esphome/issues #3264: HA Auto Discovery is broken for template select entities

kquinsland, 2022-05-05:
### The problem

I have a simple `select`: in my yaml:

```yaml
select:
  - platform: template
    internal: false
    id: tmpl_sel_active_thing
    name: "TheSelectedThing"
    optimistic: false
    options:
      - "1"
      - "2"
    set_action:
      then:
        ...
    lambda: |-
     ...
```

This yaml was working in builds done just before I did a `pip install esphome --upgrade` about an hour ago.

I did a new build and was disappointed to see that this entity was the ONLY one that wasn't automatically discoverd as part of the same device.

Here's the relevant bit from my HA logs:

```
2022-05-05 03:07:36 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_select_mqtt': ({'options': ['1', '2''], 'name': 'TheSelectedThing', 'state_topic': 'projectName/select/active_thing/state', 'command_topic': 'projectName/select/active_thing/command', 'availability_topic': 'projectName/status', 'unique_id': 'ESPselectactive_thing', 'device': {'identifiers': 'a4cf1224b30c', 'name': 'projectName', 'sw_version': None, 'model': 'mhetesp32minikit', 'manufacturer': 'espressif'}, 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 246, in async_discover
    config = schema(discovery_payload)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec
    raise e if self.msg is None else AllInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec
    v = func(v)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable
    return schema(data)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: string value is None for dictionary value @ data['device']['sw_version']
```

And this is what my MQTT broker has for the `homeassistant/select/projectName/active_thing/config`

This is what my MQTT client shows:

```
{"ops":["1","2","3","4","5","6","7","8"],"name":"TheSelectedThing","stat_t":"projectName/select/active_thing/state","cmd_t":"projectName/select/active_thing/command","avty_t":"projectName/status","uniq_id":"ESPselectactive_thing","dev":{"ids":"a4cf1224b30c","name":"projectName","sw":null,"mdl":"mhetesp32minikit","mf":"espressif"}}
```

My environment:

```shell
❯ esphome version
Version: 2022.4.0
❯ python --version
Python 3.10.4
```

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

core-2022.4.5

### What platform are you using?

ESP32

kquinsland, 2022-05-05:
I was able to recover the previous build that was working and got this from the mqtt broker:

```json
{"ops":["1","2"],"name":"TheSelectedThing","stat_t":"projectName/select/project_name/state","cmd_t":"projectName/select/project_name/command","avty_t":"projectName/status","uniq_id":"ESPselectproject_name","dev":{"ids":"807d3ab97678","name":"projectName","sw":"esphome v2022.3.2 Apr 17 2022, 11:14:08","mdl":"mhetesp32minikit","mf":"espressif"}}
```

norriscm, 2022-05-06:
I was able to resolve a related issue by reverting https://github.com/esphome/esphome/pull/3384/commits/3a9410f5f6342d56e0a752ac3123b5ab66012882

That commit causes the json component to allocate a buffer that is too small to hold the config message, which ends up truncated and invalid as a result.

kquinsland, 2022-05-18:
I am no longer able to reproduce this with the release of `2022.5.0`.  The full and proper payload is sent to the topic, now.

closed 2022-05-18 by kquinsland

derandiunddasbo, 2022-05-20:
Yes, seems to work again for meas well.