esphome/issues #3267: ble_client can only connect to one device after 2022.4.0 update

john-arvid, 2022-05-07:
### The problem

I think this pull request: https://github.com/esphome/esphome/pull/3320 broke ble_client a bit.

With firmware from version 2022.3.2 everything works ok, but after the 2022.4.0 update things stopped working as expected.

I have two am43 devices connected to my esp32, every time I reboot the esp32 one random am43 device works. It seems the connection is succesfull to each device, but only one is responsive.
I also get some errors in the logs that I don't think I got before.

### Which version of ESPHome has the issue?

2022.04.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.4.6

### What platform are you using?

ESP32

### Board

wroom-32

### Component causing the issue

ble_client

### Example YAML snippet

```yaml
esp32_ble_tracker:

ble_client:
  - mac_address: 02:14:4F:60:FA:2E
    id: lille_stuevindu
  - mac_address: 02:44:02:EC:32:6B
    id: soverom

cover:
  - platform: am43
    name: "Rullegardin lille stuevindu"
    ble_client_id: lille_stuevindu
    pin: 1234
  - platform: am43
    name: "Rullegardin soverom"
    ble_client_id: soverom
    pin: 1234

sensor:
  - platform: am43
    ble_client_id: lille_stuevindu
    battery_level:
      name: "Rullegardin lille stuevindu battery"
    illuminance:
      name: "Rullegardin lille stuevindu light"
  - platform: am43
    ble_client_id: soverom
    battery_level:
      name: "Rullegardin soverom battery"
    illuminance:
      name: "Rullegardin soverom light"
```

### Anything in the logs that might be useful for us?

```txt
[15:58:43][D][api.connection:827]: Home Assistant 2022.4.6 (::FFFF:A00:6497): Connected successfully
[15:58:52][D][ble_client:047]: Found device at MAC address [02:44:02:EC:32:6B]
[15:58:52][I][ble_client:085]: Attempting BLE connection to 02:44:02:ec:32:6b
[15:58:53][D][sensor:125]: 'Rullegardin soverom battery': Sending state nan % with 0 decimals of accuracy
[15:58:53][D][sensor:125]: 'Rullegardin soverom light': Sending state nan % with 0 decimals of accuracy
[15:58:53][W][ble_client:117]: connect to 02:44:02:ec:32:6b failed, status=133
[15:58:53][D][esp32_ble_tracker:217]: Starting scan...
[15:58:53][D][ble_client:047]: Found device at MAC address [02:14:4F:60:FA:2E]
[15:58:53][I][ble_client:085]: Attempting BLE connection to 02:14:4f:60:fa:2e
[15:58:53][D][sensor:125]: 'Rullegardin lille stuevindu battery': Sending state nan % with 0 decimals of accuracy
[15:58:53][D][sensor:125]: 'Rullegardin lille stuevindu light': Sending state nan % with 0 decimals of accuracy
[15:58:53][W][ble_client:117]: connect to 02:14:4f:60:fa:2e failed, status=133
[15:58:59][I][ble_client:085]: Attempting BLE connection to 02:44:02:ec:32:6b
[15:59:00][I][ble_client:166]: Service UUID: 0x1801
[15:59:00][I][ble_client:167]:   start_handle: 0x1  end_handle: 0x4
[15:59:00][I][ble_client:378]:  characteristic 0x2A05, handle 0x3, properties 0x20
[15:59:00][I][ble_client:166]: Service UUID: 0x1800
[15:59:00][I][ble_client:167]:   start_handle: 0x5  end_handle: 0xb
[15:59:00][I][ble_client:378]:  characteristic 0x2A00, handle 0x7, properties 0x4e
[15:59:00][I][ble_client:378]:  characteristic 0x2A01, handle 0x9, properties 0x4e
[15:59:00][I][ble_client:378]:  characteristic 0x2A04, handle 0xb, properties 0x2
[15:59:00][I][ble_client:166]: Service UUID: 0xFE50
[15:59:00][I][ble_client:167]:   start_handle: 0xc  end_handle: 0x11
[15:59:00][I][ble_client:378]:  characteristic 0xFE51, handle 0xe, properties 0x16
[15:59:00][I][am43_cover:032]: [Rullegardin soverom] Logging into AM43
[15:59:00][D][sensor:125]: 'Rullegardin soverom battery': Sending state 20.00000 % with 0 decimals of accuracy
[15:59:01][I][am43_cover:123]: [Rullegardin soverom] AM43 pin accepted.
[15:59:01][D][sensor:125]: 'Rullegardin soverom light': Sending state 0.00000 % with 0 decimals of accuracy
[15:59:01][D][cover:172]: 'Rullegardin soverom' - Publishing:
[15:59:01][D][cover:175]:   Position: 0%
[15:59:01][D][cover:188]:   Current Operation: IDLE
[15:59:04][D][ble_client:047]: Found device at MAC address [02:14:4F:60:FA:2E]
[15:59:04][I][ble_client:085]: Attempting BLE connection to 02:14:4f:60:fa:2e
[15:59:05][I][ble_client:166]: Service UUID: 0x1801
[15:59:05][I][ble_client:167]:   start_handle: 0x1  end_handle: 0x4
[15:59:05][I][ble_client:378]:  characteristic 0x2A05, handle 0x3, properties 0x20
[15:59:05][I][ble_client:166]: Service UUID: 0x1800
[15:59:05][I][ble_client:167]:   start_handle: 0x5  end_handle: 0xb
[15:59:05][I][ble_client:378]:  characteristic 0x2A00, handle 0x7, properties 0x4e
[15:59:05][I][ble_client:378]:  characteristic 0x2A01, handle 0x9, properties 0x4e
[15:59:05][I][ble_client:378]:  characteristic 0x2A04, handle 0xb, properties 0x2
[15:59:05][I][ble_client:166]: Service UUID: 0xFE50
[15:59:05][I][ble_client:378]:  characteristic 0xFE52, handle 0x11, properties 0x8
[15:59:05][D][esp32_ble_tracker:217]: Starting scan...
[15:59:06][W][ble_client:177]: No descriptor found for notify of handle 0xe
[15:59:06][I][am43_cover:032]: [Rullegardin lille stuevindu] Logging into AM43
[15:59:06][D][sensor:125]: 'Rullegardin lille stuevindu battery': Sending state 80.00000 % with 0 decimals of accuracy
[15:59:21][D][cover:078]: 'Rullegardin lille stuevindu' - Setting
[15:59:21][D][cover:086]:   Position: 0%
[15:59:27][D][cover:078]: 'Rullegardin lille stuevindu' - Setting
[15:59:27][D][cover:082]:   Command: STOP
[15:59:27][D][cover:172]: 'Rullegardin lille stuevindu' - Publishing:
[15:59:27][D][cover:175]:   Position: 68%
[15:59:27][D][cover:188]:   Current Operation: IDLE
[15:59:28][D][cover:078]: 'Rullegardin lille stuevindu' - Setting
[15:59:28][D][cover:086]:   Position: 100%
[15:59:30][D][cover:078]: 'Rullegardin soverom' - Setting
[15:59:30][D][cover:086]:   Position: 100%
[15:59:33][D][cover:172]: 'Rullegardin lille stuevindu' - Publishing:
[15:59:33][D][cover:175]:   Position: 100%
[15:59:33][D][cover:188]:   Current Operation: IDLE
[15:59:34][D][cover:078]: 'Rullegardin soverom' - Setting
[15:59:34][D][cover:082]:   Command: STOP
[15:59:36][D][cover:078]: 'Rullegardin soverom' - Setting
[15:59:36][D][cover:086]:   Position: 100%
[15:59:42][D][cover:078]: 'Rullegardin soverom' - Setting
[15:59:42][D][cover:082]:   Command: STOP
```

### Additional information

Reffering to the logs, the device that worked that time was client with mac: 02:14:4F:60:FA:2E

Let me know if I should get some other logs.
I can't downgrade unless you know of a way I can revert to a previous esphome version. (I don't have a backup of that update)

john-arvid, 2022-05-07:
This commit: https://github.com/esphome/esphome/pull/3320/commits/b124948299966dcf11208cafc8319271f276160d

jascdk, 2022-05-08:
I have exactly the same problem :( Anyone have a quick fix?

john-arvid, 2022-05-08:
Not a quick fix, but you can install esphome on another device and create the bin file there. 
I installed it on my mac with `pip3 install esphome==2022.3.2`
Then made a config folder `mkdir config`
Then started the dashboard `esphome dashboard config/`
Created a new device with the same name and copied the contect of secrets and the yaml file.

Before this I tried to revert the ble_client.cpp, ble_client.h, esp32_ble_tracker.cpp and esp32_ble_tracker.h files to the 2022.3.2 version. I got the correct files reverted, but it didn't solve the problem.
So it could be something else that caused this, but I can't be sure.

But I am now 100% sure that 2022.3.2 works as expected, and 2022.4.0 breaks something.

john-arvid, 2022-05-10:
Maybe @ammmze or @jesserockz have some thoughts? (Contributor and merger)
With 2022.3.2 it looks like it tries connecting to a device many times, with 2022.4.0 it only tries one time, then gives up reconnecting.

jascdk, 2022-05-10:
> Not a quick fix, but you can install esphome on another device and create the bin file there. I installed it on my mac with `pip3 install esphome==2022.3.2` Then made a config folder `mkdir config` [...]

That could be an option to do, but I don't have any other instance for the time right now to build an older release on. I hope the merger and contributor finds a solution soon - my wife is not happy about that only one blind works at the moment :D

brainboost, 2022-05-14:
> This commit: [esphome/esphome@b124948](https://github.com/esphome/esphome/commit/b124948299966dcf11208cafc8319271f276160d)

@ammmze

ammmze, 2022-05-14:
I'm out of town on mobile and can't do anything for the next few days. But can you also post logs running with verbose log level. As a temporary work around you can possibly try using old versions of the blue client and esp32 ble tracker using [external components](https://esphome.io/components/external_components.html) referencing the last tag of esphome that worked.

buxtronix, 2022-05-15:
I am successfully able to connect to 2 AM43 devices with the following:

```
external_components:
  - source: github://pr#3320
    components: [ ble_client, am43, esp32_ble_tracker, cover ]
```
I'll play around further to try and reproduce.

buxtronix, 2022-05-15:
2022.6.0-dev also connects to both for me.

jascdk, 2022-05-15:
> I'm out of town on mobile and can't do anything for the next few days. But can you also post logs running with verbose log level. As a temporary work around you can possibly try using old versions [...]

How to do it :) ?

sambeetm, 2022-05-26:
> 2022.6.0-dev also connects to both for me.

How do you install 2022.60-dev ?
I installed the EspHome(Dev) Aadon. Installed the bin. Still not connecting to more than 1.

brainboost, 2022-05-26:
What was working for me, rolling back to 2022.3.2 I made editing container properties in the Portainer (my setup is on docker).  I changed the image to ghcr.io/esphome/esphome-hassio-aarch64:2022.3.2 that is used to work fine. After that I had to skip updates of EspHome in the Hassio UI. I had no chance to install 2022.60-dev mentioned above.

sambeetm, 2022-05-27:
External components did not work for me either....Here is my .yaml. Is it correct ?
```
esphome:
  name: guest-blinds-group
  platform: ESP32
  board: esp32dev

external_components:
  - source: github://pr#3320
    components: [ ble_client, am43, esp32_ble_tracker, cover ]

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "password"

wifi:
  ssid: "xxxxxxx"
  password: "xxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "yyyyy"
    password: "yyyyyy"
    
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.11
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

captive_portal:

esp32_ble_tracker:

ble_client:
  - mac_address: 02:05:FC:30:61:00
    id: guest_am43
  - mac_address: 02:5b:4b:f8:00:99
    id: jack_am43
  - mac_address: 02:98:2b:00:a8:b7
    id: guestbath_am43

cover:
  - platform: am43
    name: "Guest Room Blinds"
    id: guest_cover
    ble_client_id: guest_am43
  - platform: am43
    name: "Jack Room Blinds"
    id: jack_cover
    ble_client_id: jack_am43
  - platform: am43
    name: "Guest Bath Blinds"
    id: guestbath_cover
    ble_client_id: guestbath_am43

sensor:
  - platform: am43
    ble_client_id: guest_am43
    battery_level:
      name: "Guest Blinds Battery"
    illuminance:
      name: "Guest Blinds Light"
    update_interval: 120s
  - platform: am43
    ble_client_id: jack_am43
    battery_level:
      name: "Jack Blinds Battery"
    illuminance:
      name: "Jack Blinds Light"
    update_interval: 120s
  - platform: am43
    ble_client_id: guestbath_am43
    battery_level:
      name: "Guest Bath Blinds Battery"
    illuminance:
      name: "Guest Bath Blinds Light"
    update_interval: 120s

switch:
  - platform: restart
    name: "Guest Group Restart"
  - platform: ble_client
    name: "Guest Blinds Switch"
    ble_client_id: guest_am43
  - platform: ble_client
    name: "Jack Blinds Switch"
    ble_client_id: jack_am43
  - platform: ble_client
    name: "Guest Bath Blinds Switch"
    ble_client_id: guestbath_am43
```

jascdk, 2022-09-16:
@buxtronix any fixes ?:)

jhansche, 2022-10-19:
I'm also noticing this with the `bedjet` component. My PR from esphome/esphome#3544 resolved some issues with mixing up `conn_id` (because the `conn_id` from `ESP_GATTC_CONNECT_EVT` was not the same/right `conn_id` from `ESP_GATTC_OPEN_EVT`).

As of that commit merging (into 2022.8, and cherry-picked as esphome/esphome@4c37c17d into 2022.6.2), I was seeing the two devices working correctly at the time. But now after updating to 2022.10, connecting to 2 `bedjet` devices causes both HA devices to "share" one connection; that is: the state of one unit is appearing as the state for both units in HA. It's also unable to obtain BLE characteristics for the "second" device because the BLEClient would have already flushed the discovered characteristics after the "first" device was transitioned to connected state.

I had not updated the device since my related PR for BedJet https://github.com/esphome/esphome/pull/3522 merged into 2022.8, so this seems to be a regression between 2022.8 and 2022.10. It's possible this might not be the same issue as OP, considering the mention of 2022.4 (mine was working at that time, and up through 2022.8).

danielbheyl, 2022-10-19:
> I'm also noticing this with the `bedjet` component. My PR from [esphome/esphome#3544](https://github.com/esphome/esphome/pull/3544) resolved some issues with mixing up `conn_id` (because the `conn_id` from `ESP_GATTC_CONNECT_EVT` was not the same/right `conn_id` from `ESP_GATTC_OPEN_EVT`).
> 
> As of that commit merging (into 2022.8, and cherry-picked as [esphome/esphome@4c37c17d](https://github.com/esphome/esphome/commit/4c37c17d) into 2022.6.2), I was seeing the two devices working correctly at the time. But now after updating to 2022.10, connecting to 2 `bedjet` devices causes both HA devices to "share" one connection; that is: the state of one unit is appearing as the state for both units in HA. It's also unable to obtain BLE characteristics for the "second" device because the BLEClient would have already flushed the discovered characteristics after the "first" device was transitioned to connected state.
> 
> I had not updated the device since my related PR for BedJet [esphome/esphome#3522](https://github.com/esphome/esphome/pull/3522) merged into 2022.8, so this seems to be a regression between 2022.8 and 2022.10. It's possible this might not be the same issue as OP, considering the mention of 2022.4 (mine was working at that time, and up through 2022.8).

I have this same issue with 2022.10 and the bedjet component.  One bedjet will connect fine (it varies - probably the first one to send a signal) but the other refuses to connect.  The log reads:
"[MAC ADDRESS OF NON-CONNECTED UNIT] Unexpected notify handle, wanted 0000, got 002A"
Any help would be appreciated.

jhansche, 2022-10-19:
> "Unexpected notify handle, wanted 0000, got 002A"

Yes, this is exactly what I was seeing and what caused me to start looking into it and notice that the state was being shared. The reason for the warning is what's mentioned above, because of BLEClient flushing characteristics after connect (the first one connects and gets its characteristics, and then the 2nd one transitions to connected but cannot discover characteristics because the BLEClient has already deleted them).

When it's working correctly, the BLEClient should be maintaining those services and characteristics for each distinct device connection - and right now those appear to be getting crossed somewhere.

danielbheyl, 2022-11-17:
I can confirm that this issue has been resolved as of 2022.11.1.  Two bedjet BLE devices now connect at the same time.  I think that it was fixed by this change: https://github.com/esphome/esphome/pull/3971.

brainboost, 2022-11-20:
> I can confirm that this issue has been resolved as of 2022.11.1. Two bedjet BLE devices now connect at the same time. I think that it was fixed by this change: [esphome/esphome#3971](https://github.com/esphome/esphome/pull/3971).

I also spotted my all three ble blind motors started reporting independently after the update.

closed 2022-11-21 by john-arvid