esphome/issues #3274: climate.bedjet - constantly writing to flash will destroy it!?

user897943, 2022-05-10:
### The problem

Surely this should not be writing to flash on every update?

```
[13:04:13][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-95
[13:04:13][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:04:24][D][esp32.preferences:114]: Saving preferences to flash...
[13:04:40][I][ota:105]: Boot seems successful, resetting boot loop counter.
[13:04:40][D][esp32.preferences:114]: Saving preferences to flash...
[13:04:41][D][climate:385]: 'Bedjet - Front Bedroom 1' - Sending state:
[13:04:41][D][climate:388]:   Mode: FAN_ONLY
[13:04:41][D][climate:390]:   Action: COOLING
[13:04:41][D][climate:393]:   Fan Mode: OFF
[13:04:41][D][climate:396]:   Custom Fan Mode:  45%
[13:04:41][D][climate:408]:   Current Temperature: 21.00°C
[13:04:41][D][climate:414]:   Target Temperature: 19.00°C
[13:04:50][D][esp32_ble_tracker:217]: Starting scan...
[13:04:50][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-89
[13:04:50][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:04:50][D][esp32_ble_tracker:749]:   Name: 'Front bedroom TV'
[13:04:50][D][esp32_ble_tracker:751]:   TX Power: 2
[13:04:51][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-91
[13:04:51][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:04:51][D][esp32_ble_tracker:751]:   TX Power: 2
[13:04:52][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-81
[13:04:52][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:04:53][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-76
[13:04:53][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:04:54][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-73
[13:04:54][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:04:54][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-77
[13:04:54][D][esp32_ble_tracker:747]:   Address Type: PUBLIC
[13:04:54][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-93
[13:04:54][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:04:56][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-94
[13:04:56][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:04:56][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-89
[13:04:56][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:04:56][D][esp32_ble_tracker:751]:   TX Power: 2
[13:04:57][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-85
[13:04:57][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:05:01][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-97
[13:05:01][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:05:01][D][esp32_ble_tracker:749]:   Name: 'Echo Dot-HM9'
[13:05:01][D][esp32_ble_tracker:751]:   TX Power: 2
[13:05:02][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-72
[13:05:02][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:05:03][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-86
[13:05:03][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:05:11][D][climate:385]: 'Bedjet - Front Bedroom 1' - Sending state:
[13:05:11][D][climate:388]:   Mode: FAN_ONLY
[13:05:11][D][climate:390]:   Action: COOLING
[13:05:11][D][climate:393]:   Fan Mode: OFF
[13:05:11][D][climate:396]:   Custom Fan Mode:  45%
[13:05:11][D][climate:408]:   Current Temperature: 21.50°C
[13:05:11][D][climate:414]:   Target Temperature: 19.00°C
[13:05:12][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-93
[13:05:12][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:05:12][D][esp32_ble_tracker:726]: Found device xxxx RSSI=-98
[13:05:12][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:05:24][D][esp32.preferences:114]: Saving preferences to flash...
[13:05:36][D][esp32_ble_tracker:726]: Found device xxxx  RSSI=-72
[13:05:36][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:05:41][D][climate:385]: 'Bedjet - Front Bedroom 1' - Sending state:
[13:05:41][D][climate:388]:   Mode: FAN_ONLY
[13:05:41][D][climate:390]:   Action: COOLING
[13:05:41][D][climate:393]:   Fan Mode: OFF
[13:05:41][D][climate:396]:   Custom Fan Mode:  45%
[13:05:41][D][climate:408]:   Current Temperature: 21.00°C
[13:05:41][D][climate:414]:   Target Temperature: 19.00°C
[13:05:58][D][esp32_ble_tracker:726]: Found device xxxx  RSSI=-86
[13:05:58][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:05:58][D][esp32_ble_tracker:749]:   Name: 'Front bedroom TV'
[13:05:58][D][esp32_ble_tracker:751]:   TX Power: 2
[13:06:04][D][esp32_ble_tracker:726]: Found device xxxx  RSSI=-92
[13:06:04][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:06:11][D][climate:385]: 'Bedjet - Front Bedroom 1' - Sending state:
[13:06:11][D][climate:388]:   Mode: FAN_ONLY
[13:06:11][D][climate:390]:   Action: COOLING
[13:06:11][D][climate:393]:   Fan Mode: OFF
[13:06:11][D][climate:396]:   Custom Fan Mode:  45%
[13:06:11][D][climate:408]:   Current Temperature: 21.00°C
[13:06:11][D][climate:414]:   Target Temperature: 19.00°C
[13:06:19][D][esp32_ble_tracker:726]: Found device  RSSI=-98
[13:06:19][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:06:24][D][esp32.preferences:114]: Saving preferences to flash...
[13:06:24][D][esp32_ble_tracker:726]: Found device xxxx  RSSI=-96
[13:06:24][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:06:41][D][climate:385]: 'Bedjet - Front Bedroom 1' - Sending state:
[13:06:41][D][climate:388]:   Mode: FAN_ONLY
[13:06:41][D][climate:390]:   Action: COOLING
[13:06:41][D][climate:393]:   Fan Mode: OFF
[13:06:41][D][climate:396]:   Custom Fan Mode:  45%
[13:06:41][D][climate:408]:   Current Temperature: 21.00°C
[13:06:41][D][climate:414]:   Target Temperature: 19.00°C
[13:07:11][D][climate:385]: 'Bedjet - Front Bedroom 1' - Sending state:
[13:07:11][D][climate:388]:   Mode: FAN_ONLY
[13:07:11][D][climate:390]:   Action: COOLING
[13:07:11][D][climate:393]:   Fan Mode: OFF
[13:07:11][D][climate:396]:   Custom Fan Mode:  45%
[13:07:11][D][climate:408]:   Current Temperature: 21.00°C
[13:07:11][D][climate:414]:   Target Temperature: 19.00°C
[13:07:24][D][esp32.preferences:114]: Saving preferences to flash...
[13:07:41][D][climate:385]: 'Bedjet - Front Bedroom 1' - Sending state:
[13:07:41][D][climate:388]:   Mode: FAN_ONLY
[13:07:41][D][climate:390]:   Action: COOLING
[13:07:41][D][climate:393]:   Fan Mode: OFF
[13:07:41][D][climate:396]:   Custom Fan Mode:  45%
[13:07:41][D][climate:408]:   Current Temperature: 21.00°C
[13:07:41][D][climate:414]:   Target Temperature: 19.00°C
[13:08:11][D][climate:385]: 'Bedjet - Front Bedroom 1' - Sending state:
[13:08:11][D][climate:388]:   Mode: FAN_ONLY
[13:08:11][D][climate:390]:   Action: COOLING
[13:08:11][D][climate:393]:   Fan Mode: OFF
[13:08:11][D][climate:396]:   Custom Fan Mode:  45%
[13:08:11][D][climate:408]:   Current Temperature: 21.00°C
[13:08:11][D][climate:414]:   Target Temperature: 19.00°C
[13:08:24][D][esp32.preferences:114]: Saving preferences to flash...
```

### Which version of ESPHome has the issue?

2022.5.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Not applicable

### What platform are you using?

ESP32

### Board

wemos_d1_mini32

### Component causing the issue

bedjet

### Anything in the logs that might be useful for us?

```txt
Shown above
```

jhansche, 2022-05-27:
Looks like the log you're talking about is happening every 60 seconds, which is likely the standard in HA/ESPHome. The purpose of saving state is to allow restoring that state on startup.

Looks like there is an open feature request to allow customizing the save interval: https://github.com/esphome/feature-requests/issues/461 which could in theory also apply to any device not just this specific one.

However, I think you're being a bit dramatic: "constantly writing to flash will destroy it!?"
BedJet component is not doing anything special to get this functionality: it's automatic. The only thing that BedJet does is actually make use of the existing/automatic state that is already saved. But even if we didn't, the *writes* would still be happening regardless.

So if writing once a minute would destroy the device, don't you think that the core esphome maintainers would have figured that out by now?

user897943, 2022-05-27:
> Looks like the log you're talking about is happening every 60 seconds (which is in fact _not_ every update, if you look at your own logs), and that is likely the standard in either HA/ESPHome. The purpose of saving state is to allow restoring that state on restart.
> 
> Looks like there is an open feature request to allow customizing the save interval: [esphome/feature-requests#461](https://github.com/esphome/feature-requests/issues/461) which could in theory also apply to any device not just this specific one.
> 
> However, I think you're being a bit dramatic: "constantly writing to flash will destroy it!?" BedJet component is not doing anything special to get this functionality: it's automatic. The only thing that BedJet does is actually make use of the existing/automatic state that is already saved. But even if we didn't, the _writes_ would still be happening regardless.
> 
> So if writing once a minute would destroy the device, don't you think that the core esphome maintainers would have figured that out by now?

First the "dramatic" thing. I don't think it's constructive to use terms like that. I am not being dramatic, hardware failure is not fun for anybody, and flash storage has a limited number of programming cycles. I didn't know that ESPHome writes to flash every 60 seconds. I think it's perfectly fine to question why things are done a certain way. Why did you write the integration instead of using the already made app? It's because you didn't agree with the status quo. Were you being dramatic? No. You think it could be improved. I think it should only write values to flash if they change, not every 60 seconds, what is the logic rationale for writing data that hasn't changed? It's a waste of write cycles and processing power, even milliwatts have a cumulative cost.

jhansche, 2022-05-28:
And I'll again point you to the fact that this is not a bedjet problem, it's built into esphome, and if you want it changed the issue should be for esphome core, where people responsible for that can explain it better.

user897943, 2022-05-28:
> And I'll again point you to the fact that this is not a bedjet problem, it's built into esphome, and if you want it changed the issue should be for esphome core, where people responsible for that can explain it better.

Yep yep I got that, I'll add some comments to the issue you linked, thanks.

jhansche, 2022-06-06:
@user897943 looks like you can configure this in your config: https://esphome.io/components/esphome.html#adjusting-flash-writes
It defaults to every 60sec. You can increase that if flash durability is a concern.

user897943, 2022-06-06:
That's awesome, thanks a lot!!

closed 2022-06-06 by user897943