esphome/issues #3275: Pronto code is not sent via remote_transmitter

rmeissn, 2022-05-10:
### The problem

I'm trying to reverse-engineer a climate controls IR remote and started reading the IR codes. As a simple test to turn the device on and off I set up a template switch, sending pronto codes via an IR transmitter. The IR receiver, used to log the IR codes, is still wired.

When I set it up as a ir-climate, the IR receiver records sent codes by the transmitter. When I set it up as a template switch, used to send pronto codes, the IR receiver won't notice anything. I can only guess that nothing is sent via the IR diode.

I think I've implemented the template switch the correct way (see yaml and https://esphome.io/components/remote_transmitter.html?highlight=transmitter#setting-up-infrared-devices )

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

remote_transmitter

### Example YAML snippet

```yaml
esphome:
  name: test
  platform: ESP8266
  board: nodemcuv2

logger:
  baud_rate: 0

api:

ota:
  password: "password"

wifi:
  ssid: "hidden"
  password: "password"
  
remote_receiver:
  pin: 
    number: D4
    inverted: true
  dump: all
  
remote_transmitter:
  pin: D7
  carrier_duty_percent: 50%
  
switch:
  - platform: template
    name: "Test"
    turn_on_action:
      - remote_transmitter.transmit_pronto:
          data: "0000 006D 0072 0000 0085 0040 0013 002F 0010 0031 0013 000E 0013 000E 0014 000E 0013 002F 0013 000E 0013 000E 0013 002F 0013 002F 0012 000E 0013 0030 0012 000E 0014 000E 0013 0030 0012 0030 0012 000D 0014 002F 0013 002F 0013 000E 0013 000E 0013 0030 0012 000E 0014 000E 0013 0030 0010 0011 0013 000D 0014 000E 0013 000E 0013 000D 0014 000E 0013 000E 0014 000E 0011 0011 0013 000E 0014 000E 0013 000E 0014 000E 0013 000E 0013 000E 001"
    turn_off_action:
      - remote_transmitter.transmit_pronto:
          data: "0000 006D 0072 0000 0085 0040 0012 002F 0012 0031 0011 000F 0013 000E 0014 000E 0013 002F 0013 000E 0013 000E 0013 002F 0014 002E 0012 000F 0013 0030 0012 000D 0014 000E 0013 002F 0013 0030 0012 000E 0013 002F 0013 002F 0013 000E 0013 000E 0013 0030 0012 000E 0014 000E 0013 0030 0010 0010 0013 000E 0014 000E 0013 000E 0013 000E 0013 000E 0013 000E 0014 000E 0012 0010 0013 000E 0014 000E 0013 000E 0014 000E 0013 000E 0013 000E 001"
```

### Anything in the logs that might be useful for us?

```txt
[19:32:01][D][switch:013]: 'Test' Turning ON.
[19:32:01][D][remote.pronto:101]: Send Pronto: frequency=38kHz
[19:32:01][D][remote.pronto:106]: Send Pronto: intros=228
[19:32:01][D][remote.pronto:107]: Send Pronto: repeats=0
[19:32:01][D][remote_transmitter:075]: Sending remote code...
```

ssieb, 2022-08-22:
That pronto code is not valid which is why you were getting strange errors from the transmitter.

closed 2022-08-23 by jesserockz

rmeissn, 2022-08-24:
@ssieb In which way is the pronto code wrong? That's precisely the code, that was recorded by an IR receiver.

ssieb, 2022-08-24:
It's truncated.  Notice how the last number only has 3 digits.  The pronto string is too long for the log message so it got cut off.  I will see about fixing that.

ssieb, 2022-09-15:
@rmeissn, can you try https://github.com/esphome/esphome/pull/3812 to see if it gives you the whole code?