esphome/issues #3276: shtc3 sensor under virtual i2c bus using tca9548a

matteobusolin, 2022-05-11:
### The problem

Hi,
yesterday I tried to use the SHTC3 sensor under a TCA9548a i2c bus multiplexer. When the ESP32-POE-ISO board boot I can see from the logs I have a CRC error when the system is trying to read the sensor ID. If I connect directly the sensor to the i2c bus (without the TCA9548a) I can use the sensor with no problem.
I checked the TCA9548a connecting other devices in multiple virtual i2c bus line with no problem. When I connect also the SHTC3 the entirely virtual i2c bus line doesn't work.
I checked the cable connection (it's 5cm long) and there's no problem from it.

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

ESP32-POE-ISO

### Component causing the issue

i2c shtc3

### Example YAML snippet

```yaml
i2c:
  sda: 13
  scl: 16
  scan: true
  id: bus_a
  frequency: 50kHz

tca9548a:
  - address: 0x70
    id: multiplex0
    i2c_id: bus_a
    channels:
      - bus_id: multiplex0channel0
        channel: 0
      - bus_id: multiplex0channel1
        channel: 1
      - bus_id: multiplex0channel2
        channel: 2
      - bus_id: multiplex0channel3
        channel: 3
      - bus_id: multiplex0channel4
        channel: 4
      - bus_id: multiplex0channel5
        channel: 5
      - bus_id: multiplex0channel6
        channel: 6
      - bus_id: multiplex0channel7
        channel: 7

pcf8574:
  - id: 'pcf8574_hub_1'
    i2c_id: multiplex0channel0
    address: 0x20
    pcf8575: false

sensor:
  - platform: shtcx
    i2c_id: multiplex0channel2
    temperature:
      name: "Room Temperature"
    humidity:
      name: "Room Humidity"
    address: 0x70
    update_interval: 12s

  - platform: bh1750
    i2c_id: multiplex0channel2
    name: "Room Illuminance"
    address: 0x23
    update_interval: 12s
```

jesserockz, 2022-05-12:
Can you try on the latest beta release? There was a PR to fix the channels in the `tca9548a` recently

matteobusolin, 2022-05-12:
> Can you try on the latest beta release? There was a PR to fix the channels in the `tca9548a` recently

I tried (using esphome versione 2022.5.0b1) with no result.
For more information here the log:

`[08:55:05][I][logger:242]: Log initialized
[08:55:05][C][ota:461]: There have been 1 suspected unsuccessful boot attempts.
[08:55:05][D][esp32.preferences:114]: Saving preferences to flash...
[08:55:05][I][app:029]: Running through setup()...
[08:55:05][I][i2c.arduino:161]: Performing I2C bus recovery
[08:55:05][C][tca9548a:023]: Setting up TCA9548A...
[08:55:05][D][tca9548a:030]: Channels currently open: 1
[08:55:05][C][pcf8574:010]: Setting up PCF8574...
**[08:55:05][C][shtcx:028]: Setting up SHTCx...
[08:55:05][E][sensirion_i2c:026]: CRC8 Checksum invalid at pos 0! 0xC8 != 0xE2
[08:55:05][E][shtcx:040]: Error reading Device ID**
[08:55:05][E][component:112]: Component shtcx.sensor was marked as failed.
[08:55:05][C][bh1750.sensor:041]: Setting up BH1750 'Room Illuminance'...
[08:55:05][E][component:112]: Component bh1750.sensor was marked as failed.
[08:55:05][C][wifi:037]: Setting up WiFi...
[08:55:05][D][wifi:373]: Starting scan...
[08:55:05][D][binary_sensor:034]: 'PCF8574 Hub 1 Pin #0': Sending initial state OFF
[08:55:05][D][binary_sensor:034]: 'PCF8574 Hub 1 Pin #1': Sending initial state OFF
[08:55:05][D][binary_sensor:034]: 'PCF8574 Hub 1 Pin #2': Sending initial state OFF
[08:55:05][D][binary_sensor:034]: 'PCF8574 Hub 1 Pin #3': Sending initial state OFF
[08:55:05][D][binary_sensor:034]: 'PCF8574 Hub 1 Pin #4': Sending initial state OFF
[08:55:05][D][binary_sensor:034]: 'PCF8574 Hub 1 Pin #5': Sending initial state OFF
[08:55:05][D][binary_sensor:034]: 'PCF8574 Hub 1 Pin #6': Sending initial state OFF
[08:55:05][D][binary_sensor:034]: 'PCF8574 Hub 1 Pin #7': Sending initial state OFF
......
[08:55:08][D][wifi:514]: Disabling AP...
[08:55:08][C][web_server:096]: Setting up web server...
[08:55:08][C][ota:085]: Over-The-Air Updates:
[08:55:08][C][ota:086]:   Address: fatboy.local:3232
[08:55:08][C][ota:089]:   Using Password.
[08:55:08][C][api:025]: Setting up Home Assistant API server...
[08:55:08][I][app:062]: setup() finished successfully!
[08:55:08][I][app:102]: ESPHome version 2022.5.0b1 compiled on May 12 2022, 08:54:34
......
[08:55:08][C][logger:274]: Logger:
[08:55:08][C][logger:275]:   Level: DEBUG
[08:55:08][C][logger:276]:   Log Baud Rate: 115200
[08:55:08][C][logger:277]:   Hardware UART: UART0
[08:55:08][C][i2c.arduino:038]: I2C Bus:
[08:55:08][C][i2c.arduino:039]:   SDA Pin: GPIO13
[08:55:08][C][i2c.arduino:040]:   SCL Pin: GPIO16
[08:55:08][C][i2c.arduino:041]:   Frequency: 50000 Hz
[08:55:08][C][i2c.arduino:044]:   Recovery: bus successfully recovered
[08:55:08][I][i2c.arduino:054]: Results from i2c bus scan:
[08:55:08][I][i2c.arduino:060]: Found i2c device at address 0x20
[08:55:08][I][i2c.arduino:060]: Found i2c device at address 0x70
[08:55:08][C][tca9548a:033]: TCA9548A:
[08:55:08][C][tca9548a:034]:   Address: 0x70
[08:55:08][C][pcf8574:021]: PCF8574:
[08:55:08][C][pcf8574:022]:   Address: 0x20
[08:55:08][C][pcf8574:023]:   Is PCF8575: NO
[08:55:09][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'PCF8574 Hub 1 Pin #0'
[08:55:09][C][gpio.binary_sensor:016]:   Pin: 0 via PCF8574
[08:55:09][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'PCF8574 Hub 1 Pin #1'
[08:55:09][C][gpio.binary_sensor:016]:   Pin: 1 via PCF8574
[08:55:09][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'PCF8574 Hub 1 Pin #2'
[08:55:09][C][gpio.binary_sensor:016]:   Pin: 2 via PCF8574
[08:55:09][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'PCF8574 Hub 1 Pin #3'
[08:55:09][C][gpio.binary_sensor:016]:   Pin: 3 via PCF8574
[08:55:09][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'PCF8574 Hub 1 Pin #4'
[08:55:09][C][gpio.binary_sensor:016]:   Pin: 4 via PCF8574
[08:55:09][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'PCF8574 Hub 1 Pin #5'
[08:55:09][C][gpio.binary_sensor:016]:   Pin: 5 via PCF8574
[08:55:09][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'PCF8574 Hub 1 Pin #6'
[08:55:09][C][gpio.binary_sensor:016]:   Pin: 6 via PCF8574
[08:55:09][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'PCF8574 Hub 1 Pin #7'
[08:55:09][C][gpio.binary_sensor:016]:   Pin: 7 via PCF8574
[08:55:09][C][shtcx:059]: SHTCx:
[08:55:09][C][shtcx:060]:   Model: SHTC3 (0000)
[08:55:09][C][shtcx:061]:   Address: 0x70
[08:55:09][E][shtcx:063]: Communication with SHTCx failed!
[08:55:09][C][shtcx:065]:   Update Interval: 12.0s
[08:55:09][C][shtcx:067]:   Temperature 'Room Temperature'
[08:55:09][C][shtcx:067]:     Device Class: 'temperature'
[08:55:09][C][shtcx:067]:     State Class: 'measurement'
[08:55:09][C][shtcx:067]:     Unit of Measurement: '°C'
[08:55:09][C][shtcx:067]:     Accuracy Decimals: 1
[08:55:09][C][shtcx:068]:   Humidity 'Room Humidity'
[08:55:09][C][shtcx:068]:     Device Class: 'humidity'
[08:55:09][C][shtcx:068]:     State Class: 'measurement'
[08:55:09][C][shtcx:068]:     Unit of Measurement: '%'
[08:55:09][C][shtcx:068]:     Accuracy Decimals: 1
[08:55:09][C][bh1750.sensor:118]: BH1750 'Room Illuminance'
[08:55:09][C][bh1750.sensor:118]:   Device Class: 'illuminance'
[08:55:09][C][bh1750.sensor:118]:   State Class: 'measurement'
[08:55:09][C][bh1750.sensor:118]:   Unit of Measurement: 'lx'
[08:55:09][C][bh1750.sensor:118]:   Accuracy Decimals: 1
[08:55:09][C][bh1750.sensor:119]:   Address: 0x23
[08:55:09][E][bh1750.sensor:121]: Communication with BH1750 failed!
[08:55:09][C][bh1750.sensor:124]:   Update Interval: 12.0s
[08:55:09][C][captive_portal:088]: Captive Portal:
[08:55:09][C][web_server:129]: Web Server:`

andreashergert1984, 2022-05-12:
> multiple

Did you remember to put also pullups on the i2c-bus BEHIND the TCA? since it is a completely new bus. 
vG Andi

matteobusolin, 2022-05-12:
> > multiple
> 
> Did you remember to put also pullups on the i2c-bus BEHIND the TCA? since it is a completely new bus. vG Andi

I'm usign this 2 boards:
https://ibb.co/0JJ0gYw
https://ibb.co/MhMZqwr

and if I attach only the BH1750 i2c sensor, the virtual bus has no problem at all.

andreashergert1984, 2022-05-12:
> 

please try to seperate the sensor each to a seperate virt-channel.. i had this in the past with sensors,e.g. bpm280 and 2320 will not work nicely on the same bus (despite different addresses). 

I assume here more a hardware issue (Signal Integrity) than a software issue. 

lG Andi

matteobusolin, 2022-05-13:
> > 
> 
> please try to seperate the sensor each to a seperate virt-channel.. i had this in the past with sensors,e.g. bpm280 and 2320 will not work nicely on the same bus (despite different addresses).
> 
> I assume here more a hardware issue (Signal Integrity) than a software issue.
> 
> lG Andi

I already tried and the SHTC3 sensor doesn't work even alone. If I move it to the main i2c bus line it has no problem

jesserockz, 2022-05-15:
Just looking at your logs, could it be an issue with the CHTC3 being on address 0x70 (which is the same as the `tca9548a`)? I dont have these so I cannot test that myself.

matteobusolin, 2022-05-15:
> Just looking at your logs, could it be an issue with the CHTC3 being on address 0x70 (which is the same as the `tca9548a`)? I dont have these so I cannot test that myself.

Interesting question. If this is the problem should be a software problem because they are connected to two different i2c buses (one real, directly from the esp32 and the other virtual). When I try to comunicate with SHTC3 I refer to **i2c_id: multiplex0channel2** which is different to **i2c_id: bus_a**

jesserockz, 2022-05-15:
Yes, but the tca9548a is still on that "virtual" i2c bus, at address 0x70.

The "virtual" bus is nothing more than code toggling the channel pins before sending the message on the esp32 hardware bus.

Can you change the address of the `tca9548a` with the A0/A1/A2 pins to test this?

andreashergert1984, 2022-05-15:
> Yes, but the tca9548a is still on that "virtual" i2c bus, at address 0x70.
> 
> The "virtual" bus is nothing more than code toggling the channel pins before sending the message on the esp32 hardware bus.
> 
> Can you change the address of the `tca9548a` with the A0/A1/A2 pins to test this?

for sure.. i did not see the same address first.. this cannot work with the same address..

matteobusolin, 2022-05-15:
> Yes, but the tca9548a is still on that "virtual" i2c bus, at address 0x70.
> 
> The "virtual" bus is nothing more than code toggling the channel pins before sending the message on the esp32 hardware bus.
> 
> Can you change the address of the `tca9548a` with the A0/A1/A2 pins to test this?

Yes, I can from the board. I will do it ASAP, thanks for the suggestion. I will tell you the result ^^

matteobusolin, 2022-05-15:
> Yes, but the tca9548a is still on that "virtual" i2c bus, at address 0x70.
> 
> The "virtual" bus is nothing more than code toggling the channel pins before sending the message on the esp32 hardware bus.
> 
> Can you change the address of the `tca9548a` with the A0/A1/A2 pins to test this?

You are the man! Thanks to solve the problem! I putted the TCA9548A id to 0x71 and now everything is working with no problem. I was thinking it wasn't a problem 'cos two different i2c bus

matteobusolin, 2022-05-15:
thanks to everyone for the support! You both are amazing :)

closed 2022-05-23 by matteobusolin