esphome/issues #3277: Boot Loop on AI Thinker ESP32-C Modules (2Mb Variant)

dmberrycom, 2022-05-13:
### The problem

ESPHome image on ESP-C3-32S (2Mb Variant) is stuck in a boot loop after programming 

ESP-C3-13U (4Mb Variant) - Works Fine
ESP-C3-32S (2M Variant) - Boot Loop

https://docs.ai-thinker.com/en/esp32c3

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.4

### What platform are you using?

ESP32-IDF

### Board

Ai thinker ESP32-C3-32-S - 2MB 

### Component causing the issue

esphome

### Example YAML snippet

```yaml
esphome:
  name: esp32-c3-2mb
  platformio_options:
    board_build.flash_mode: dio  (tried adding these settings but they don't seem to apply)
    board_build.flash_size: 2MB  (tried adding these settings but they don't seem to apply)
    
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

logger:
 deassert_rts_dtr: true
```

### Anything in the logs that might be useful for us?

```txt
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:QIO, clock div:1
load:1z3fcd6100,len:0x19a0
ets_loader.c 78
```

warthog9, 2022-07-02:
Still running into this on 53139c293ba77c457484f8f08ae94b4323d0228f, also tried setting the spi_flash size and that didn't work

waicool20, 2022-09-09:
I have this issue as well, I think its because of the default partition tables/bootloader, if I flash the blufi example project from esp-idf repo then flash a legacy image of esphome onto the 0x10000 address, it will boot fine but OTAs wont work obviously

nabilmerk, 2022-10-16:
Hi,

I have the same ESP-C3-32S (2Mb Variant) board and issue. I managed to boot the board with ESPHome using this information: https://community.home-assistant.io/t/esp32-c3-boot-loop-2mb-flash/466311

The reason behing the issue is the size of the binary file which doesn't fit in the 2MB of external flash. I'm not sure that solves the actual issue but it gives some workaround to play with ESPHome on these boards.