esphome/issues #3278: ESP32-C3 DSMR support

mhendriks, 2022-05-13:
### The problem

The ESP32-C3 and dsmr feature gives problems with the latest ESPHome version and underlying packages. "This board doesn't support arduino framework!" is the message.

With alternative packages android support is available but the Crypto header file hwcrypto/aes.h is not available. Adding this header file in the Crypto folder is a solution. This solution is good for local build but causes rebuild/update issues.

Is this a known issue...or is dsmr not yet supported for esp32c3?

### Which version of ESPHome has the issue?

2022.5.0b2 but 4.x too

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

esp32-c3-devkitm-1

### Component causing the issue

dsmr

### Example YAML snippet

```yaml
esphome:
  name: esp32c3-tst    
  platformio_options:
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.3
    platform:
      - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
    
esp32:
  board: esp32-c3-devkitm-1
  framework: 
    type: arduino
    
status_led:
 pin:
   number: 7
   inverted: true
    
api:

ota:

wifi:
# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "esp32c3"
    ap_timeout: 15s

captive_portal:

web_server:
  port: 80
  
# Enable logging
logger:
  level: DEBUG
  baud_rate: 0

uart:
  rx_pin:
    number: 10
    inverted: true
  baud_rate: 115200
  rx_buffer_size: 1700
  
dsmr:
 id: dsmr_instance
 receive_timeout: 200ms
 max_telegram_length: 1700
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration esp32c3-test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32c3-tst (board: esp32-c3-devkitm-1; framework: arduino; platform: 
https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream)
------------------------------------------------------------------------------------------------------------------------------------------------
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2-patch3
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2-patch3
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2-patch3
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2-patch3
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2-patch3
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2-patch3
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-arduinoespressif32 2.0.3+sha.142fceb 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 2.0.0
|-- <FS> 2.0.0
|-- <Update> 2.0.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 2.0.0
|-- <ESPmDNS> 2.0.0
|-- <ArduinoJson> 6.18.5
|-- <Dsmr> 0.5.0
|-- <Crypto> 0.2.0
Compiling .pioenvs/esp32c3-tst/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/esp32c3-tst/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/esp32c3-tst/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/esp32c3-tst/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/esp32c3-tst/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/esp32c3-tst/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/esp32c3-tst/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/esp32c3-tst/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/esp32c3-tst/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/esp32c3-tst/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/esp32c3-tst/src/esphome/components/dsmr/dsmr.cpp.o
Compiling .pioenvs/esp32c3-tst/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/esp32c3-tst/src/esphome/components/esp32/gpio_arduino.cpp.o
Compiling .pioenvs/esp32c3-tst/src/esphome/components/esp32/gpio_idf.cpp.o
Compiling .pioenvs/esp32c3-tst/src/esphome/components/esp32/preferences.cpp.o
In file included from src/esphome/components/dsmr/dsmr.cpp:6:
.piolibdeps/esp32c3-tst/Crypto/AES.h:200:10: fatal error: hwcrypto/aes.h: No such file or directory
 #include "hwcrypto/aes.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
```

glmnet, 2022-05-14:
I have a branch in my fork of ESPHome you can import it with external components and it will work

bakermat, 2022-06-11:
@mhendriks did you find a solution for this?
@glmnet: I tried adding this, assuming that's the correct branch? I get an error when compiling:
```
external_components:
  - source: github://glmnet/esphome@new-dsmr
    components: [dsmr]
```

Error:
```
INFO Reading configuration /config/esphome/bla.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing xxx (board: esp32-c3-devkitm-1; framework: arduino; platform: espressif32@4.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 2.0.0
|-- <FS> 2.0.0
|-- <Update> 2.0.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 2.0.0
|-- <ESPmDNS> 2.0.0
|-- <ArduinoJson> 6.18.5
Compiling /data/xxx/.pioenvs/xxx/src/esphome/components/api/api_connection.cpp.o
Compiling /data/xxx/.pioenvs/xxx/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/xxx/.pioenvs/xxx/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/xxx/.pioenvs/xxx/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/xxx/.pioenvs/xxx/src/esphome/components/api/api_server.cpp.o
Compiling /data/xxx/.pioenvs/xxx/src/esphome/components/api/list_entities.cpp.o
Compiling /data/xxx/.pioenvs/xxx/src/esphome/components/api/proto.cpp.o
Compiling /data/xxx/.pioenvs/xxx/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/xxx/.pioenvs/xxx/src/esphome/components/api/user_services.cpp.o
Compiling /data/xxx/.pioenvs/xxx/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling /data/xxx/.pioenvs/xxx/src/esphome/components/dashboard_import/dashboard_import.cpp.o
Compiling /data/xxx/.pioenvs/xxx/src/esphome/components/dsmr/AES128.cpp.o
In file included from src/esphome/components/dsmr/AES128.cpp:23:
src/esphome/components/dsmr/AES.h:100:3: error: 'esp_aes_context' does not name a type
   esp_aes_context ctx;
   ^~~~~~~~~~~~~~~
*** [/data/xxx/.pioenvs/xxx/src/esphome/components/dsmr/AES128.cpp.o] Error 1
```

mhendriks, 2022-06-12:
@bakermat yes, the external components trick did his job.
problem is the outdated Crypto version in esphome/dsmr/__init__.py

use
```
external_components:
  - source: github://mhendriks/esphome-p1@main
    components: [dsmr]
```

Total config
```
substitutions:
  device_name: p1-dongle-pro
  device_description: "P1 Pro (ESP32) module to read your smart meter"
  friendly_name: P1 Dongle Pro

external_components:
  - source: github://mhendriks/esphome-p1@main
    components: [dsmr]

esphome:
  name: ${device_name}
  comment: "${device_description}" 
  name_add_mac_suffix: false
  project: 
    name: "smartstuff.p1_dongle_pro"
    version: "0.9.6"
  platformio_options:
    upload_speed: 921600
    platform: espressif32@4.3.0
    
esp32:
  board: esp32-c3-devkitm-1
  framework: 
    type: arduino
    version: 2.0.3
    
status_led:
  pin:
    number: 7
    inverted: true

api:

ota:

wifi:

# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}
    ap_timeout: 15s

dashboard_import:
  package_import_url: github://mhendriks/esphome-p1/p1-dongle-pro.yaml@main
  
captive_portal:

web_server:
  port: 80
  
# Enable logging
logger:
  level: DEBUG

uart:
  rx_pin:
    number: 10
    inverted: true
  baud_rate: 115200
  rx_buffer_size: 1700
  
dsmr:
 id: dsmr_instance
 receive_timeout: 200ms
 max_telegram_length: 1700

  
sensor:
  - platform: dsmr
#    energy_delivered_lux:
#      name: "Energy Consumed Luxembourg"
#      state_class: total_increasing
    energy_delivered_tariff1:
      name: "Energy Consumed Tariff 1"
      state_class: total_increasing
    energy_delivered_tariff2:
      name: "Energy Consumed Tariff 2"
      state_class: total_increasing
#    energy_returned_lux:
#      name: "Energy Produced Luxembourg"
#      state_class: total_increasing
    energy_returned_tariff1:
      name: "Energy Produced Tariff 1"
      state_class: total_increasing
    energy_returned_tariff2:
      name: "Energy Produced Tariff 2"
      state_class: total_increasing
    power_delivered:
      name: "Power Consumed"
      unit_of_measurement: "W"
      state_class: "measurement"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_returned:
      name: "Power Produced"
      unit_of_measurement: "W"
      state_class: "measurement"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    electricity_failures:
      name: "Electricity Failures"
      icon: mdi:alert
    electricity_long_failures:
      name: "Long Electricity Failures"
      icon: mdi:alert
    voltage_l1:
      name: "Voltage Phase 1"
    voltage_l2:
      name: "Voltage Phase 2"
    voltage_l3:
      name: "Voltage Phase 3"
    current_l1:
      name: "Current Phase 1"
    current_l2:
      name: "Current Phase 2"
    current_l3:
      name: "Current Phase 3"
    power_delivered_l1:
      name: "Power Consumed Phase 1"
      unit_of_measurement: "W"
      state_class: "measurement"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_delivered_l2:
      name: "Power Consumed Phase 2"
      unit_of_measurement: "W"
      state_class: "measurement"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_delivered_l3:
      name: "Power Consumed Phase 3"
      unit_of_measurement: "W"
      state_class: "measurement"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_returned_l1:
      name: "Power Produced Phase 1"
      unit_of_measurement: "W"
      state_class: "measurement"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_returned_l2:
      name: "Power Produced Phase 2"
      unit_of_measurement: "W"
      state_class: "measurement"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    power_returned_l3:
      name: "Power Produced Phase 3"
      unit_of_measurement: "W"
      state_class: "measurement"
      accuracy_decimals: 0
      filters:
        - multiply: 1000
    gas_delivered:
      name: "Gas Consumed"
      state_class: total_increasing
    gas_delivered_be:
      name: "Gas Consumed Belgium"
      state_class: total_increasing
  - platform: uptime
    name: "Uptime"
  - platform: wifi_signal
    name: "Wi-Fi Signal"
    update_interval: 60s
      
text_sensor:
  - platform: dsmr
    identification:
      name: "DSMR Identification"
    p1_version:
      name: "DSMR Version"
    p1_version_be:
      name: "DSMR Version Belgium"
  - platform: wifi_info
    ip_address:
      name: "IP Address"
    ssid:
      name: "Wi-Fi SSID"
    bssid:
      name: "Wi-Fi BSSID"
  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true
```

bakermat, 2022-06-12:
Thanks, that did the trick!

mhendriks, 2022-10-11:
Issue is solved with the latest ESPHome version.

closed 2022-10-11 by mhendriks