esphome/issues #3280: Unexpected error with relay on newer firmwares

puerto-rico, 2022-05-14:
### The problem

I flashed a new firmware via the update function to discover some odd behavior on the esp. after reboot

it has been running flawless with a 2x relay attached for more than a year. And that relay does not turn on a boot. default is off.
But something happened with one of the latest firmwares, not sure if is 3.2.0 but it worked 2 months ago.

What happens is that when the esp reboot's or get's powered on the relay turns on for a fraction of a sec. and then the relay closes again. this has never happened before. it is running: espressif8266 @ 3.2.0

### Which version of ESPHome has the issue?

ESPHome version 2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 2022.3.8

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

Firmware

### Example YAML snippet

```yaml
esphome:
  name: hormann

esp8266:
  board: nodemcuv2

#Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "hidden"

wifi:
  ssid: hidden
  password: hidden

switch:
  - platform: gpio
    pin: 14
    inverted: True
    id: realy
    name: "Hormann Open"
    icon: "mdi:garage-open"
    on_turn_on:
    - delay: 500ms
    - switch.turn_off: realy
  - platform: gpio
    pin: 12
    inverted: True
    id: closed
    name: "Hormann Closed"
    icon: "mdi:garage"
    on_turn_on:
    - delay: 500ms
    - switch.turn_off: closed
```

pmannk, 2022-05-31:
This is actually a GPIO behaviour of the ESP8266 during boot. It's covered in the data sheets, but they're a really tough read.
The following article covers it in a much more digestable format based on probing with a scope: https://rabbithole.wwwdotorg.org/2017/03/28/esp8266-gpio.html
If you can switch from GPIO12&14 to GPIO4&5 then you should be fine. I had this problem about 2 years ago when I first setup my home-brew garage door opener. About 20% of reboots would be _just_ enough to trigger the relays. After switching GPIOs I haven't had a single false trigger.

If you really care about maximum security then you might want to consider other hardware interlocks too.

bkaufx, 2022-06-02:
You have to have the following in your config:

```
esp8266:
  early_pin_init: false
```

You may need to update your ESPHome for that to work.  Also, you may need to reinstall the addon from a different repository in order to upgrade.  This is the addon repo you need to add in homeassistant: https://github.com/esphome/home-assistant-addon

pmannk, 2022-06-02:
>   early_pin_init: false

I'd be careful with this. It might work around the issue but it's an exercise in luck (ie, the relay board is too slow to respond to the brief pull-up that occurs at reset).

The espressif pin documentation is very specific ( https://www.espressif.com/en/support/documents/technical-documents?keys=ESP8266+Pin+List ). GPIO12-15 are all subject to a weak-pull up _at reset_. GPIO4 & 5 (and some others) are not.

puerto-rico, 2022-06-07:
The funny thing here is that i already thought i had done this (used GPIO that are low to start with.)
But i must have got it mixed up. But is has never been a problem before, the changed behavior started  after an update.

i found out when changed the pins to GPIO4 and GPIO5 in the code, (without moving the wires from 12 and 14)
it stopped tricking the relay at boot, but when i moved  the to wires to GPIO4(d2) and GPIO5(d1)  it started happening again.
like it was in software. The board did not tricker the relay when the GPIO was not used in the code.

i got i fixed with   " early_pin_init: false " thank you, @bkaufx 

so the conclusion must be that something changed and now i have to use  "early_pin_init: false "  :)

bkaufx, 2022-06-07:
Yea, the early pin init functionality caused this problem for you.  Here's the original issue that the early pin init functionality was supposed to fix:
https://github.com/esphome/issues/issues/3001

And the PR that added the early pin init functionality:
https://github.com/esphome/esphome/pull/3144

And here is the issue where someone pointed out the problem with relays toggling.  This issue resulted in the early_pin_init option so the early pin init functionality could be disabled:
https://github.com/esphome/issues/issues/3263