esphome/issues #3281: PN532 over I2C stops responding after a while

chrispernold, 2022-05-14:
### The problem

I had this RC522 module for a while in a jukebox for my kids to play different MP3s but was always leaning towards trying out the PN532 module as it seemed more integrated to the ESPHome landscape. Before moving over I used a version of the https://github.com/HausnerR/esphome-door-lock code to get it working similar to `on_tag `and `on_tag_removed`.

**Fast forwarding a few years and I've replaced the RC522 with a PN532 module, hoping I could stick to the default PN532 code in ESPHome and remove the custom scripts. While it works immediately after a boot of the NodeMCUV2 (over I2C) it stops responding after a while, let's say a few hours. Only a hard reset (unplugging the power cable) will get it back to responding to RFID cards/tags.**

The setup looks like the following: 

ESP device with RFID PN532 module reads a RFID card/tag e.g. 12-34-56-78 and send this information to a Node Red flow that triggers an MP3 to be played on one of my Sonos speakers. 

If the RFID card/tag is removed it sends a 00-00-00-00 state that in turn pause the Sonos.

### Which version of ESPHome has the issue?

v2022.4.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.3.5

### What platform are you using?

ESP8266

### Board

NODEMCU V2

### Component causing the issue

PN532 over I2C

### Example YAML snippet

```yaml
esphome:
  name: node2-jukebox
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "XXXXXXX"
  password: "XXXXXXX"
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: XXXXXXX
    # Set this to the IP address of the router. Often ends with .1
    gateway: XXXXXXX
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

# Enable logging
logger:
  level: debug
  
# Enable OTA updates
ota:
  password: "XXXXXXX"

# Enable Home Assistant API
api:
  password: "XXXXXXX"

switch:
  - platform: restart
    name: "node2 jukebox restart"

i2c:
  sda: D4
  scl: D3
  scan: true
  #id: pn532_board
  #frequency: 400kHz

pn532_i2c: # or rc522_i2c
  id: pn532_board
  update_interval: 1s
  on_tag:
    then:
    - text_sensor.template.publish:
        id: rfid_tag
        state: !lambda |
          return x;
    - delay: 0.2s #to fix slow component
  on_tag_removed:
    then:
    - text_sensor.template.publish:
        id: rfid_tag
        state: !lambda 'return "00-00-00-00";'

text_sensor:
  - platform: template
    name: "RFID Tag"
    id: rfid_tag
  - platform: template
    name: "RFID Tag removed"
    id: rfid_tag_removed
```

### Anything in the logs that might be useful for us?

```txt
It doesn't show any errors in debug mode and in very_verbose it shows amongst other errors the following 

Timed out waiting for readiness from PN532!

Component pn532 took a long time for an operation (0.17 s). <- this varies
```

### Additional information

Someone at the [HA community](https://community.home-assistant.io/t/no-tag-scnned-esp32-with-pn532-testes-oder-spi-and-i2c/348601/3) suggested the D1 mini board instead of the NodeMCU. So I'm thinking this could be a HW error where the signal gets out of sync somehow and the board is unable to get it back to sync. What do you think?

chrispernold, 2022-05-15:
I changed to a D1 mini and it seems to work now. What I also did was to make sure I used the [recommended GPIO](https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/)s for the board, which I will strictly follow from here on out.

closed 2022-05-15 by chrispernold

VMmatty, 2022-09-01:
I know this issue was closed but @chrispernold I am having the exact same problem.  Can you share your updated yaml maybe?  I'm using a D1 mini connected via i2c and the scanner works for a few hours and then stops accepting scans.

I'm open to suggestions.  Thanks!

0x90ChuckTessela, 2022-11-19:
I've found the PN532 stops working every other soft reboot.  It works when completely unpowered for a moment.  However, safe reboots/restarts cause it to fail every other reboot.  Not sure if this is a software issue or a hardware issue.

In all events, the device enumerates as an I2C device, it just never seems to produce any output until the next soft reboot or power cycle.

ssieb, 2022-11-20:
Watch the serial logs to see what the difference is between working and not working.

0x90ChuckTessela, 2022-11-21:
so the (relevant) logs for working are:

```
[C][pn532:018]: Setting up PN532...
[VV][i2c.arduino:140]: 0x24 TX 0000FF02FED4022A00
[V][pn532:246]: Reading ACK...
[VV][i2c.arduino:116]: 0x24 RX 01
[VV][i2c.arduino:116]: 0x24 RX 010000FF00FF00
[V][pn532:257]: ACK valid: YES
[V][pn532_i2c:043]: Reading response
[VV][i2c.arduino:116]: 0x24 RX 01
[VV][i2c.arduino:116]: 0x24 RX 010000FF06FAD5
[V][pn532:262]: Sending NACK for retransmit
[VV][i2c.arduino:140]: 0x24 TX 0000FFFF0000
[V][pn532_i2c:049]: Reading response of length 5
[VV][i2c.arduino:116]: 0x24 RX 01
[VV][i2c.arduino:116]: 0x24 RX 010000FF06FAD50332010607E800
[D][pn532:033]: Found chip PN532
[D][pn532:034]: Firmware ver. 1.6
[VV][i2c.arduino:140]: 0x24 TX 0000FF05FBD4140114010200
```

whereas, performing a soft-reset (restart the device via esphome's restart platform):
```
[C][pn532:018]: Setting up PN532...
[VV][i2c.arduino:140]: 0x24 TX 0000FF02FED4022A00
[V][pn532:246]: Reading ACK...
[VV][i2c.arduino:116]: 0x24 RX 01
[VV][i2c.arduino:116]: 0x24 RX 010000FF02FED5
[V][pn532:257]: ACK valid: NO
[E][pn532:022]: Error sending version command
[E][component:113]: Component pn532 was marked as failed.
```

performing a second soft-reset (same manner as the first) results in success, with logs the same as a power-on:

it appears that the unit itself is still responding to a previous configuration command from before the power-on.  I've tested this on a board with nothing else but that PN532 hooked up.  I'll keep looking into it, but anyone else have any ideas?

mrkylegp, 2022-12-31:
I have the same issues with the PN532 modules. I think based on everyone experiencing the same issue here that there is a problem with the module on a hardware/firmware level.