esphome/issues #3282: restore_from_flash causes boot loops after OTA

platima, 2022-05-14:
### The problem

When I set `restore_from_flash: true` then flash via OTA I get some very odd behavior. Basically it'll flash, then it reboots, and then randomly reboots every 1-10s. It joins the wifi for a split second, but never long enough to flash again.

If I then flash with serial, it works fine, until the next OTA.

If I flash serial with `restore_from_flash: true`, then while it's running flash with `restore_from_flash: false` via OTA, it works fine, so it appears to be something specific to `restore_from_flash: true` _after_ an OTA flash.

Happy to grab any logs required!

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.4

### What platform are you using?

ESP8266

### Board

esp8285

### Component causing the issue

board

### Example YAML snippet

```yaml
esphome:
  name: csh-plgusbpm

esp8266:
  board: esp8285
  restore_from_flash: true
  
preferences:
  flash_write_interval: 1min
```

jesserockz, 2022-05-15:
Can you please grab serial logs from when it is rebooting?

Maybe even enable `logger` `level: VERY_VERBOSE` too

platima, 2022-05-23:
Okay so I cannot tell if this is related to #3284 or not. 

I am struggling to get serial logs due to the nature of the device(s), plus have not had time this week.

I did also flash some smart lights (ESP8266ex chips) and they worked fine, then I enabled restore from flash, and now it's basically bricked. The only other change I made in the code during that flash was to set restore mode always_on.

It will now boot randomly, and join the wifi for a few seconds, before crashing. I tried captive portal by disabling my IoT SSID, but even then it only shows up in safe mode for a few seconds. Flicking it on and off 12 times to trigger OTA safe mode also did not seem to do anything.

I actually flashed 3 of them, and all 3 bugged out like this, however, two I managed to get back by just hitting 'retry' over and over and eventually catching them online for long enough. This took two days though. 

To try to rule out a thermal issue I left it off for 12 hours. In the morn same result. Right now I have a loop running in terminal
`while [ $? -gt 0 ]; do /usr/bin/python3 esphome upload --device 10.12.20.25 /config/esphome/e14-03.yaml; done` and occasionally it catches it online but never gets past ~5% upload. I stripped pretty much everything out the firmware too just to minimise upload size.

platima, 2022-05-23:
Just found two more bulbs, different brand but same issue; these are GU10's also running ESP8266ex, and after enabling restore_from_flash they kept rebooting. I managed to re-flash all 3 of them again using lots of retries, however, one is now fine, the other two won't even try to join wifi, don't go into fallback AP mode, etc. Seem completely bricked.

FYI these are the three products so far;
- https://templates.blakadder.com/connect_smarthome_CSH-E14WW10W.html (2 of 3 I recovered)
- https://templates.blakadder.com/kogan-KASPEMHUSBA.html (I actually have the Connect branded one, but exact same)
- https://templates.blakadder.com/connect_smarthome_CSH-GU10WW5W.html (2 of 2 bricked)
- https://templates.blakadder.com/laser_LSH-E14RGB5W.html (1 of 1 recovered)

All were older gen Tuya that I converted and worked fine for a while until I made that change.

platima, 2022-05-24:
Okay so this is interesting. This device is a light bulb, the other was a smart switch. I noticed that pretty much every power cycle it would join wifi, but if I tried to flash it, it'd time out after 3-4% uploaded.

Having cracked both open, both boot and run fine when on 3.3V via FTDI. I connected to this CSH-E14WW10W that I could not recover via the UART pins and it seemed perfectly stable. That being said, when I tried flashing it it crashed at 74% uploaded when doing OTA. I did it again and monitored serial this time. ESPHome web UI showed 11% uploaded, this is all I got on serial;
```
20:32:27.366 -> [0;36m[D][ota:139]: Starting OTA Update from 10.12.40.80...[0m
20:32:27.451 -> [0;36m[D][ota:308]: OTA in progress: 0.3%[0m
20:32:36.271 -> 
20:32:36.271 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,2)
20:32:36.271 -> 
20:32:36.271 -> wdt reset
```
Power cycling it and trying again, pretty much exact same; UART shows no messages for boot (I forget what logger mode I had), and when I flashed it it crashed.

To flash over serial with esphome-flasher, even though the device would join wifi and come online, I had to pull GPIO0 low to put it in programming mode. Doing it this way with esphome-flasher also does a full flash erase, so I was confident any garbage data was cleared. But, of course it crashed after detecting the chip and saying 'Erasing flash';
```
Using 'COM7' as serial port.
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Connecting...

Chip Info:
 - Chip Family: ESP8266
 - Chip Model: ESP8266EX
 - Chip ID: 000708B8
 - MAC Address: D8:F1:5B:07:08:B8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 1MB
 - Flash Mode: dout
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Unexpected error: Error while erasing flash: Packet content transfer stopped (received 8 bytes)
```
I did this twice, and it timed out both times after 8 bytes. I also found it interesting that it was using 460800 baud when it had a 26Mhz oscillator onboard, as it should in theory be 74880. But perhaps that's just a programming mode fault that I am unaware of.

So, after two failed flashes, I power cycled it again, and it's not joining the wifi so clearly something was wiped a bit. Tapping the RST to ground gives me
```
21:05:05.045 -> 
21:05:05.045 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
21:05:05.045 -> 
21:05:05.045 -> ets_main.c 
```
And I realise I should have done this before when it still had crappo firmware on it, but ... vodka. I don't understand the boot modes but I am hoping it helps you wizards decode what is going on.

Out of dumb lack of ideas, I pulled 0 low again, rebooted, and told ESPHome web UI to flash via USB.... And it did. I honestly have no idea why this worked but perhaps because it does not try to clear flash, which would align with it crashing/looping when we save state to flash using restore_from_flash.

As this may board dependent, not just chip (FYI I am using `board: esp01_1m` in my YAML`) here is a pic of it for you; https://imgur.com/2UzZ7mO. If you need any chip close ups let me know and I'll bust out the microscope.

Cheers.

closed 2022-10-07 by github-actions[bot]