esphome/issues #3283: ERROR Error resolving IP address of athom-smart-plug-v2.local. Is it connected to WiFi?

nickh66, 2022-05-15:
### The problem

Hi, I believe this to be a bug, so lets see what the team thinks.  But first some context.

I purchased esphome pre-flashed athom.tech smart plug v2 & set it up out of the box successfully & was connected to my wifl.  Also the plug was discovered & configured successfully & then working fine in home-assistant.  I did also find a new device in the esphome addon console that allows to 'adopt' the plug.  I thought this was a good idea so i could update the firmware in the future if i wanted or needed.

The problem comes when trying to adopt the plug.  Firstly, when doing this a new esphome device is automatically created during the process & when clicking the install button on the wizard, then choosing the wireless install option i eventually see this error at the end of the build/compile process.

I think this particlar error about "resolving ip address" is because it's perhaps hardcoded for the incorrect plug hostname that it was born with out of the box & also because my local dns suffix is not .local but instead .localdomain.  In the case of this particlar plug the actual device hostname is "athom-smart-plug-v2-xxxxxx.localdomain" (x=mac of device).  So the connection for upload couldn't be made.  Looks like there is an assumed device name that does not match the actual hostname & maybe hardcoded to .local dns suffix instead of utilising the one provided by my dhcp server.

I did then follow the instructions in the esphome guide that is included in the error output.  That did seem to allow the firmware to be uploaded to the plug but once the upload was complete the plug would not reconnect to my wifi & I also lost local connectivity to the plug.

Cheers

====== [SUCCESS] Took 210.23 seconds ========================
INFO Successfully compiled program.
INFO Resolving IP address of athom-smart-plug-v2.local
ERROR Error resolving IP address of athom-smart-plug-v2.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.4

### What platform are you using?

ESP8266

### Board

Athom PG05V2-AU16A

### Example YAML snippet

```yaml
Sorry, i don't have this any more.
```

jesserockz, 2022-05-15:
First thing I want to point out is `athom-smart-plug-v2-xxxxxx.local` is correct, (not `.localdomain`) as ESPHome utilizes mDNS **not** DNS. If your machine or network blocks mDNS/multicast traffic then that could explain why it is not working out of the box.

> I did then follow the instructions in the esphome guide that is included in the error output. That did seem to allow the firmware to be uploaded to the plug but once the upload was complete the plug would not reconnect to my wifi & I also lost local connectivity to the plug.

Could you explain what you actually did here?

nickh66, 2022-05-15:
Hi,

Good point about mDNS.  So i accept it's not about my chosen domain suffix.  My mDNS is definately working though because without it home-assistant (on a separate LAN for my wired devices isn't able to find my google home devices that live on a different subnet with all my other iot devices, like these smart plugs.

So, In this instance i think the issue i experienced might be because the hostname used for the connection after the compile completed was not the same hostname the actual device was running with.  The device hostname was "athom-smart-plug-v2-xxxxxx" while esphome was attempting to "athom-smart-plug-v2".

That's the only thing i thought might have been a bug.

The fact the device is now unreachable & is totally on me.  I'm pretty sure i made a mistake somewhere in the changes i made to the config yaml that rendered the device lost.  Honestly, i don't recall exactly what changes i made that broke things but the only changes i made were around setting an static ip so perhaps i messed up somewhere in the wifi: section.  It's possible i accidentally added both the ap: and captive_portal: sections. In any case i'm sure i stuffed it somewhere.

Just to be clear. My mistake that lost me my connection to the device is not related to what i thought might be a problem with hostnames.

Cheers
Nick

jesserockz, 2022-05-15:
When adopting a device, the device advertises its full hostname, including the mac address for adoption. Unless you changed the yaml file and removed the mac suffix from the `name` then it would have tried to connect to the full name.

Without the device back in that state though we will never know now...haha

As for your second issue:

`ap` and `captive_portal` are supposed to work together so that you have a fallback if the device cannot connect to your wifi.
But if you have set a static ip with an incorrect ip, or subnet etc then it will connect to wifi, but you will have a hard time connecting to it.
If it is not even connecting to your wifi then `ap` should come up. 
One way to test that is to take it away from the house, (or turn off the wifi) so the `ap` turns on and you can connect to that and upload a new firmware.

nickh66, 2022-05-16:
Thanks Jesse, Thats a great tip to try at another location away from my home wifi.

tempus2016, 2022-06-28:
@nickh66 did you fix this problem? I too have the same issue and I can't connect to the v2 plug.  The AP shows, I'm connected it it but it won't show me any page to update the original bin file.

closed 2022-11-11 by github-actions[bot]