esphome/issues #3284: esp8285 reboot due to incorrect API or wireless timeout

platima, 2022-05-15:
### The problem

I have a pile of Espressif MCU's connected to ESPHome within HA. Everything latest.

I've found that the ESP8285 crashes every 15-30 mins, and I narrowed that down to either a Wifi or API timeout. Setting both to 0s works fine.

The YAML I am using now is 
```
esphome:
  name: csh-plgusbpm
  comment: Labelled Plug01

esp8266:
  board: esp8285

logger:

api:
  reboot_timeout: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  reboot_timeout: 0s
```

However it'd be great to actually not have to set these, as they are not actually timing out. My AP does not show it disconnecting until after the reboot, and when I watch the logs everything is working fine until the API says read error on the 15 or 30 min mark, then it reconnects a few seconds later. So it seems like the timeout is not being counted correctly.

After the unexpected reboot (I can recreate and get verbose logs if required), the status LED is blinking every ~1 second and the emergency wifi AP is enabled, even though it's also joined again in STA mode.

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.4

### What platform are you using?

ESP8266

### Board

TYWE2S (ESP8285)

### Component causing the issue

api and wifi

### Example YAML snippet

```yaml
esphome:
  name: csh-plgusbpm
  comment: Labelled Plug01

esp8266:
  board: esp8285

logger:

api:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
```

### Additional information

I can grab any debug logs that you may need, just let me know what logging level to use.

platima, 2022-05-23:
Okay so I cannot tell if this is related to #3282 or not.

ssieb, 2022-09-21:
Serial logs would be best, but OTA logs would probably work too.  And the normal DEBUG level is fine.