esphome/issues #3285: ST7735 driver not working correctly with ST7735S

Scoff123, 2022-05-15:
### The problem

I have a Waveshare 0.96" 80x160 TFT SPI Display which uses the ST7735S chip. The screen protector film has a Blue tab. This display is connected to my Lolin D1 Mini Pro ESP8266 board.
Configuring my ESPHome code to use the ST7735 platform doesn't work correctly with my ST7735S Blue tab.
There is no Blue tab option to specify in the code, so I am using the model INITR_MINI160X80
Using default settings, the resulting display does not fill the screen, and there are borders of undriven screen.
To drive these borders and fill the screen, I have to set the width as 82, and height as 161. 

Also, the RGB colours are either inverted or not mapped correctly. For example, if I try to fill the screen with RED only, I define:
color
- id: my_red
  red: 100%
  green: 0%
  blue: 0%
and create a display lambda
lambda: |-
      it.fill(id(my_red));
and the result is a yellow screen. 

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.4

### What platform are you using?

ESP8266

### Board

Lolin D1 Mini Pro

### Component causing the issue

Display/ST7735 Platform

### Example YAML snippet

```yaml
display:
  - platform: st7735
    model: "INITR_MINI160X80"
    reset_pin: GPIO16
    cs_pin: GPIO15
    dc_pin: GPIO0
    rotation: 90
    device_width: 82
    device_height: 161
    col_start: 0
    row_start: 0
    eight_bit_color: true
    update_interval: 1s
```

jesserockz, 2022-05-15:
This is probably more of a feature-request than a bug to be honest.
It will require someone adding support for that display.

Scoff123, 2022-05-15:
Thanks, yeah I was a little unsure whether to raise it as a feature request, or whether the support was supposed to already be there (within the ST7735 Platform) and just wasn't working correctly. I'll close this and open as a feature request instead.

Scoff123, 2022-05-16:
Raised as feature request here instead: https://github.com/esphome/feature-requests/issues/1730

closed 2022-05-16 by Scoff123