esphome/issues #3286: BedJet - can't set fan mode using automation/template

user897943, 2022-05-15:
### The problem

I have this code:

```
service: climate.set_fan_mode
data:
  fan_mode:  {{ ((states('input_number.bedjet_speed')|float) / 1)|int }}%
target:
  entity_id: climate.bedjet_front_bedroom_1
```
I also have a input_number named input_number.bedjet_speed helper that has a minimum 5, max 100, step 5. I can adjust the slider and then use the service climate.set_fan_mode to read the number on the slider and divide by 1 to get an integer (otherwise if I set the slider to 45 it will show as 45.0), then add the % symbol to get 45% (code is self explanatory).

If I try to call the service, it fails with "Unknown error". I went to the states tab on developer tools and noticed that the values under fan modes are formatted as such:

```
fan_modes:   5%,  10%,  15%,  20%,  25%,  30%,  35%,  40%,  45%,  50%,  55%,  60%,  65%,  70%,  75%,  80%,  85%,  90%,  95%, 100%
```
It's very subtle but I noticed that there is a space character before each number, so when calling the service with value "35%" it fails, because the fan mode is actually formatted as " 35%" - notice the space before 35. If I add the space, the service sets the value (but weirdly the bedjet doesn't change the fan speed, nor does the value change on the states page for the entity).

This means: can't set the fan speed using automations as above. I have an automation to read the value from the input number, and then call the service climate.set_fan_mode which sets the fan speed to the same value as the input_number slider. Automations using climate.set_fan_mode service will fail because the values are not set on the bedjet and are also formatted with extra unnecessary space as attributes.

### Which version of ESPHome has the issue?

2022.6.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.5.4

### What platform are you using?

ESP32

### Board

wemos d1 esp32

### Component causing the issue

bedjet

jhansche, 2022-05-27:
This was discussed previously, and the reason for adding the space is due to HA frontend sorting the fan modes lexicographically (alphabetically). That means the modes were being sorted like:
* 1%
* 10%
* 15%
* 100%
* 20%
* 25%
* ...
* 45%
* 5%
* 50%
* 55%
* ... etc

During the initial component PR, it was suggested to use a leading 0, but that looked awkward as well. I chose the space because the UI seemed to swallow the extra space but still sorted it in the correct order. But I did not consider how that impacts using it from automations or services.

Your PR was already merged to remove the padding spaces, and I'm okay with that, so I think this can be closed. I may open a PR for HA frontend to have better sorting of these fields (this is usually a well known sorting strategy called "natural" sort order, which is still alphabetical but considers numeric values as numbers rather than strictly lexical).

user897943, 2022-05-27:
> This was discussed previously, and the reason for adding the space is due to HA frontend sorting the fan modes lexicographically (alphabetically). That means the modes were being sorted like:
>
> * 1% [...]

I appreciate that you're letting the changes remain, the automations have been running beautifully now, speeds are in sync with the remote etc. I have full control of the BedJet from HA. I think making the change to HA is the way to go, and will benefit other integrations that are affected. That's a great idea. Thank you for developing this, by the way. It turns a good product into a great one. With a bed occupancy sensor this thing can be set and forget. If there's one thing that's missing, it's showing the "current temperature" correctly, right now it doesn't match what is shown on the remote as "actual". Remote shows actual as 79F and the sensor on your integration on HA shows 71, a huge discrepancy, so it makes automating based on temperature very unreliable.

jhansche, 2022-05-27:
> Remote shows actual as 79F and the sensor on your integration on HA shows 71, a huge discrepancy, so it makes automating based on temperature very unreliable.

This was intentional. The reason is because the "Actual" temp on the remote is showing the air temp of what the fan is blowing out; in cool mode that's going to be the same as ambient temp (there is no active cooling in BedJet), and in heat mode, it's going to very closely track "Target" temp, because it will regulate the output air in order to match the target. So to me, that makes it not very useful.

Instead I use "Ambient" temp for current temperature, which makes it usable as another room temperature sensor, and that also more closely tracks to what another climate device (like an AC unit) would do. It's uncommon for a device to show the temp of the air it's putting out (i.e., an AC might show an output temp of 55° even though the target is 69 and ambient is 71 - it isn't all that helpful to know the output temp in this case), because what really matters is the ambient temp. The only downside is it makes the climate history graph look unusual because it might show a target temp of ~85°, with a current temp of ~71°, which kind of makes it look like it's failing to hit the requested target.

user897943, 2022-05-27:
> This was intentional. The reason is because the "Actual" temp on the remote is showing the air temp of what the fan is blowing out; in cool mode that's going to be the same as ambient temp (there is [...]

I understand that this is how the unit operates, glad you confirmed the concept. How do you get the "ambient" temp, is there an actual hardware sensor inside it for that?

I actually think it should mirror the BedJet. It's not an actual climate device (as it doesn't cool like AC does), it's a "comfort" device so I think it's logical to have the temps show the way it does on the remote as you wanna know what the air coming out is. Isn't there a way to make three sensors, one for setpoint, one for "actual" (and make that the "currently" temp on HA), AND a sensor showing room temp like you've got for those people who might want to use that for automation, then they still have the available value. I think the implementation should mirror what the manufacturer intended.

jhansche, 2022-05-27:
"Ambient" is built into the BedJet. It doesn't display on the remote, but it displays in the app.

I hear what you're saying re how the manufacturer intended, but I just don't agree that there's any reason for it. If you want to make a template sensor that shows the actual air temp blowing out of the device, you could make it choose based on current mode: when cooling -> choose `current_temperature`, when heating -> choose `max(target_temperature, current_temperature)`. That's really what it comes down to, and how the device will regulate itself anyway.

As for creating a 3rd sensor for the extra value, that isn't possible with a Climate component. The ESPHome Climate component is derived from the Home Assistant Climate entity (because that's where it has to be published): https://developers.home-assistant.io/docs/core/entity/climate/, with only these fields available to export (temp related):
```
  /// The current temperature of the climate device, as reported from the integration.
  float current_temperature{NAN};

  union {
    /// The target temperature of the climate device.
    float target_temperature;
    struct {
      /// The minimum target temperature of the climate device, for climate devices with split target temperature.
      float target_temperature_low;
      /// The maximum target temperature of the climate device, for climate devices with split target temperature.
      float target_temperature_high;
    };
  };
```

There just isn't another attribute that can be used for this, and ESPHome does not allow adding custom attributes.

The suggestion would require splitting the climate component into a controller that tracks BLE state, and dispatching the shared state to multiple unrelated entities. While it might get what you're after, it's really not worth the effort imo.

But I still don't understand your use case that would require showing the output air temp over ambient air temp? How is the "Actual" temp actionable for your use case?

Since adding the integration, I don't even use the app at all anymore, and I rarely use the remote either (just to quickly switch mode or bump the temp up or down). So I really don't care all that much about what's on the remote or app screen, or how that compares to what's in HA. Instead I care about what it looks like _in_ HA only, and what you're proposing would look strange, unhelpful, and teetering on downright incorrect.

Also, "Actual" temp is not valid when the fan is not actively blowing, while "Ambient" temp continues to emit correct data all throughout the day even while the unit is not active.  If we switched to actual temp, then the history chart would be mostly empty except while the unit is blowing. Why not make use of real actionable data?

> It's not an actual climate device (as it doesn't cool like AC does), it's a "comfort" device

Actually, according to their website, it's marketed as "BedJet 3 Climate Comfort Sleep System" 😄

user897943, 2022-05-27:
I see your points, I'm all for having options, that's all. The use case is
that I primarily use the remote and it's a little OCD having data display
inconsistently between the remote the the card on HA. If I'm using heat
mode then "actual" would let me make sure the air blowing out of the BJ is
actually what I set it to be. I personally have no use for ambient, but I
can totally see the benefit for automating at some point and don't want it
to go away. I simply wish there was a way to have the info on the HA card
show up like the same on the remote.

As for how they market it, well they have a ton of buzzwords but to me it's
a comfort device not a climate one, that's why you have a separate fan
component on HA, when "cooling" the BedJet is nothing more than a ducted
fan. To be a climate device in my opinion requires actual air conditioning
which it doesn't have.

I use the memory function a lot, and as you know you can set time until it
shuts down and it counts down, is it possible and are you willing to pull
this "current program countdown time" and turn into a sensor? That would be
useful, again, to basically have the same info as is on the remote. The
reason why I don't stop using the remote is exactly because the integration
does not provide all the data that the remote does.

Don't take that as a criticism! I absolutely love having the integration,
it's fantastic, but could be improved further.

jhansche, 2022-05-27:
> To be a climate device in my opinion requires actual air conditioning which it doesn't have.

This isn't true. A climate device implies being able to alter the climate in some way. You're right that a simple fan _does not_ do that, and hence has its own dumb device type. But a _heater_ does, and heaters do in fact appear in HA as climate devices. BedJet does have active heating. The lack of active cooling does not negate the heater.

> current program countdown time

This data is already available in the BedjetStatusPacket struct: https://esphome.io/api/structesphome_1_1bedjet_1_1_bedjet_status_packet.html#ab5f1b8cc0f745a5b09b3a3276bfe24e1
I had no use for it, and no place to put it as a Climate device, so it's just ignored. All we would need to do is expose the BedjetCodec object with a public function, and you'd be able to write a template sensor to copy those and other values.

I just don't think it's necessary, and it isn't possible to expose _anything else_ with the Climate device type.

I know @jesserockz has suggested other ways to work around this limitation, as mentioned above: splitting the BedjetCodec that is currently ingrained in the BedjetClimate object, and move it to its own controller object, and then add extra per-domain component implementations. One might be a climate device, one could be a sensor (for your Actual temp, program countdown timer, current firmware version, and a bunch of other things currently in BedjetStatusPacket that don't fit in the Climate component), could also add a fan component that allows controlling the fan speed as a slider rather than the awkward spinner dropdown.

I like the idea, I just don't know how to move in that direction, such that we share a single BLE subscription and status packet, and we're able to share that packet information to all other component implementations that need it. If anyone can point me to an example of how that is done with any other component (I haven't found one), I'm willing to consider it.

> Don't take that as a criticism ... but could be improved further.

I don't, and I'm happy with the way it has turned out. It suits my needs. The fact that it doesn't suit yours is also okay.

Maybe you would be better off switching to one of the other options out there. I know I've seen one circulating that uses Node-Red and the Home Assistant server as the BLE client, and a lovelace card to go with it that actually does look much more like the remote does. https://community.home-assistant.io/t/bedjet-integration/144822/6
I would not be offended if you prefer that one 🥲

user897943, 2022-05-27:
I'll stick to yours, thanks.

If you are only willing to provide functions that fit your needs, rather
than exposing all the functions for the community to use creatively, that's
totally your call, you own your time, but it is disappointing from my
perspective as your implementation works elegantly but it's a waste to have
all the data available and not expose it.

As for the climate/comfort debate, it's really not important to me how it's
referred as, just like cilantro/coriander is the same thing.

jhansche, 2022-05-27:
> The reason why I don't stop using the remote is exactly because the integration does not provide all the data that the remote does.

I understand that. But I would turn that around again, and ask why? When my BedJet is active, it's because I'm in bed (hopefully sleeping). I don't care how many minutes are left.

> If you are only willing to provide functions that fit your needs

That's not what I said. I said all it takes is making BedjetCodec public. Then you can add in your yaml file a lambda that can extract that data however you please, including stuffing it into a template sensor. 

I agree that it's a waste to have the data there and not use it. But I think you're misunderstanding the problem. The problem isn't that I don't want to expose it, it's that there **is no way** to expose it as an entity in an ESPHome Climate component. I couldn't add the actual temp or countdown timer to the climate component, if I did want to. ESPHome does not allow custom attributes.

Therefore the closest thing I can offer, is making the BedjetCodec object available publicly, thus allowing you to access it from a lambda.

jhansche, 2022-05-27:
The more I think about it, I think the automation callback/lambda is the ticket.
That's a "useful, compelling, actionable" use case 😉

user897943, 2022-05-27:
I am happy to explain my use case but I hope you're not asking questions as
a "gotcha". If I wake up and I am too tired still, I might wanna add an
extra hour to the program, so I look at the remote and check how much is
remaining. I never turn off the BedJet manually, I let it run for a bit
after I wake up to air the sheets, basically until the program is finished
at which point it turns itself off for safety and saving power. If I don't
have this info on the HA I have to keep using the remote which I'd rather
not, so yes it would be good to have the extra data.

Isn't it possible to have more than one component on a yaml? You're saying
it can be exposed as a lambda sensor, if you write documentation explaining
how, that would be amazing. Using a grid card could allow me to basically
replicate a virtual remote on my HA for example and finally put the
original remote in the drawer for good.

jhansche, 2022-05-28:
Nope, not a "gotcha". Like I said, I want a feature request to be actionable. I can't act on "because of my OCD". 

In the use case you're describing, you would wake up, open a HA dashboard, look at the data, and decide if you want to reset or extend the program?

I'm asking questions because it helps me understand the use case. I don't just build something because, I build it for a purpose. I need to know that purpose in order to design it correctly.

This would also require adding the TIME_UP/DN buttons as well, and a way to execute it from outside the component, which is not currently implemented (that button is not used or defined in the component).

So it sounds like you have a few feature requests to open (and this one can be closed as you already resolved it):

1. Automation lambda(s)
2. Expose button access for lambdas

user897943, 2022-05-28:
> Nope, not a "gotcha". Like I said, I want a feature request to be actionable. I can't act on "because of my OCD".
>
> In the use case you're describing, you would wake up, open a HA dashboard, look at [...]

That's basically it. I look at the time, increase or decrease as needed. I'll open the requests later, I'll reference this post for background info, thanks.

user897943, 2022-05-28:
Issue resolved, as code changes have been merged, closing issue.

closed 2022-05-28 by user897943

jhansche, 2022-05-29:
In the spirit of sharing use cases and highlighting the power of automations, this is how I integrate the bedjet with my sleep schedule.

I use a sleep tracking app on my phone. That app has mqtt support, and I set that up to feed sleep related events to HA. One of those events is a "time to get ready for bed soon" alert based on the next day's alarm time. Another is when sleep tracking is started, when the "smart wake period" begins, when the alarm starts, when I snooze the alarm, and when I dismiss the alarm.

I use the "ready for bed soon" event to start the bedjet, so that it's warm when I get in. When the smart wake period begins, it opens my bedroom blackout shades. When the alarm starts, it switches from ext-heat mode to dry mode as a kind of incentive to get up. When I dismiss the alarm, it turns off the bedjet. An automation could just as easily set a timer to let the dry mode keep running for a period of time after dismissing (or with a bed occupancy sensor, after actually getting up out of bed).

I don't say this to downplay your use case, just to highlight that smart home devices and automations can render obsolete some routines that we're used to doing manually. And when I said above I never use the bedjet app and hardly ever use the remote, the reality is I also hardly ever use the HA climate card either, because automations take care of basically everything I need.

Thanks for bearing with me.

I'll work on the automation lambda in the new ticket, and update the docs to explain how to use it.

user897943, 2022-05-29:
Thanks for sharing your use case, interesting and pretty much what I want to achieve except using bed occupancy. I have some 250 automations running on my HA and totally get what you. My aim is to minimize the amount of user input as much as possible but I will in this case control the timing because bed occupancy sensing isn't reliable enough for me yet. Need to explore better sensing options. The aim is to never have to touch anything.
I appreciate your work to enhance the already amazing stuff you've provided.

mwolter805, 2022-05-31:
Don’t want to put words in @user897943’s mouth but it sounds like he’s asking for the ESPHome bedjet integration to more closely follow the remote. As have I and has another in issue https://github.com/esphome/issues/issues/3314 where the heat map was discussed. Appears like the majority of users that are reporting issues or asking for FR’s are asking for this as well. Not trying to make a big deal about this but to dismiss it is “kicking the can down the road” and will most certainly lead to more issues and FR’s in the future. In other words, future users will be confused by the integration because it’s operation has been changed from the devices operation.  

I would argue that the Bedjet is not exactly a climate device, it’s a climate comfort device. Which is a super set of a typical climate device. As you’ve seen it has many more functions/sensors than a typical climate device. Fan speed, an auto-off-timer, special heat modes, output temp sensor, etc. Since ESPHome and HA, for that matter, does not have a “climate comfort” component or domain, climate is being used. It’s great that it mostly “fits”, but to argue it’s a climate device is not accurate. 

IMO it would be better and require less programming for the user (lambdas and templates) if these sensors and functions were exposed as separate sensors and components to ESPHome. Do not integrate them into the climate component but make them separate sensors and devices and only utilize lambdas if absolutely necessary. 

It would also be good if there was a map made, similar to the one made in https://github.com/esphome/esphome/pull/3519 but to allow the “current temp” of the climate component to be mapped to either “output temp” or “ambient temp” of the bedjet. As you mentioned ambient temp is not available on the remote but is separately available on the app. I don’t know about you all but other than the initial setup of the bedjet, I haven’t used the app and didn’t even know ambient was available and to suddenly see it on the ESPHome integration was confusing. This would allow you to “customize” the climate component to fit your needs, but the default operation should mimic the remote.

I’ll make a separate FR for all this as it sounds like the two FR’s above would be incorporated in this.

user897943, 2022-05-31:
> Don’t want to put words in @user897943’s mouth but it sounds like he’s asking for the ESPHome bedjet integration to more closely follow the remote. As have I and has another in issue #3314 where the heat map was discussed. Appears like the majority of users that are reporting issues or asking for FR’s are asking for this as well. Not trying to make a big deal about this but to dismiss it is “kicking the can down the road” and will most certainly lead to more issues and FR’s in the future. In other words, future users will be confused by the integration because it’s operation has been changed from the devices operation.
> 
> I would argue that the Bedjet is not exactly a climate device, it’s a climate comfort device. Which is a super set of a typical climate device. As you’ve seen it has many more functions/sensors than a typical climate device. Fan speed, an auto-off-timer, special heat modes, output temp sensor, etc. Since ESPHome and HA, for that matter, does not have a “climate comfort” component or domain, climate is being used. It’s great that it mostly “fits”, but to argue it’s a climate device is not accurate.
> 
> IMO it would be better and require less programming for the user (lambdas and templates) if these sensors and functions were exposed as separate sensors and components to ESPHome. Do not integrate them into the climate component but make them separate sensors and devices and only utilize lambdas if absolutely necessary.
> 
> It would also be good if there was a map made, similar to the one made in [esphome/esphome#3519](https://github.com/esphome/esphome/pull/3519) but to allow the “current temp” of the climate component to be mapped to either “output temp” or “ambient temp” of the bedjet. As you mentioned ambient temp is not available on the remote but is separately available on the app. I don’t know about you all but other than the initial setup of the bedjet, I haven’t used the app and didn’t even know ambient was available and to suddenly see it on the ESPHome integration was confusing. This would allow you to “customize” the climate component to fit your needs, but the default operation should mimic the remote.
> 
> I’ll make a separate FR for all this as it sounds like the two FR’s above would be incorporated in this.

You echo my sentiments pretty much.

jhansche, 2022-06-01:
I made some progress today on breaking apart the bedjet component, opening the door for additional sensors to be created. Still more work to do to get it back to stable; and more work yet on top of that, to realize the end results mentioned above.

user897943, 2022-06-01:
> I made some progress today on breaking apart the bedjet component, opening the door for additional sensors to be created. Still more work to do to get it back to stable; and more work yet on top of that, to realize the end results mentioned above.

Your effort is really appreciated, buddy. Thanks very much 🙏

kquinsland, 2022-06-01:
> * ... etc
>
>
> During the initial component PR, it was suggested to use a leading 0, but that looked awkward as well. I chose the space because the UI seemed to swallow the extra space but still sorted [...]

So this means that there's nothing that can (reasonably) be done on the ESPHome side of things to get the sort order fixed... short of marking the bedjet internal and re-implementing a template climate or a template select entity which respects the absolute order of user supplied options?

jhansche, 2022-06-02:
No there are definitely ways to fix this in HA, for sure. I think I mentioned it above here. It just needs to be sorted using a "natural" comparator. I haven't worked much in the HA codebase, and not at all in the frontend. Not sure if the sorting happens in frontend or API for that matter.

It would probably even be faster to open an issue for the problem in HA/frontend, where someone more familiar with how that works would have a better understanding of where the fix should be made. If I have time I'll try to dig in on that and track down where the sort happens. There are relatively simple implementations of [natural sorting](https://en.m.wikipedia.org/wiki/Natural_sort_order) in most languages, so whether it's in python or JavaScript, it should be easy to do. And the change would only impact the order of values with multi-digit numbers, otherwise it's identical to alphabetical sorting.

jhansche, 2022-06-02:
Another possible way to address this could be to change the behavior of climate fan mode. Right now they have a small handful of normal fan modes (like high/medium/low, and a couple others), and that's it. While the `fan` entity on the other hand has a min/max/step config, which means the fan speed is actually displayed as a slider with predefined set points. Ideally we would be able to set a supported_features flag to say that we have a linear fan speed range (compared to L/M/H style), and let the UI change to reflect that slider style.

But it's kind of a big change to make in the climate entity, when most won't actually suffer from this problem. I imagine the proposal would not get much traction.

With a fan bedjet platform, we could expose the climate entity in one way and also expose a fan entity just for controlling the fan speed. Could even link the fan entity to the climate entity, which would allow us to disable the fan mode in the climate UI if there's already a separate fan entity for it.

jhansche, 2022-06-02:
I'm about to open a PR that refactors the component into a Hub with child components (currently only Climate as a child). As part of that refactor, I've also been able to address the "throttle" issues, and expose simpler methods from the `bedjet` component (the "Hub"), which will open the door for more interactions with it.

Try it out, using [external_components](https://esphome.io/components/external_components.html) in your config. I'll update here once the PR is ready.

jhansche, 2022-06-03:
https://github.com/esphome/esphome/pull/3522 is the PR. You can use the external_components link above to test it.

Note that this is a breaking change and requires shifting the yaml config around (explained in the PR, visualized in the PR's test1.yaml changes, and also documented in the linked docs PR as well).

user897943, 2022-06-03:
Unfortunately unable to test until the new board arrives, the one I'm using might be defective.

user897943, 2022-06-03:
@jhansche do you have any idea why my log keeps showing that it can't connect to the BJ unit? It's connected to WiFi... I hope it's this stupid board and the new one will fix...

[18:01:56][I][ble_client:085]: Attempting BLE connection to 08:3a:f2:4b:fe:1e
[18:01:56][D][bedjet:610]: [BedJet 1 - Front room] Not connected, enabled=true
[18:01:57][W][ble_client:117]: connect to 08:3a:f2:4b:fe:1e failed, status=133

jhansche, 2022-06-03:
I can't say for sure, except that mine shows the same status=133 (esp ble "generic error" status code) intermittently. It eventually connects, but it frequently takes multiple attempts. Couple that with the instability while streaming logs, and mine actually does that a couple times a minute. When I stop watching logs, it stabilizes.

Does yours eventually connect, or is it constantly showing that 133 error? I have found that sometimes power cycling the bedjet will clear things out and let it connect successfully, but I haven't identified exactly what triggers it to fail in the first place.

For what it's worth, that error code is coming straight from the BLEClient's disconnect event, which just forwards that from the built-in esp ble library, and not something that my component is introducing. You can verify that by using the simple ble_client sensor component and giving it your bedjet's mac and one of the bedjet gatt service characteristics, and you'll see the same connectivity behaviors.

mwolter805, 2022-06-04:
Interesting, I haven’t seen the error 133… yet. Knock on wood. Don’t want to jinx myself 🙂 It might be because I don’t use WiFi and have used either a QuinESP ABE or Olimex ESP32-POE for ESP’s doing Bluetooth. Throwing this out there as it might help narrowing down what’s causing this issue.

user897943, 2022-06-04:
> I can't say for sure, except that mine shows the same status=133 (esp ble "generic error" status code) intermittently. It eventually connects, but it frequently takes multiple attempts. Couple that [...]

Mine never connects.. sometimes I changes, says it's trying to connect then it doesn't, gets refused I think.

It was working before. Can't wait to get the new board, I got a ESP-WROOM-32U which comes with a U.FL connector, and a 2.5db antenna that plugs into that, should provide the most reliable radio pathway possible. @mwolter805 could be right, I think because the ESP shares the same antenna for WiFi and BT it's probably overloading the SoC.

@jhansche could you open an issue about tweaking the BT delays in ESPHome/BT stack, I think the delays are too short and causing issues because the device doesn't have enough time to do the handshake/transfer the data fully before the next command is issued. Some googling suggested this to be a potential cause for this error 133. For reference, I use a Ubiquiti UniFi AP-HD that sits about 5ft from the ESP board, and the ESP board sits about 3ft from the BedJet 3, so it should definitely not be struggling signal-wise.

jhansche, 2022-06-05:
I don't believe this is an Espressif or ESPHome issue to address. The BLE notification spec is intended for the BLE server (bedjet in this case) to notify when a characteristic changes. In other devices that might happen once or a few times a minute, like when the temp changes or something like that, or for something like a contact sensor it would happen whenever the sensor opens or closes. It's the bedjet that is sending events more rapidly than usual.

It's been a while since I looked into that, and I think there could be some level of negotiation that happens where the client tells the server what it's packet window is, or something like that, and iirc that might be where this could potentially be modified. But from what little I can remember, I don't think that we have the ability to configure that. I'll see if I can look into that again, because I agree that it would help if we could limit that to even once per second or so.

Also note that there can only be one client connected to the server, so if the bedjet thinks a device is still connected and doesn't recognize that the new connection is the same client, it could be refusing the connection in that case. The same would happen if you had the bedjet app open at the time, because the app would prevent the esp32 from connecting. Somewhat related, I've seen at least once where the bedjet got into a state where it just wouldn't accept my connection anymore, and oddly enough I think opening the app did something to "reset" it, so that when I closed the app, the esp32 was able to connect then

kquinsland, 2022-06-05:
> I'll see if I can look into that again, because I agree that it would help if we could limit that to even once per second or so.

It's been a _long_ time since I did any detailed BTLE protocol analysis but if memory serves, writing to the CCCD (Client Characteristic Configuration Descriptor) would allow you to silence them?

Wait a few seconds and then re-enable them / listen for a while?

It would be _ideal_ if it was possible to negotiate a rate but thought that was up to the device/server to implement :(.

user897943, 2022-06-05:
I tried to follow the above suggestions regarding app, no dice.

One thing I noticed, when the ESPHome logs appears, it's missing speeds... Could that be related, like the BJ is comparing what the "remote" from ESPHome supports, sees it is missing speeds then refuses to connect? So far I factory reset the BJ, tried to do the open app/close app thing, pressed the low power button twice to put BT in pairing mode, took batteries out of remote... It refuses to connect and gives 133. This damn replacement ESP is taking a while to get here.. but the current esp is connecting to wifi no problem... I am not sure if it's related but I was on the 2022.5 dev branch, it was working, then upgrade to 2022.6 dev branch and these problems appeared around the same time so maybe it's the Arduino library? I'm just guessing, not a coder.

```
14:05:21][C][wifi:491]: WiFi:
[14:05:21][C][wifi:353]:   Local MAC: 7C:9E:BD:66:5B:D0
[14:05:21][C][wifi:354]:   SSID: [redacted]
[14:05:21][C][wifi:355]:   IP Address: 10.0.0.142
[14:05:21][C][wifi:357]:   BSSID: [redacted]
[14:05:21][C][wifi:358]:   Hostname: 'bedjetv3-fb1'
[14:05:21][C][wifi:360]:   Signal strength: -76 dB ▂▄▆█
[14:05:21][C][wifi:364]:   Channel: 11
[14:05:21][C][wifi:365]:   Subnet: 255.255.255.0
[14:05:21][C][wifi:366]:   Gateway: 10.0.0.1
[14:05:21][C][wifi:367]:   DNS1: 10.0.0.1
[14:05:21][C][logger:275]:   Level: DEBUG
[14:05:21][C][logger:276]:   Log Baud Rate: 115200
[14:05:21][C][logger:277]:   Hardware UART: UART0
[14:05:21][C][template.binary_sensor:018]: Template Binary Sensor 'BedJet FB 1 Connected'
[14:05:21][C][template.binary_sensor:018]:   Device Class: 'connectivity'
[14:05:21][C][homeassistant.time:010]: Home Assistant Time:
[14:05:21][C][homeassistant.time:011]:   Timezone: 'UTC0'
[14:05:21][C][ble_switch:035]: BLE Client Switch 'BedJet Enable (FB 1)'
[14:05:21][C][ble_switch:035]:   Icon: 'mdi:bluetooth'
[14:05:21][C][restart:022]: Restart Switch 'BedJet FB Bridge Restart'
[14:05:21][C][restart:022]:   Icon: 'mdi:restart'
[14:05:21][C][esp32_ble_tracker:712]: BLE Tracker:
[14:05:21][C][esp32_ble_tracker:713]:   Scan Duration: 30 s
[14:05:21][C][esp32_ble_tracker:714]:   Scan Interval: 500.0 ms
[14:05:21][C][esp32_ble_tracker:715]:   Scan Window: 50.0 ms
[14:05:21][C][esp32_ble_tracker:716]:   Scan Type: ACTIVE
[14:05:21][C][ble_client:035]: BLE Client:
[14:05:21][C][ble_client:036]:   Address: 08:3a:f2:4b:fe:1e
[14:05:21][C][mdns:084]: mDNS:
[14:05:21][C][mdns:085]:   Hostname: bedjetv3-fb1
[14:05:21][C][ota:085]: Over-The-Air Updates:
[14:05:21][C][ota:086]:   Address: 10.0.0.142:3232
[14:05:21][C][ota:089]:   Using Password.
[14:05:21][C][api:138]: API Server:
[14:05:21][C][api:139]:   Address: 10.0.0.142:6053
[14:05:21][C][api:143]:   Using noise encryption: NO
[14:05:21][C][bedjet:057]: BedJet Climate 'BedJet 1 - Front room'
[14:05:21][C][bedjet:060]:   Supported modes:
[14:05:21][C][bedjet:062]:    - OFF
[14:05:21][C][bedjet:062]:    - HEAT
[14:05:21][C][bedjet:062]:    - FAN_ONLY
[14:05:21][C][bedjet:062]:    - DRY
[14:05:21][C][bedjet:065]:   Supported fan modes:
[14:05:21][C][bedjet:070]:    - 20% (c)
[14:05:21][C][bedjet:070]:    - 25% (c)
[14:05:21][C][bedjet:070]:    - 30% (c)
[14:05:21][C][bedjet:070]:    - 35% (c)
[14:05:21][C][bedjet:070]:    - 40% (c)
[14:05:21][C][bedjet:070]:    - 45% (c)
[14:05:21][C][bedjet:070]:    - 5% (c)
[14:05:21][C][bedjet:070]:    - 50% (c)
[14:05:21][C][bedjet:070]:    - 55% (c)
[14:05:21][C][bedjet:070]:    - 60% (c)
[14:05:21][C][bedjet:070]:    - 65% (c)
[14:05:21][C][bedjet:070]:    - 70% (c)
[14:05:21][C][bedjet:078]:    - M3 (c)
[14:05:24][D][ble_client:047]: Found device at MAC address [08:3A:F2:4B:FE:1E]
[14:05:24][I][ble_client:085]: Attempting BLE connection to 08:3a:f2:4b:fe:1e
[14:05:24][W][ble_client:117]: connect to 08:3a:f2:4b:fe:1e failed, status=133
[14:05:24][D][esp32_ble_tracker:217]: Starting scan...
[14:05:24][D][ble_client:047]: Found device at MAC address [08:3A:F2:4B:FE:1E]
[14:05:24][I][ble_client:085]: Attempting BLE connection to 08:3a:f2:4b:fe:1e
[14:05:24][W][ble_client:117]: connect to 08:3a:f2:4b:fe:1e failed, status=133
[14:05:24][D][esp32_ble_tracker:217]: Starting scan...
[... 75 lines cut ...]
[14:05:37][D][ble_client:047]: Found device at MAC address [08:3A:F2:4B:FE:1E]
[14:05:37][I][ble_client:085]: Attempting BLE connection to 08:3a:f2:4b:fe:1e
[14:05:37][W][ble_client:117]: connect to 08:3a:f2:4b:fe:1e failed, status=133
[14:05:37][D][esp32_ble_tracker:217]: Starting scan...
[14:05:37][D][esp32_ble_tracker:726]: Found device 47:33:39:A6:8B:2F RSSI=-81
[14:05:37][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[14:05:38][D][ble_client:047]: Found device at MAC address [08:3A:F2:4B:FE:1E]
[14:05:38][I][ble_client:085]: Attempting BLE connection to 08:3a:f2:4b:fe:1e
[14:05:38][W][ble_client:117]: connect to 08:3a:f2:4b:fe:1e failed, status=133
[14:05:38][D][esp32_ble_tracker:217]: Starting scan...
[14:05:40][D][esp32_ble_tracker:726]: Found device 62:AB:EF:98:68:48 RSSI=-93
[14:05:40][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[14:05:40][D][esp32_ble_tracker:726]: Found device 6C:EC:EC:E4:CD:46 RSSI=-79
[14:05:40][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[14:05:40][D][ble_client:047]: Found device at MAC address [08:3A:F2:4B:FE:1E]
[14:05:40][I][ble_client:085]: Attempting BLE connection to 08:3a:f2:4b:fe:1e
[14:05:40][W][ble_client:117]: connect to 08:3a:f2:4b:fe:1e failed, status=133
[14:05:40][D][esp32_ble_tracker:217]: Starting scan...
[14:05:40][D][ble_client:047]: Found device at MAC address [08:3A:F2:4B:FE:1E]
[14:05:40][I][ble_client:085]: Attempting BLE connection to 08:3a:f2:4b:fe:1e
[14:05:40][W][ble_client:117]: connect to 08:3a:f2:4b:fe:1e failed, status=133
[14:05:40][D][esp32_ble_tracker:217]: Starting scan...
[14:05:41][D][esp32_ble_tracker:726]: Found device 60:C7:94:BF:9F:85 RSSI=-72
[14:05:41][D][esp32_ble_tracker:747]:   Address Type: RANDOM
```

mwolter805, 2022-06-05:
A guess but if I remember correctly the BedJet is sending status updates every second as the auto-off timer decrements. I believe a while back I was noticing this when decoding the BLE data received. Combine these frequent updates with the overhead from debug logging and the Wi-Fi/Bluetooth resource conflicts and it could be causing these errors. 

I believe the BedJet is using an ESP32 internally. Wish we could load ESPHome directly on it and just use MQTT! Local control would be so much easier!

kquinsland, 2022-06-05:
> I believe the BedJet is using an ESP32 internally. 

V3 is.

> Wish we could load ESPHome directly on it and just use the API or MQTT. Local control would be so much easier!

That was my original goal. Unfortunately the ESP32 is not running the show. Heat/Fan control is done with a dedicated micro-controller. I hit a variety of issues while trying to do static analysis on the ESP <->  8051 communications. a TLDR of some of my findings is here: https://github.com/esphome/issues/issues/3292#issuecomment-1141608916

jhansche, 2022-06-05:
The BLE notification is sent by the BedJet, and it happens more frequently than once per second (wrt the comment about auto-off countdown).

The connection parameters is where we should be able to set the min/max interval. But nothing I've tried has worked successfully.

As for trying to toggle the CCCD, I tried that early on when I was trying to throttle the notifications, but that was not successful. It's been a while since I tried it so I don't recall what the exact issue was - just that it did not do what I was hoping for. There were other issues as well though, such as the CCCD code in ble_client didn't seem to be written correctly (sending an 8-bit value when BLE spec calls for 16-bit). I worked around that by rewriting my own control of CCCD, and that resolved most of the issues around staying connected.

I tried to inspect the GAP connection parameters using `esp_ble_get_current_conn_params`, but for some reason I can't get that to compile 🤷  Which doesn't make sense because we are definitely using other gap_ble_api functions in other ble_* components. And using `esp_ble_gap_set_prefer_conn_params` to set the min/max interval, does not seem to have any affect at all, even though it returns ESP_OK indicating success. But I have a feeling that this negotiation is still dependent on the BedJet being willing to accept the requested parameters, and from what I can tell, it seems that it just ignores the preferred params. I do get ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT events, showing that it is using different values for the min/max from what I asked for, and then actually lowering them to the ESP absolute minimum (6ms) - which I don't know where that request is coming from:
```
// not what I asked for:
[16:43:43][V][ble_client:230]: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT: status=0, min=16, max=32
// absolute min:
[16:43:43][V][ble_client:230]: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT: status=0, min=6, max=6
```

There is no code in ESPHome that is calling the `esp_ble_gap_update_conn_params` function, so this tells me it's either happening automatically in the esp32/arduino library, or the BedJet is specifically setting it and we get the callback as a result. If I react to that event by trying to update the params again, I get back `status=19`, which resolves to `ESP_BT_STATUS_ERR_ILLEGAL_PARAMETER_FMT`.

So it seems that we're just not able to force the BedJet into using the parameters we want.

At some point I want to try a test switching to esp32-idf and see if that's any better.

user897943, 2022-06-06:
Guys you're clearly experienced in dealing with this, could you please take a look at this:
https://github.com/esphome/feature-requests/issues/1734

The manufacturer refuses to send me the original bin to reflash the ESP32 and insist the unit is "bricked" which is total bullshit.. I am recovering from surgery and being able to control both the AC and the Bedjet is important, at least with the bedjet I have the app.. the AC only works with the IR remote which I cannot use from a distance.

user897943, 2022-06-06:
Do you know that removing the battery from the remote, then inserting it back while the MENU key is pressed down can give you a test menu, with the up/down keys there are some options like RX and TX tests... could this be useful in trying to decipher the data refresh frequency for example?

I am still getting those 133 errors. I tried to reset the unit, turn it on and off, press and hold low power button, press the button twice... Only hope now is that the new ESP arriving in a few days will fix this. It's weird because it was working before, but since then I basically butchered the ESP trying different things, so I might have damaged hardware but the fact that it's scanning devices indicates the BT portion is working!? I also downgraded ESPhome back to 2022.5.1 which was working before, no dice.

kquinsland, 2022-06-07:
> Do you know that removing the battery from the remote, then inserting it back while the MENU key is pressed down can give you a test menu, with the up/down keys there are some options like RX and TX tests... could this be useful in trying to decipher the data refresh frequency for example?

Ohh I was wondering how to get in there. There's a lot of interesting strings related to test/debug mode in the firmware for the remote. Unfortunately, the remote uses ANT wireless not BTLE so i doubt any of the rf metrics you'd see in that test mode are going to be useful for BTLE.

> I am still getting those 133 errors. I tried to reset the unit, turn it on and off, press and hold low power button, press the button twice...

Turn the BT radio off on every mobile device you've used to connect to the bedjet. I had some issues with ESPHome not being able to connect and it was due to my iPad still being connected to the device... even though the bed jet app was no longer open. I turned BT off on the iPad and my phone and waited about 10s. The ESPHome device / bedjet component was immediately able to connect once I did that.

user897943, 2022-06-07:
> Ohh I was wondering how to get in there. There's a lot of interesting strings related to test/debug mode in the firmware for the remote. Unfortunately, the remote uses ANT wireless not BTLE so i [...]

Should be just my cellphone, nothing else. I'll try turning off BT, I had a hemorrhage and now I'm stuck in the hospital, sigh. Thanks for the suggestion.

user897943, 2022-06-09:
Back home. Got the new board. Assembled everything, uploaded the code, it's working now.
But got an error on the log, in case you're interested:
```
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 10.0.0.142 using esphome API
INFO Successfully connected to 10.0.0.142
[11:02:30][I][app:102]: ESPHome version 2022.6.0-dev compiled on Jun  9 2022, 11:01:13
[11:02:32][I][ble_client:085]: Attempting BLE connection to 08:3a:f2:4b:fe:1e
[11:02:35][I][ble_client:166]: Service UUID: 0x1801
[11:02:35][I][ble_client:167]:   start_handle: 0x1  end_handle: 0x5
[11:02:35][I][ble_client:378]:  characteristic 0x2A05, handle 0x3, properties 0x20
[11:02:35][I][ble_client:166]: Service UUID: 0x1800
[11:02:35][I][ble_client:167]:   start_handle: 0x14  end_handle: 0x1c
[11:02:35][I][ble_client:378]:  characteristic 0x2A00, handle 0x16, properties 0x2
[11:02:35][I][ble_client:378]:  characteristic 0x2A01, handle 0x18, properties 0x2
[11:02:35][I][ble_client:378]:  characteristic 0x2AA6, handle 0x1a, properties 0x2
[11:02:35][I][ble_client:166]: Service UUID: 00001000-BED0-0080-AA55-4265644A6574
[11:02:35][I][ble_client:167]:   start_handle: 0x28  end_handle: 0xffff
[11:02:36][I][ble_client:378]:  characteristic 00002000-BED0-0080-AA55-4265644A6574, handle 0x2a, properties 0x1a
[11:02:36][I][ble_client:378]:  characteristic 00002001-BED0-0080-AA55-4265644A6574, handle 0x2d, properties 0xa
[11:02:36][I][ble_client:378]:  characteristic 00002002-BED0-0080-AA55-4265644A6574, handle 0x2f, properties 0xa
[11:02:36][I][ble_client:378]:  characteristic 00002003-BED0-0080-AA55-4265644A6574, handle 0x31, properties 0x8
[11:02:36][I][ble_client:378]:  characteristic 00002004-BED0-0080-AA55-4265644A6574, handle 0x33, properties 0x8
[11:02:36][I][ble_client:378]:  characteristic 00002005-BED0-0080-AA55-4265644A6574, handle 0x35, properties 0xa
[11:02:36][I][ble_client:378]:  characteristic 00002006-BED0-0080-AA55-4265644A6574, handle 0x37, properties 0xa
[11:02:37][W][bedjet:319]: Error writing descr at handle 0x0043, status=13
```

jhansche, 2022-06-09:
I assume you're referring to the last [W] line?

I log that as a warning, because it's technically unexpected, and I don't want to filter it out for any reason. But the status=13 error code is for an invalid attr length, which is something I had to work around.

BLEClient requires that we tell it we want status notifications (otherwise we won't get them), and when we tell it that, it automatically writes the CCCD notify flag. However, BedJet doesn't accept the CCCD flag in the format that BLEClient sends it: BLEClient sends it as an 8-bit attribute, but BedJet and the BLE spec expect that to be a 16-bit attribute. So I have my own code that writes the descriptor as 16-bit.

After that [W] log, you should also see another log that indicates the bedjet component wrote the notify flag successfully, which is the workaround I added.

It's probably technically a bug in BLEClient, but given that it's used by other components that presumably don't respond with this error code, it probably stands to reason that some devices accept it as 8-bit, and maybe they would then start failing if we "fix" BLEClient to send 16-bit.

I've considered making a PR to change BLEClient, but that was my hesitation. I guess I could submit the fix anyway, and at least start the dialog with other BLEClient components.

kquinsland, 2022-06-09:
>  I guess I could submit the fix anyway, and at least start the dialog with other BLEClient components.

+1 for this. Bedjet can't be the only BTLE device out there that has this requirement.

user897943, 2022-06-09:
> I assume you're referring to the last [W] line?
>
> I log that as a warning, because it's technically unexpected, and I don't want to filter it out for any reason. But the status=13 error code is for [...]

Thanks for explaining. I think you should submit it. Your integration is working to spec, if others break they need to fix theirs.

jhansche, 2022-06-10:
PR submitted to fix this issue, as well as another issue dealing with multiple BLE connections: https://github.com/esphome/esphome/pull/3544

jhansche, 2022-08-22:
Additional improvement at https://github.com/esphome/esphome/pull/3735 adds a new `fan` component, which can be used to present a simpler Fan widget (has simple on/off switch and numeric slider from 5%-100%).

Feel free to test it out using [`external_components`](https://esphome.io/components/external_components.html):
```yaml
external_components:
  - source: github://pr#3735
    components: [bedjet]
```

Note that because this will share the same upstream connection to the device with the `climate` platform, changes in one entity will also cause related changes in the other. In the future, we might be able to configure the `climate` component to disable the fan modes if the BedJet Fan component was used.