esphome/issues #3288: INA226 alert pin Esphome

tiimsvk, 2022-05-16:
### The problem

is it planned to add for ina226 alert pin to esphome?
[image]
[Datasheet](https://www.ti.com/lit/ds/symlink/ina226.pdf) section 7.3.1.2 Alert Pin

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.4

### What platform are you using?

ESP8266

### Component causing the issue

ina226

martgras, 2022-05-16:
As an alternative you can use https://esphome.io/components/sensor/index.html#on-value-range and trigger another action

tiimsvk, 2022-05-18:
> As an alternative you can use https://esphome.io/components/sensor/index.html#on-value-range and trigger another action

Well thank you. of course I use this component