esphome/issues #3289: MAX31865 faulty reading when module not connected

ThomDietrich, 2022-05-16:
### The problem

Hey all,
I am working with the max31865 and a PT1000 and everything is great in good conditions. The issue I have observed is that the generated entity receives various faulty values when a the max31865 module is not actually connected.

**Config:**

```
  - platform: max31865
    name: "PT1000 Temperatur 1"
    update_interval: 10s
    cs_pin: GPIO5
    reference_resistance: 430 Ω
    rtd_nominal_resistance: 1000 Ω
    rtd_wires: 2
```

**Result:**

[image]

In my case, the invalid values seem to lie around -142°C as well as -242°C, among other values as can be seem in graph.

**Expectation:**

No SPI communication with the MAX31865 should result in a deterministic "unkown" or "unavailable" state for the sensor reading.

CC @DAVe3283

### Which version of ESPHome has the issue?

 2022.2.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2022.5

### What platform are you using?

ESP8266

### Board

az-delivery-devkit-v4

### Component causing the issue

max31865

DAVe3283, 2022-05-16:
My MAX31865 is buried inside a soda fountain so I can't easily get to it and disconnect wires. It would be helpful if you could set logging to more verbose, and send me the output from as many bad readings as possible.

For example, add this to your YAML:
```yaml
logger:
  # Get detailed output from max31865
  level: VERY_VERBOSE
  logs:
    # Silence unrelated chatty components
    api: DEBUG
    api.connection: DEBUG
    api.service: DEBUG
    app: DEBUG
    scheduler: DEBUG
    sensor.filter: DEBUG
    sensor: DEBUG
```

I do know that I didn't do a sanity check for all zero or all one responses, which occasionally comes up for me too, but like 2-3 times a year. I will at least get that check added, which likely will solve your problem too. But please get me examples if you can.

DAVe3283, 2022-05-16:
> > buried inside a soda fountain
> 
> Please please tell me more xD

<details>
<summary>Expand for off-topic soda fountain stuff</summary>

I bought a used soda fountain that had the chiller controller fail, so I made my own. It is a counter electric (ice bank) style machine, so I need to maintain a certain level of ice around the chiller. I do this by letting it thaw back past the sensor, then turn on the chiller and grow ice over it. Seems to work very well.
[image]

When all is said and done, it isn't cheaper than buying cans of soda, but it is a lot more fun!
[image]
</details>

> Would need to read up on SPI but I hoped it should be super simple to judge whether a chip actually responded or not? Actually, while the state of my entity is currently "Unknown", would it make sense to report "unavailable" in that case? Not sure what the guideline from esphome is for this.

On most faults, the sensor publishes NAN, which I believe reports as unknown. My understanding is Unavailable indicates a sensor is offline, while Unknown is reported for a fault/problem while the ESPHome device is online. I will see if I can find any official guidance from ESPHome on this, since I am going to be in there anyway.

> I will get you those logging lines! I am currently traveling and will post anything I can dig up next week.

I ordered a PT1000 setup from AdaFruit and will hopefully be able to replicate your setup exactly next week. I had been meaning to do this for my soda fountain for some time now anyway, but mine only glitches very rarely, likely due to the EMI when the compressor kicks on at the exact same time a reading is taken.

Actually, looking at your config in the issue, do you have a PT1000 with a 430 Ω reference resistor? I thought it was supposed to be a 4300 Ω reference resistor for the PT1000, and the 430 for the PT100. Or was that just a typo here?

DAVe3283, 2022-05-24:
Thanks for the data, that is great! I just got a PT1000 in to test with, I will get started after work today on updating the code to handle things better.

Your 2nd example shows the MAX31865 reporting a fault. Check your wiring is good between the MAX and the PT1000 sensor, especially the connection.

ThomDietrich, 2022-05-24:
> Your 2nd example shows the MAX31865 reporting a fault.

I believe that's either due to the module being not great or its related to the longer wires between. Not sure if you can do something in terms of not publishing "nan", which arguably might be a solution or shading another issue for other users.

DAVe3283, 2022-05-26:
> I believe that's either due to the module being not great or its related to the longer wires between. Not sure if you can do something in terms of not publishing "nan", which arguably might be a [...]

You can easily mask that on your end. For example, I have a DS18B20 that sometimes fails to read due to long wires, and I just filter out the failure:

```yaml
sensor:
  - platform: dallas
    name: "Garage Freezer Temperature"
    address: 0xAE032197799CC128
    accuracy_decimals: 2
    # Due to bit errors on the 1-Wire bus, some readings fail checksum
    # Rather than report NaN, simply filter these transient errors out
    filters:
      - filter_out: NaN
```

ThomDietrich, 2022-05-27:
Right, of course. That was too obvious :D

Were you able to reproduce the `0.00Ω --> -242.02°C` error on your side?

DAVe3283, 2022-05-27:
> Right, of course. That was too obvious :D
> 
> Were you able to reproduce the `0.00Ω --> -242.02°C` error on your side?

Yes, and I hope to have a PR at some point this weekend to fix that.