esphome/issues #3290: Build fails on macOS and Windows

ivankravets, 2022-05-18:
### The problem

Hi,

We've started testing each new commit to the PlatformIO Core with the popular community projects. ESPHome does not build on Windows and macOS => https://github.com/platformio/platformio-core/actions/runs/2345627975

Is this normal?

### Which version of ESPHome has the issue?

Git

### What type of installation are you using?

pip

### What platform are you using?

ESP32

ivankravets, 2022-05-19:
What is the right way to test ESPHome with PlatformIO for CI? See https://github.com/platformio/platformio-core/blob/develop/.github/workflows/projects.yml

jesserockz, 2022-05-19:
Yeah I checked out how you were doing it. What you have done is probably correct. I am currently working on fixing the windows build, but I do not have a mac to test that one.

Also your matrix excludes are wrong as there is no `os` key and probably should be `name` or similar.

ivankravets, 2022-05-19:
I forgot to add, that it now compiles on macOS. So, only Windows fails.

closed 2022-10-01 by github-actions[bot]