esphome/issues #3293: ESP32 no longer able to make or receive TCP connections

jsnfwlr, 2022-05-19:
### The problem

My original ESP32 DevkitV1 (loaded as a `nodemcu-32s` in the config) was working fine until I upgraded ESPHome to 2022.4.0 and reflashed the board with an unmodified but updated version of the config. After reflashing the board worked, but I woke the next day to find it offline.

I plugged it in to my machine and pulled up the logs and saw it was showing all the normal information, but was disconnecting from MQTT immediately after attempting to connect.

I wasn't able to ping the board either.

I have since stripped out all the components until the build is just wifi, api, ota, and web-server (added just to see if I could trick the board in to doing something, anything!)

I can't ping the board, but both nmap and my own custom built network scanner can find it on the network, and my router reports it is there too.

Thinking it could be an issue with the latest build, I rolled my docker container back to 2022.3.5 and performed a clean build on a new board with a new config name, and the error persisted.

### Which version of ESPHome has the issue?

2022.4.0 and 2022.5.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.5.4

### What platform are you using?

ESP32

### Board

nodemcu-32s and esp32s

### Component causing the issue

mqtt, web-server, api

### Example YAML snippet

```yaml
esphome:
  name: loungescanner
  comment: BLE Scanner for HA beacons

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
  level: VERY_VERBOSE

wifi:
  ssid: !secret wifissid
  password: !secret wifipass
  manual_ip:
    static_ip: 192.168.3.225
    gateway: 192.168.0.1
    subnet: 255.255.252.0
  
ota:
  password: !secret otapass
  
api:
  password: !secret apipass
  encryption:
    key: !secret encryptionkey

text_sensor:
[... 249 lines cut ...]
```

### Additional information

$ nmap -Pn 192.168.3.225 -p 80,8266,6053
Starting Nmap 7.92 ( https://nmap.org ) at 2022-05-19 13:54 AWST
Nmap scan report for 192.168.3.225
Host is up.

PORT     STATE    SERVICE
80/tcp   filtered http
6053/tcp filtered x11
8266/tcp filtered espeasy-p2p

Nmap done: 1 IP address (1 host up) scanned in 3.05 seconds

$ ping 192.168.3.225
PING 192.168.3.225 (192.168.3.225) 56(84) bytes of data.
From 192.168.0.41 icmp_seq=1 Destination Host Unreachable
From 192.168.0.41 icmp_seq=2 Destination Host Unreachable
From 192.168.0.41 icmp_seq=3 Destination Host Unreachable
^C
--- 192.168.3.225 ping statistics ---
11 packets transmitted, 0 received, +3 errors, 100% packet loss, time 10218ms
pipe 4

rcambrj, 2022-05-20:
I had a similar problem whereby a fresh ESP32 would not respond over wifi until I downgraded esphome to 2022.1.4, flashed the same esphome configuration to it via USB (but the old version), then upgraded again to the latest 2022.05 (beta), this time via wifi and it's now responding OK. Maybe tomorrow it won't, like @jsnfwlr 's.

Something's up with the latest version, but I can't say exactly what.

jsnfwlr, 2022-05-20:
I have tried running ESPrense on the ESP32 as a way to avoid this issue - I just need some BLE scanners - and while it works for the initial setup via the captive portal with web server, it doesn't work once I complete the configuration and restart the device.

@rcambrj out of curiosity, what router and wifi ap are you using. I have a wired mesh of TP-Link Omada EAP245HD and a TP-Link Omada ER7206. I am considering setting up another AP on a WiFi NIC added to my server to see if that improves matters at all.

jsnfwlr, 2022-05-20:
I've just created a new, clean docker container using 2021.12.0, and it still isn't working.

jsnfwlr, 2022-05-20:
I got it working under 2021.12.0 by flashing it first with a generic config using ESP-IDF for the framework, then going back to what I was using before

jsnfwlr, 2022-05-22:
I was able to flash it with an updated bin from 2022.5.0 and had it online for most of the day yesterday, but then it reverted back to failing this morning, and I can't get it working with the same 2022.5.0 bin.

pmannk, 2022-05-31:
Strange. I've had this behaviour with ESP32s when using BLE (they use a shared radio and making heavy BLE+WiFi demands on them at the same time creates problems), but yours seems to be doing this even with a stripped down config?
I've flashed a few ESP32s with 2022.5.0 and 2022.5.1 and they've remained stable.

One thing that might help you debug further is to remove your WiFi client settings from the ESP and switch the device to run captive portal instead (https://esphome.io/components/captive_portal.html). Then you can connect a client device to it and see if the WiFi remains stable when the ESP operates in Access Point mode.

It doesn't explain why the device seemed to be working fine previously, but something thing to check on your Access Point(s) is whether they have 20MHz or 40MHz channel widths configured. I had to drop my APs down to 20MHz width for some devices to remain stable. I also use Short Preamble.

closed 2022-10-18 by github-actions[bot]