esphome/issues #3294: Having issues setting temperature compensation to EZO pH Sensor

rchamp26, 2022-05-20:
### The problem

I may be having an issue with understanding how to implement the lambda call, but I receive an issue when trying to compile. it seems that esphome doesn't know what the set_tempcomp_value flag is.

any help in how I can potentially implement the temperature compensation would be great. I noticed after the past couple weeks that the ezo read is .2-.4 off from my other handheld pH sensors. I already calibrated and during calibration inside, the readings were within .05-.1 which is ok in my aquaponics setup, however after moving the sensor back outside, the reading seems to be way off an I believe it may be due to not being able to compensate for the elevated temperatures outside and in the water, which my handhelds account for

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.0

### What platform are you using?

ESP32

### Board

hiletgo esp-wroom-32

### Component causing the issue

i2c

### Example YAML snippet

```yaml
sensor:
  - platform: ezo
    id: ph_ezo
    name: "pH sensor"
    address: 99
    unit_of_measurement: "pH"
    accuracy_decimals: 2
    update_interval: 10s
    
    
  - platform: ezo
    id: rtd_ezo
    name: "RTD Temperature"
    address: 102
    accuracy_decimals: 2
    unit_of_measurement: "°C"
    device_class: temperature
    update_interval: 10s
  
  - platform: template
    id: ph_ezo_tempcomp
    name: "pH Temp Comp"
    unit_of_measurement: "pH"
    lambda: |-
      return id(ph_ezo).set_tempcomp_value(id(rtd_ezo).state);
```

### Anything in the logs that might be useful for us?

```txt
Here is the error I get 

Compiling /data/aquaponics-control/.pioenvs/aquaponics-control/src/main.cpp.o
/config/esphome/aquaponics-control.yaml: In lambda function:
/config/esphome/aquaponics-control.yaml:64:40: error: could not convert 'ph_ezo->esphome::ezo::EZOSensor::set_tempcomp_value(rtd_ezo->esphome::ezo::EZOSensor::<anonymous>.esphome::sensor::Sensor::state)' from 'void' to 'esphome::optional<float>'
       return id(ph_ezo).set_tempcomp_value(id(rtd_ezo).state);
                                        ^
*** [/data/aquaponics-control/.pioenvs/aquaponics-control/src/main.cpp.o] Error 1
```

rchamp26, 2022-05-20:
I think I may have figured out the implementation, but not sure if it has implemented and is actually compensating the value based on the temp being passed in. this is what I have now. my main question the more i look at this, is, do i put the lambda in the RTD temp sensor so on value, it pushes that temp comp to the PH or do I leave as is? I tried moving to the temp sensor, and the values being returned by the ph_ezo sensor seem to be the same regardless of whether the lambda is in the RTD or ph sensor. I do think that the temp comp is working as i did notice a .001-.01 increase in value, which is what the Atlas scientific compensation calculator estimates the change would be, but is still way off from my other handhelds. I may need to pick up an additional ezo and ph probe to see if that is the issue as well as perhaps some other tests in UART mode offline to see if this is a probe/ezo issue or something different happening in reporting of values over UART vs i2c

```  
  - platform: ezo
    id: ph_ezo
    name: "pH sensor"
    address: 99
    unit_of_measurement: "pH"
    accuracy_decimals: 2
    update_interval: 10s
    on_value:
      then: 
        - lambda: |-
            id(ph_ezo).set_tempcomp_value(id(rtd_ezo).state);
          
    
  - platform: ezo
    id: rtd_ezo
    name: "RTD Temperature"
    address: 102
    accuracy_decimals: 2
    unit_of_measurement: "°C"
    device_class: temperature
    update_interval: 10s
```

ssieb, 2022-07-22:
Github issues are not a support forum.  Please use the discord server instead.  You should use the temperature on_value to set the compensation instead.

```yaml
  - platform: ezo
    id: ph_ezo
    name: "pH sensor"
    address: 99
    unit_of_measurement: "pH"
    accuracy_decimals: 2
    update_interval: 10s

  - platform: ezo
    id: rtd_ezo
    name: "RTD Temperature"
    address: 102
    accuracy_decimals: 2
    unit_of_measurement: "°C"
    device_class: temperature
    update_interval: 10s
    on_value:
      then: 
        - lambda: id(ph_ezo).set_tempcomp_value(x);
```

closed 2022-07-22 by ssieb