esphome/issues #3295: addressable_flicker not accepting brightness change

ezhick, 2022-05-21:
### The problem

When I try to set the brightness or color_brightness for the LED strip it works with no affect, with any othe addressable affects, but does not work with addressable_flicker. The brightness drops down to 20-30% as soon as the effect is enabled and does not change in any way. 

If the brightness is reset to 0 after 100% (see code below, after 100% the encoder and the brightness are reset to 0) the LED shall not turn on until 75% of brightness and the color of the light will be 100% blue, not the one set in the code. After a couple of changes the light will get the right colour.

ALL other effects work fine.

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2022.5.5

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

light

### Example YAML snippet

```yaml
sensor:
  - platform: rotary_encoder
    name: "Left cooker"
    pin_a: D1
    pin_b: D2
    #min_value: 20
    id: left_knob
    resolution: 4
    on_clockwise:
      then:
        - if:
            condition:
              light.is_off: kitchen_light
            then:
              - light.turn_on: 
                  id: kitchen_light
                  red: 81%
                  green: 75%
                  blue: 100%
                  brightness: 10%
                  effect: "Rainbow"
              - sensor.rotary_encoder.set_value:
                  id: left_knob
                  value: 10
            else:
              - light.turn_on:
                  id: kitchen_light
                  brightness: !lambda |-
                    return id(left_knob).state / 100.0;
    on_anticlockwise:
      then:
        - if:
            condition:
              light.is_off: kitchen_light
            then:
              - light.turn_on: 
                  id: kitchen_light
                  red: 81%
                  green: 75%
                  blue: 100%
                  brightness: 10%
                  effect: "Rainbow"
              - sensor.rotary_encoder.set_value:
                  id: left_knob
                  value: 10
            else:
              - light.turn_on:
                  id: kitchen_light
                  brightness: !lambda |-
                    return id(left_knob).state / 100.0;
    on_value_range:
      above: 100
      then:
      - light.turn_on:
          id: kitchen_light
          brightness: 0%
      - sensor.rotary_encoder.set_value:
          id: left_knob
          value: 0
light:
  - platform: neopixelbus
    type: GRB
    variant: WS2812x
    pin: GPIO03
    method:
      type: esp8266_dma
    num_leds: 18
    name: "Kitchen lights"
    id: kitchen_light
    default_transition_length: 0s
    restore_mode: ALWAYS_OFF
    effects:
      - addressable_flicker:
          name: "Flicker"
          update_interval: 16ms
          intensity: 15%
```

ezhick, 2022-06-15:
Looks like nobody noticed the issue...

ssieb, 2022-10-16:
The effects don't necessarily take into account the light settings.  They can use their own settings for color and brightness.
I don't understand what you mean in your second paragraph.

ezhick, 2022-11-01:
> The effects don't necessarily take into account the light settings. They can use their own settings for color and brightness. I don't understand what you mean in your second paragraph.

But it does accepts in all other effects - the brightness changes as expected. And Flicker (as other effects I tried) does not have it's own settings, otherwise I would have used them. 

In the second paragraph I mean that if I set the brightness to 0 the light goes down and will not light up if the brightness is set les then to 75%.