esphome/issues #3296: Frame: Invalid when receiving sync'ed WLED color

mrand, 2022-05-21:
### The problem

After seeing that #2256 should allow RGB/RGBW bulbs to sync to WLED, I have attempted to enable that in my config, but couldn't get it to work.  After turning the light partition on and selecting WLED effect, I couldn't get anything useful happening.  After turning on VERY_VERBOSE logging, I see this error when I click on a color over on my WLED install that is set to transmit SYNC:

````yaml
[16:28:24][D][wled_light_effect:071]: Frame: Invalid (size=39, first=0x00).
````

### Which version of ESPHome has the issue?

2022.05

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.05

### What platform are you using?

ESP8266

### Board

RGBW bulb

### Component causing the issue

light

### Example YAML snippet

```yaml
`
wled:

light:
  - platform: rgbw
    id: ${devicename}_light
    name: "RGBW 3"
    default_transition_length: 0s
    red: output_red
    green: output_green
    blue: output_blue
    white: cw

  - platform: partition
    name: "RGBW 3 Partition"
    segments:
      - single_light_id: ${devicename}_light
    effects:
      - wled:
`
```

### Anything in the logs that might be useful for us?

```txt
VERY_VERBOSE output of turning on the light, and then selecting a color on a WLED that is configured to transmit/sync:

`
<< Plenty of log above this, but didn't see anything useful... this is where I turned the partition on, and then select the WLED effect >>
[16:27:52][VV][scheduler:195]: Running interval '' with interval=10000 last_execution=104365 (now=114372)
[16:27:53][VV][api.service:432]: on_light_command_request: LightCommandRequest {
  key: 881916492
  has_state: YES
  state: YES
  has_brightness: NO
  brightness: 0
  has_color_mode: YES
  color_mode: COLOR_MODE_RGB_WHITE
  has_color_brightness: NO
  color_brightness: 0
  has_rgb: NO
  red: 0
  green: 0
  blue: 0
  has_white: NO
  white: 0
  has_color_temperature: NO
  color_temperature: 0
  has_cold_white: NO
  cold_white: 0
  has_warm_white: NO
  warm_white: 0
  has_transition_l
[16:27:53][D][light:035]: 'RGBW 3 Partition' Setting:
[16:27:53][D][light:046]:   State: ON
[16:27:53][D][light:084]:   Transition length: 1.0s
[16:27:53][V][json:031]: Attempting to allocate 512 bytes for JSON serialization
[16:27:53][V][json:051]: Size after shrink 204 bytes
[16:27:53][VV][api.service:122]: send_light_state_response: LightStateResponse {
  key: 881916492
  state: YES
  brightness: 1
  color_mode: COLOR_MODE_RGB_WHITE
  color_brightness: 1
  red: 1
  green: 1
  blue: 1
  white: 1
  color_temperature: 0
  cold_white: 1
  warm_white: 1
  effect: 'None'
}
[16:27:53][V][component:199]: Component api took a long time for an operation (0.09 s).
[16:27:53][V][component:200]: Components should block for at most 20-30ms.
[16:27:53][VV][scheduler:195]: Running interval '' with interval=5000 last_execution=111123 (now=116127)
[16:27:53][VV][light.addressable:015]: Addressable Light 'RGBW 3 Partition' (effect_active=NO)
[16:27:53][VV][light.addressable:018]:   [ 0] Color: R=189 G=189 B=189 W=189
[16:27:53][VV][light.addressable:021]:  
[16:27:56][VV][api.service:432]: on_light_command_request: LightCommandRequest {
  key: 881916492
  has_state: YES
  state: YES
  has_brightness: NO
  brightness: 0
  has_color_mode: YES
  color_mode: COLOR_MODE_RGB_WHITE
  has_color_brightness: NO
  color_brightness: 0
  has_rgb: NO
  red: 0
  green: 0
  blue: 0
  has_white: NO
  white: 0
  has_color_temperature: NO
  color_temperature: 0
  has_cold_white: NO
  cold_white: 0
  has_warm_white: NO
  warm_white: 0
  has_transition_l
[16:27:56][D][light:035]: 'RGBW 3 Partition' Setting:
[16:27:56][D][light:108]:   Effect: 'WLED'
[16:27:56][V][json:031]: Attempting to allocate 512 bytes for JSON serialization
[16:27:56][V][json:051]: Size after shrink 204 bytes
[16:27:56][VV][api.service:122]: send_light_state_response: LightStateResponse {
  key: 881916492
  state: YES
  brightness: 1
  color_mode: COLOR_MODE_RGB_WHITE
  color_brightness: 1
  red: 1
  green: 1
  blue: 1
  white: 1
  color_temperature: 0
  cold_white: 1
  warm_white: 1
  effect: 'WLED'
}
[16:27:56][V][component:199]: Component api took a long time for an operation (0.08 s).
[16:27:56][V][component:200]: Components should block for at most 20-30ms.
[16:27:57][VV][api.service:355]: on_ping_request: PingRequest {}
[16:27:57][VV][api.service:043]: send_ping_response: PingResponse {}
[16:27:58][VV][scheduler:195]: Running interval '' with interval=5000 last_execution=116123 (now=121124)
[16:27:58][VV][light.addressable:015]: Addressable Light 'RGBW 3 Partition' (effect_active=YES)
[16:27:58][VV][light.addressable:018]:   [ 0] Color: R=  0 G=  0 B=  0 W=  0
[16:27:58][VV][light.addressable:021]:  
[16:27:59][VV][api.service:355]: on_ping_request: PingRequest {}
[16:27:59][VV][api.service:043]: send_ping_response: PingResponse {}
[16:28:02][VV][scheduler:195]: Running interval '' with interval=10000 last_execution=114365 (now=124365)
[16:28:03][VV][scheduler:195]: Running interval '' with interval=5000 last_execution=121123 (now=126123)
[16:28:03][VV][light.addressable:015]: Addressable Light 'RGBW 3 Partition' (effect_active=YES)
[16:28:03][VV][light.addressable:018]:   [ 0] Color: R=  0 G=  0 B=  0 W=  0
[16:28:03][VV][light.addressable:021]:  
[16:28:08][VV][scheduler:195]: Running interval '' with interval=5000 last_execution=126123 (now=131123)
[16:28:08][VV][light.addressable:015]: Addressable Light 'RGBW 3 Partition' (effect_active=YES)
[16:28:08][VV][light.addressable:018]:   [ 0] Color: R=  0 G=  0 B=  0 W=  0
[16:28:08][VV][light.addressable:021]:  
[16:28:12][VV][scheduler:195]: Running interval '' with interval=10000 last_execution=124365 (now=134365)
[16:28:12][VV][api.service:355]: on_ping_request: PingRequest {}
[16:28:12][VV][api.service:043]: send_ping_response: PingResponse {}
[16:28:13][VV][scheduler:195]: Running interval '' with interval=5000 last_execution=131123 (now=136126)
[16:28:13][VV][light.addressable:015]: Addressable Light 'RGBW 3 Partition' (effect_active=YES)
[16:28:13][VV][light.addressable:018]:   [ 0] Color: R=  0 G=  0 B=  0 W=  0
[16:28:13][VV][light.addressable:021]:  
[16:28:14][VV][api.service:355]: on_ping_request: PingRequest {}
[16:28:14][VV][api.service:043]: send_ping_response: PingResponse {}
[16:28:18][VV][scheduler:195]: Running interval '' with interval=5000 last_execution=136123 (now=141123)
[16:28:18][VV][light.addressable:015]: Addressable Light 'RGBW 3 Partition' (effect_active=YES)
[16:28:18][VV][light.addressable:018]:   [ 0] Color: R=  0 G=  0 B=  0 W=  0
[16:28:18][VV][light.addressable:021]:  
[16:28:22][VV][scheduler:195]: Running interval '' with interval=10000 last_execution=134365 (now=144365)
[16:28:23][VV][scheduler:195]: Running interval '' with interval=5000 last_execution=141123 (now=146123)
[16:28:23][VV][light.addressable:015]: Addressable Light 'RGBW 3 Partition' (effect_active=YES)
[16:28:23][VV][light.addressable:018]:   [ 0] Color: R=  0 G=  0 B=  0 W=  0
[16:28:23][VV][light.addressable:021]:  
[16:28:24][D][wled_light_effect:071]: Frame: Invalid (size=39, first=0x00).

<< this is where I selected a different color on my WLED device.  Log message resumed with the continuous debug output >>
`
```

Drun555, 2022-06-30:
RGBCT, RGBWW lights. Trying to wrap it up with LedFX. Even if pixel count setted up to 1, the error still pops up.

`18:52:38 | [D] | [wled_light_effect:071] | Frame: Invalid (size=3, first=0x00).`

digitydogs, 2022-09-17:
Bug is still present and affects native WLED issues Sync commands.  Tested using several combinations of pins used, segments, and number of LEDs. Also tried using several different ports.  All result in the same error message in the ESPHome logs, depending on the WLED device sending the sync data, the size variable varies. 
[00:15:11][D][wled_light_effect:071]: Frame: Invalid (size=489, first=0x00).
[01:18:32][D][wled_light_effect:071]: Frame: Invalid (size=39, first=0x00).

E131 control works great for doing controlled displays via xlights, but for day to day spur of the moment effects being able to sync to a single WLED device and access the inbuilt effects is invaluable.

Alternately I would love to see ESPHome import the various effects and color pallets provided by WLED for native use, and add native sync channels for lighting but that is more of a request than an issue.

digitydogs, 2022-09-17:
I admittedly know next to nothing about code, so this is a literal shot in the dark... 

looking over code revisions I notice that just before line 71 of wled_light_effects.cpp (the line throwing the error) that this section of code has moved the std::vector outside of the while statement at some point in time since the integration was knows to work.   Could this be the reason for the invalid frame errors? 

Current Code:
std::vector<uint8_t> payload;
  while (uint16_t packet_size = udp_->parsePacket()) {
    payload.resize(packet_size);

Old Code:
while (uint16_t packet_size = udp_->parsePacket()) {
    std::vector<uint8_t> payload;  
    payload.resize(packet_size);

lux4rd0, 2023-09-28:
Trying to sync with my WLED HyperCube - running into the same:

`[14:49:05][D][wled_light_effect:071]: Frame: Invalid (size=104, first=0x00).`