esphome/issues #3297: A4988 frequency slow

camnewnham, 2022-05-22:
### The problem

I'm setting up a stepper component and I have run into an issue where the motor won't turn any faster, but isn't having any torque issues and is otherwise running as normal.

The motor is configured at 1/16 microsteps, and won't turn faster than 1.5RPM. At 200 steps per rev this means the pin state is toggled at around 4800Hz. I've tested with multiple drivers (DRV8825, A4988 and TMC2100) all to the same result.

Increasing `stepper.set_speed` beyond this point doesn't have any effect. It seems that the loop speed is the limiting factor. 

ESPHome code:
```
script:
  - id: stop_stepper
    then: 
      - stepper.report_position:
          id: my_stepper
          position: 0
      - stepper.set_target:
          id: my_stepper
          target: 0
      - output.turn_off: led1
      
  - id: run_stepper
    then:
      - output.turn_on: led1
      - script.execute: stop_stepper
      - stepper.set_speed:
          id: my_stepper
          speed: 10000 steps/s
      - stepper.set_target:
          id: my_stepper
          target: -10000000
      - wait_until:
          condition:
            binary_sensor.is_on: endstop_0
          timeout: 30s
      - script.execute: stop_stepper

stepper:
  - platform: a4988
    id: my_stepper
    sleep_pin:
      number: D1
      inverted: true
    step_pin:
      number: D2
    dir_pin: 
      number: D3
      inverted: false
    max_speed: 10000 steps/s
    acceleration: inf
    deceleration: inf
```

For comparison, this loop running from an Arduino IDE deployment does manage to increase the frequency far beyond this:
```
void loop()
{
  for (int i=0;i<STEPS_PER_REVOLUTION*10; i++)
  {
      digitalWrite(STEP,HIGH);
      delayMicroseconds(100);
      digitalWrite(STEP,LOW);
      delayMicroseconds(100);
  }
}
```

Is there anything I can do to increase the loop frequency? Or could multiple steps be performed in the loop?

https://github.com/esphome/esphome-core/blob/254d059945932adb64f5cbe8f4928ea19b4f6f00/src/esphome/stepper/a4988.cpp#L32

Also: I have noticed that the stepper can be much choppier when controlled from ESPHome vs. the above Arduino IDE sketch - I assume this is also a loop timing issue?

### Which version of ESPHome has the issue?

2022.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.5

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

stepper

WeekendWarrior1, 2022-05-25:
How many RPMs do you see using your Arduino Sketch?
What stepper motor are you using?

A simple solution (which doesn't address the root cause of this issue) would be reducing your microsteps (ie. 1/8 should double your RPM).

camnewnham, 2022-05-25:
Oh apologies; I meant 1.5 revolutions per _second_ using ESPHome (90RPM) at 1/16 steps. Updated above.
The Arduino sketch will run around 3x this speed before the motor can't keep up. I can get some more exact numbers if useful.

The stepper is from an ANet A8 3D printer, specs here: https://caggius.wordpress.com/anet-a8-hardware-specifications/

Reducing microsteps is certainly a way to increase speed, but comes with noise (microPlyer on TMC drivers being an exception).

Would I expect to see any difference in the loop timings or consistency switching from an ESP82666 to an ESP32?

surfingbytes, 2022-10-23:
Why is this closed, the issue is still there, it's unbeliavably slow. Can it be improved please?

ssieb, 2022-10-23:
When you use the Arduino sketch, you take over the CPU for only your task.  You can't do that with esphome.  e.g. try running wifi also in your sketch.

surfingbytes, 2022-10-24:
That makes sense @ssieb . However, with 2 cores on ESP32, couldn't wifi + everything firmware has to do on one core, and components defined in yaml (stepper motor in this case) on the other one? Or better, having property on each component, which sets the core on which it runs?

surfingbytes, 2022-10-24:
Another possibility - add a settings (default false), and when it's turned on, wifi handling is skipped (or wifi off) when HighFrequencyLoopRequester is started.

swifty99, 2022-12-09:
Hi,
I also struggle with the stepper component. It is super slow and interfered even by the logger!
Stepping needs to run in hardware, not software.
There is hope: https://github.com/gin66/FastAccelStepper
This project has a nice dedicated stepping library. I do not not know internal ESPhome structure well enough to integrate this. I could definitely help. Let's give stepping component a use!

camnewnham, 2022-12-11:
FWIW, I ended up moving away from ESPHome and instead used PlatformIO and MQTT to add to home assistant.
This also meant it was easier to integrate the [TMCStepper ](https://github.com/teemuatlut/TMCStepper) library and use UART - essentially just setting the speed and letting the controller handle the steps (which is fine for a cover with endstops).

ssieb, 2022-12-12:
In esphome, the stepper component will have to be split into ESP8266 and ESP32 versions to take advantage of hardware acceleration.

swifty99, 2022-12-12:
> In esphome, the stepper component will have to be split into ESP8266 and ESP32 versions to take advantage of hardware acceleration.

true. 
Super smooth stepping would be worth it. 
I use TMC Steppers with additional SPI features (super quiet stepping). For that ESP8266 has not enough pins anyway.

albasee, 2023-10-14:
> FWIW, I ended up moving away from ESPHome and instead used PlatformIO and MQTT to add to home assistant. This also meant it was easier to integrate the [TMCStepper [...]

Hmm, I'm fighting with the same problem on ESPHome. Do you have an example of your solution with PlatformIO and MQTT?

camnewnham, 2023-10-15:
> Do you have an example of your solution with PlatformIO and MQTT?

[Here you go.](https://github.com/camnewnham/motorboard/tree/c6819f260d3066b71be85ddc59d46a94409a6bc7)

iqubik, 2024-01-17:
Really need new ESP32 realisation of stepper hard lib. Not smooth moving now

lual, 2024-09-07:
I wasted many hours trying to figure out why my stepper moves so slow  (my esp32 board: az-delivery-devkit-v4).

Because I don't have an oscilloscope or anything similar, I wired the STEP pin to another GPIO and a “Pulse Counter Sensor” to count the step pulses. 
I determined the following values: 
* with “max_speed: 10 steps/s” the counter reads “539.46 pulses/min” so that's about 9 steps/s. That fits roughly.
* with “max_speed: 200 steps/s” the counter reads “3701.49 pulses/min” that is about 62 steps/s.

**62 steps/s is the maximum**. None of my attempts made it faster. So in “Full Step Mode” we have a limit of at least 3s for one revolution. (Even slower with microsteps, of course)

The documentation should at least warn about this.