esphome/issues #3298: Preferences are stored only once on ESP32 with ESP-IDF

alengwenus, 2022-05-22:
### The problem

On an ESP32 with ESP-IDF, when storing values to flash using `restore_value: true` parameter, the values are stored only the first time they are written to flash.
This happens with any component which writes values to flash (number, switch, select, globals).

In the logs the write process is correctly documented for the first time of writing:
```
[07:12:04][D][number:054]: 'Test number' - Setting number value
[07:12:04][D][number:113]:   New number value: 10.000000
[07:12:04][D][number:012]: 'Test number': Sending state 10.000000
[07:12:04][D][esp32.preferences:114]: Saving preferences to flash...
```

However, any subsequent attempt to store the value results in not storing the values:
```
[07:12:29][D][number:054]: 'Test number' - Setting number value
[07:12:29][D][number:113]:   New number value: 20.000000
[07:12:29][D][number:012]: 'Test number': Sending state 20.000000
[07:12:29][D][esp32.preferences:114]: Saving preferences to flash...
[07:12:29][D][esp32.preferences:131]: NVS data not changed skipping -113489577  len=4
```

### Which version of ESPHome has the issue?

2022.5.0

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2022.5.5

### What platform are you using?

ESP32-IDF

### Board

esp32dev

### Component causing the issue

number, switch, select, global

### Example YAML snippet

```yaml
esphome:
  name: test

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended

preferences:
  flash_write_interval: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.2.69
    gateway: 192.168.2.1
    subnet: 255.255.255.0

logger:

ota:

api:

number:
  - platform: template
    name: "Test number"
    min_value: 0
    max_value: 100
    step: 1
    initial_value: 0
    optimistic: true
    restore_value: true
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration test.yaml...
INFO Starting log output from 192.168.2.69 using esphome API
INFO Successfully connected to 192.168.2.69
[07:11:26][I][app:102]: ESPHome version 2022.5.0 compiled on May 22 2022, 07:08:57
[07:11:26][C][wifi:491]: WiFi:
[07:11:26][C][wifi:353]:   Local MAC: ***
[07:11:26][C][wifi:354]:   SSID: '***'
[07:11:26][C][wifi:355]:   IP Address: 192.168.2.69
[07:11:26][C][wifi:357]:   BSSID: 2C:91:AB:DC:FA:48
[07:11:26][C][wifi:358]:   Hostname: 'test'
[07:11:26][C][wifi:360]:   Signal strength: -72 dB ▂▄▆█
[07:11:26][C][wifi:364]:   Channel: 6
[07:11:26][C][wifi:365]:   Subnet: 255.255.255.0
[07:11:26][C][wifi:366]:   Gateway: 192.168.2.1
[07:11:26][C][wifi:367]:   DNS1: 0.0.0.0
[07:11:26][C][wifi:368]:   DNS2: 0.0.0.0
[07:11:26][C][logger:274]: Logger:
[07:11:26][C][logger:275]:   Level: DEBUG
[07:11:26][C][logger:276]:   Log Baud Rate: 115200
[07:11:26][C][logger:277]:   Hardware UART: UART0
[07:11:26][C][template.number:050]: Template Number 'Test number'
[07:11:26][C][template.number:051]:   Optimistic: YES
[07:11:26][C][template.number:052]:   Update Interval: 60.0s
[07:11:26][C][mdns:084]: mDNS:
[07:11:26][C][mdns:085]:   Hostname: test
[07:11:26][C][ota:085]: Over-The-Air Updates:
[07:11:26][C][ota:086]:   Address: 192.168.2.69:3232
[07:11:26][C][api:138]: API Server:
[07:11:26][C][api:139]:   Address: 192.168.2.69:6053
[07:11:26][C][api:143]:   Using noise encryption: NO
[07:12:04][D][number:054]: 'Test number' - Setting number value
[07:12:04][D][number:113]:   New number value: 10.000000
[07:12:04][D][number:012]: 'Test number': Sending state 10.000000
[07:12:04][D][esp32.preferences:114]: Saving preferences to flash...
[07:12:29][D][number:054]: 'Test number' - Setting number value
[07:12:29][D][number:113]:   New number value: 20.000000
[07:12:29][D][number:012]: 'Test number': Sending state 20.000000
[07:12:29][D][esp32.preferences:114]: Saving preferences to flash...
[07:12:29][D][esp32.preferences:131]: NVS data not changed skipping -113489577  len=4
```

martgras, 2022-05-22:
Can you please test this fix:

```
external_components:
  - source: github://martgras/esphome@nvs-fix
    components: [ esp32]
```

alengwenus, 2022-05-22:
> Can you please test this fix:
> 
> ```
> external_components:
>   - source: github://martgras/esphome@nvs-fix
>     components: [ esp32]
> ```

Looks good. This seems to fix it. 👍

closed 2022-05-23 by jesserockz

jangrewe, 2022-05-23:
man, am i happy to find this... i was so close to throwing my ESPs out the window! ;-)