esphome/issues #3303: Bedjet: Error: Failed to call service climate/set preset mode.

mwolter805, 2022-05-22:
### The problem

Within the Home Assistant frontend, when selecting a preset mode via the climate entity created by the integration, the following error is received.

Interestingly, the preset is sent and applied to the BedJet. So it works, but there is a toast error HA. 

Also, there are no errors related to this in the Home Assistant logs.  

[image]
[image]

```
Failed to call service climate/set preset mode. string value is None for dictionary value @ data['preset_mode']
```
Calling the service via developer tools does not have the same error toast.
```
service: climate.set_preset_mode
target:
  entity_id: climate.bed_esp32_matts_bedjet
data:
  preset_mode: "M1"
```

### Which version of ESPHome has the issue?

2022.5.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.4.6

### What platform are you using?

ESP32

### Board

Olimex ESP32-POE

### Component causing the issue

BedJet

### Example YAML snippet

```yaml
substitutions:
  name: bed-esp32
  friendly_name: Bed-ESP32
  ip_address: 192.168.6.12

esphome:
  name: $name
  platform: ESP32
  board: esp32-poe

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12
  use_address: $ip_address

mdns:
  disabled: true
  
web_server:
  port: 80

# Enable Home Assistant API
api:
  password: !secret api_password
  reboot_timeout: 0s

# Enable over-the-air flashing
ota:
  password: !secret ota_password
  on_begin:
    then:
      - logger.log: "Disconnecting BedJet for OTA update..."
      - switch.turn_off: matts_bedjet_enable

time:
  - platform: homeassistant
    id: time_ha
    
logger:
  baud_rate: 0 # disable UART logging
  level: error  # normal level for logger
#  level: debug # only use this level when debugging an issue
#  level: VERY_VERBOSE # only use this level if debug is not producing enough info to debug an issue
  logs:
    esp32_ble_tracker: error # normal level for this function
#    esp32_ble_tracker: debug # enable when discovering BLE devices

# Include the following until the include is added to bedjet, most likely in version 2022.6
external_components:
  - source: github://pr#3490
    components: [bedjet]

# Uptime Sensor    
  - platform: uptime
    name: "${friendly_name} Uptime"
    id: utime

esp32_ble_tracker:

ble_client:

# Matts BedJet - Address
  - mac_address: 11:11:11:11:11:11
    id: matts_bedjet_ble

climate:

# Matts BedJet - Climate Entity
  - platform: bedjet
    id: matts_bedjet
    name: $friendly_name Matts BedJet
    ble_client_id: matts_bedjet_ble
    time_id: time_ha
    icon: mdi:hvac
    visual:
      min_temperature: 70 °F
      max_temperature: 85 °F

switch:

# Matts BedJet - Enable BLE
  - platform: ble_client
    ble_client_id: matts_bedjet_ble
    id: matts_bedjet_enable
    name: $friendly_name Matts BedJet Enable BLE
    icon: mdi:bluetooth-connect
```

### Anything in the logs that might be useful for us?

```txt
There are no errors in home assistant or on the ESP.
```

jhansche, 2022-05-27:
I have seen this myself on occasion, but like you mentioned, it seems to be happening only in the climate UI. Feels like a bug in HA.

It's possible this might be happening specifically because these are "custom preset" entries. So you know if the same happens with a standard preset, like Boost?

mwolter805, 2022-05-27:
There is no toast error message when the BedJet is off and the Boost preset is selected. It appears the issue might be related to displaying the proper status. Below is a sequence of events 

Off -> Boost -> M1
1. The BedJet is Off and Preset: Boost is selected. Immediately the status for the entity changes to Operation: Heat, Preset: Boost. 
2. I then change the preset to M1. There is no status change immediately. Approx 20 seconds later Operation changes to Heat, Preset changed to blank and the same toast error message in the original report is displayed. 

Off -> M1
1. The BedJet is Off and Preset: M1 is selected. There is no immediate change to the status, Operation is still Off, Preset is blank, and the toast error message appears.  Approx 30 seconds later the status changes to Operation: Heat and Preset is blank. 

Off -> Heat -> M1
1. The BedJet is Off and Operation: Heat is selected. Immediately the status changes to Operation: Heat, Preset: Blank.
2. Waited 30 seconds then selected Preset: M1. The status didn't change (Operation: Heat, Preset: Blank) and the toast error message appeared.

Thanks and let me know if you need any more info.

jhansche, 2022-05-27:
There are a couple things to note:

First is the up to ~30 second delay in status updates. This was mentioned in the HA community, and I have some ideas for how we can address that.

Second is understanding the relationships between mode and preset. In most climate devices, a preset is a kind of extra mode that's applied on top of an existing operation: you turn on your A/C unit, and then press the "Eco" button. "Eco" preset doesn't mean anything if the unit is off.

However, in the BedJet, "Boost/TURBO", and the M1-3 presets, actually turn the device on, _and_ might set it to variety of actual operational modes (modes that could actually change throughout the night if it was set to run a Biorhythm program). I think it's possible that this might be one of the side effects we're running into. Maybe HA UI assumes you can't set a preset if the state is off?

Related to that: I do know this is also a problem if you try to use a preset in a scene, without also changing the state... HA assumes that any entity listed in a scene MUST be setting the state (and can optionally modify a different attribute as well). So you can't just have a scene that sets `preset_mode: eco`, you have to also set `state: cool` or `heat`. As a result, the only way to get a scene that runs a preset, is to also set the mode. HA will raise an error in "Check Configuration" if you try to only set a preset in a scene. And I think that's what's happening.

Unfortunately we don't have a way to create "custom modes" (like we can for custom fan modes or custom presets), which is really what those are.

I don't know the inner workings of the HA climate devices/services, which is actually where all this information belongs. I've only worked with the ESPHome side of the implementation.

mwolter805, 2022-05-28:
Thanks for the explanation. The only reason I use the M1 preset is to set an extended Auto Off time when using the heat mode. Other than that the preset is not needed. 

An alternative is adding the ability to set the Auto Off time and maybe even view the current Auto Off time. I'll add an FR for this.

jhansche, 2022-05-28:
Another person just asked for displaying the countdown. While we can't add that as a stock feature of this (a climate) component, we can add a way for you to access the data, which you can then mirror over to a template sensor and use it in HA.

As for the duration of heat, I opened another issue myself because I want to revert that change that was merged a few weeks ago. The Heat mode only lasts 15-30 min on mine (or up to 4 hours), and that's not long enough in my opinion. Instead I want to change it back to what it was before, so that heat mode sets the bedjet Ext Heat mode, which lasts for 12 hours.

Would that solve your use case?

mwolter805, 2022-05-28:
It would be great to view and set the count down, if possible.

In my case, when pressed from the controller, EXT HEAT sets the timer to 10 hours and HEAT sets it to 30 minutes. Would prefer the timer be set to 12 hours. 

If we had the ability to set the timer, I think it would be the best of both worlds. All the modes and presets available in the integration would match the BedJet’s built-in functions, and we could set the mode with a service call and set the timer with a second service call. 

Thank you

mwolter805, 2022-05-28:
Closing this issue and copying the last comment https://github.com/esphome/issues/issues/3314 so it can be seen and discussed by more users.

closed 2022-05-28 by mwolter805