esphome/issues #3304: Max7219 Digit 255 character limit?

gaussey, 2022-05-22:
### The problem

I’m using MQTT to populate the display but it’s cropping everything after 254 characters. Is this a limitation with the display or ESPHome and is there any way are to increase this limit?

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.1

### What platform are you using?

ESP8266

### Board

NodeMCU

### Component causing the issue

Display

gaussey, 2022-05-23:
The relevant bits :     
lambda: |-
      it.printf(0, 0, id(digit_font), "%s",id(newsdata).state.c_str());

text_sensor:

  - platform: mqtt_subscribe
    name: "matrixnews"
    id: newsdata
    topic: matrix/all

But even just doing a direct PrintF of > 255 characters is cropped to 254.

jesserockz, 2022-05-23:
Try this: `it.print(0, 0, id(digit_font), id(newsdata).state.c_str());`

`print` not `printf`, the printf has a character limit of 256 for the result due to a fixed buffer size.

gaussey, 2022-05-23:
Holy crap, that did it. I was under the assumption that I had to use PrintF because the text was formatted with the text ID, not that I knew PrintF had a character limit on it anyway. Thank you!