esphome/issues #3305: Docker fails to download/install toolchain-xtensa, tries many times

skandragon, 2022-05-22:
### The problem

Unable to compile:

```docker-compose run esphome compile shopfans.yaml```

```
Tool Manager: Installing platformio/toolchain-xtensa @ ~2.100300.0
Unpacking  [####################################]  100%
Tool Manager: Warning! Package Mirror: [('/config/.esphome/platformio/cache/tmp/pkg-installing-1ofgds5j/lib/libcc1.so.0', '/config/.esphome/platformio/packages/toolchain-xtensa/lib/libcc1.so.0', '[Errno 2] No such file or directory')]
Tool Manager: Looking for another mirror...
Unpacking  [####################################]  100%
Tool Manager: Warning! Package Mirror: [('/config/.esphome/platformio/cache/tmp/pkg-installing-hwq9fqvk/lib/libcc1.so.0', '/config/.esphome/platformio/packages/toolchain-xtensa/lib/libcc1.so.0', '[Errno 2] No such file or directory')]
Tool Manager: Looking for another mirror...
```

Loops at least 5 times...

### Which version of ESPHome has the issue?

2022.5.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N.A.

### What platform are you using?

ESP8266

### Board

sonoff_th (esp8266)

### Component causing the issue

platformio/toolchain-xtensa

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration shopfans.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing shopfan (board: sonoff_th; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-xtensa @ ~2.100300.0
Unpacking  [####################################]  100%
Tool Manager: Warning! Package Mirror: [('/config/.esphome/platformio/cache/tmp/pkg-installing-1ofgds5j/lib/libcc1.so.0', '/config/.esphome/platformio/packages/toolchain-xtensa/lib/libcc1.so.0', '[Errno 2] No such file or directory')]
Tool Manager: Looking for another mirror...
Unpacking  [####################################]  100%
Tool Manager: Warning! Package Mirror: [('/config/.esphome/platformio/cache/tmp/pkg-installing-hwq9fqvk/lib/libcc1.so.0', '/config/.esphome/platformio/packages/toolchain-xtensa/lib/libcc1.so.0', '[Errno 2] No such file or directory')]
Tool Manager: Looking for another mirror...
```
```

OttoWinter, 2022-06-13:
What host system do you use on top of docker? What filesystem (ext4, ntfs, ...) is the `config` dir mounted in?

skandragon, 2022-09-20:
I believe this was amd64, using ext4 filesystem.