esphome/issues #3307: BedJet - Clock will not update

mjoshd, 2022-05-23:
### The problem

Unable to set the clock on the BedJet via ESPHome/Home Assistant.

I have set up the [Home Assistant Time Source](https://esphome.io/components/time.html#home-assistant-time-source):
```
time:
  - platform: homeassistant
    id: homeassistant_time
```
and I'm using it for the [time_id](https://esphome.io/components/climate/bedjet.html#configuration-variables):
```
climate:
  - platform: bedjet
    id: climate_bedjet_1
    name: ${friendly_name}
    ble_client_id: ble_bedjet_1
    time_id: homeassistant_time
```

I'm not sure if the time-setting functionality is supposed to be automatic or not but it doesn't seem to be working for me.

I tried making a lambda similar to the firmware upgrade one but it won't compile and says that `send_local_time_()` is a "protected member function".

Could this possibly be related to the firmware update BedJet sent out ~1-2 months ago?

BedJet info (from app):
```
App         V1.0.4 (6)
BedJet 3    V1.20 (140)   
Remote      V1.20 (182)
```

### Which version of ESPHome has the issue?

2022.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.5

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

bedjet

### Example YAML snippet

```yaml
# Full yaml (sensitive info redacted)

# https://esphome.io/guides/configuration-types.html#substitutions
substitutions:
  device_name: REDACTED
  friendly_name: REDACTED
  icon: "mdi:bed-outline"

# https://esphome.io/components/esphome.html
esphome:
  name: ${device_name}

# https://esphome.io/devices/esp32.html
esp32:
  board: nodemcu-32s
  framework:
    type: arduino

# https://esphome.io/components/logger.html
logger:
  baud_rate: 0

# https://esphome.io/components/api.html
api:
  password: REDACTED
  encryption:
    key: REDACTED

# https://esphome.io/components/ota.html
ota:
  password: !secret esphome_ota_password
  on_begin:
    then:
      - logger.log: "Disconnecting clients for OTA update..."
      - switch.turn_off: bedjet_1_ble_connection

# https://esphome.io/components/wifi.html
wifi:
  ssid: REDACTED
  password: REDACTED

##############################################################
##############################################################
##############################################################

# https://esphome.io/components/ble_client.html
ble_client:
  - mac_address: REDACTED
    id: ble_bedjet_1

# https://esphome.io/components/button/restart.html
# https://esphome.io/components/climate/bedjet.html#lambda-calls
button:
  - platform: restart
    name: ${friendly_name} Restart
  - platform: template
    name: "Check BedJet Firmware"
    on_press:
      then:
      - lambda: |-
          id(climate_bedjet_1).upgrade_firmware();

# https://esphome.io/components/climate/bedjet.html
# https://esphome.io/components/climate/index.html#base-climate-configuration
climate:
  - platform: bedjet
    id: climate_bedjet_1
    name: ${friendly_name}
    ble_client_id: ble_bedjet_1
    time_id: homeassistant_time
    # icon: ${icon}
    visual:
      min_temperature: 18.8889 °C
      max_temperature: 42.7778 °C

# https://esphome.io/components/esp32_ble_tracker.html
esp32_ble_tracker:
  scan_parameters:
    interval: 0s
    window: 0s
    # active: false

# https://esphome.io/components/switch/ble_client.html
# https://esphome.io/components/switch/restart.html
switch:
  - platform: ble_client
    ble_client_id: ble_bedjet_1
    name: ${friendly_name} BLE Connection
    id: bedjet_1_ble_connection

# https://esphome.io/components/time.html#home-assistant-time-source
time:
  - platform: homeassistant
    id: homeassistant_time
```

mjoshd, 2022-05-25:
The issue with setting the time appears to be solved in ESPHome 2022.5.1.

Unfortunately, since flashing the ESP with a new .bin compiled in ESPHome 2022.5.1 there are very frequent disconnects (~60s interval) and if I try to control the BedJet I get the same error shown in #3303.

[image]

closed 2022-05-25 by mjoshd

jhansche, 2022-05-27:
I've added a PR to make the clock sync method public, so that you can add a lambda for it. I also added a method that lets you set the clock to a specific time, which will work even if you don't have a Time component added.