esphome/issues #3308: BedJet - HA frontend takes a long time to update (~30s)

mjoshd, 2022-05-23:
### The problem

The BedJet accepts commands and responds right away but the HA front end takes some time to reflect changes (usually between 10-30 seconds).

BedJet info (from app):
```
App         V1.0.4 (6)
BedJet 3    V1.20 (140)   
Remote      V1.20 (182)
```

### Which version of ESPHome has the issue?

2022.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.5

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

bedjet

### Example YAML snippet

```yaml
# Full yaml (sensitive info redacted)

# https://esphome.io/guides/configuration-types.html#substitutions
substitutions:
  device_name: REDACTED
  friendly_name: REDACTED
  icon: "mdi:bed-outline"

# https://esphome.io/components/esphome.html
esphome:
  name: ${device_name}

# https://esphome.io/devices/esp32.html
esp32:
  board: nodemcu-32s
  framework:
    type: arduino

# https://esphome.io/components/logger.html
logger:
  baud_rate: 0

# https://esphome.io/components/api.html
api:
  password: REDACTED
  encryption:
    key: REDACTED

# https://esphome.io/components/ota.html
ota:
  password: REDACTED
  on_begin:
    then:
      - logger.log: "Disconnecting clients for OTA update..."
      - switch.turn_off: bedjet_1_ble_connection

# https://esphome.io/components/wifi.html
wifi:
  ssid: REDACTED
  password: REDACTED

##############################################################
##############################################################
##############################################################

# https://esphome.io/components/ble_client.html
ble_client:
  - mac_address: REDACTED
    id: ble_bedjet_1

# https://esphome.io/components/button/restart.html
# https://esphome.io/components/climate/bedjet.html#lambda-calls
button:
  - platform: restart
    name: ${friendly_name} Restart
  - platform: template
    name: "Check BedJet Firmware"
    on_press:
      then:
      - lambda: |-
          id(climate_bedjet_1).upgrade_firmware();

# https://esphome.io/components/climate/bedjet.html
# https://esphome.io/components/climate/index.html#base-climate-configuration
climate:
  - platform: bedjet
    id: climate_bedjet_1
    name: ${friendly_name}
    ble_client_id: ble_bedjet_1
    time_id: homeassistant_time
    # icon: ${icon}
    visual:
      min_temperature: 18.8889 °C
      max_temperature: 42.7778 °C

# https://esphome.io/components/esp32_ble_tracker.html
esp32_ble_tracker:
  scan_parameters:
    interval: 0s
    window: 0s
    # active: false

# https://esphome.io/components/switch/ble_client.html
# https://esphome.io/components/switch/restart.html
switch:
  - platform: ble_client
    ble_client_id: ble_bedjet_1
    name: ${friendly_name} BLE Connection
    id: bedjet_1_ble_connection

# https://esphome.io/components/time.html#home-assistant-time-source
time:
  - platform: homeassistant
    id: homeassistant_time
```

### Additional information

Creating issue [as requested](https://community.home-assistant.io/t/bedjet-integration/144822/24?u=mjoshd).

jhansche, 2022-05-27:
Copying notes from the HA community thread:

> The delay in updating is something I deal with as well. There’s a couple problems leading to this, the biggest one is that the esp32 doesn’t currently have a way to configure how frequently the BTLE notification gets triggered, which means by default it becomes a “fire hose” that receives new updates as quickly as possible, usually on the order of every 100-200ms.
>
> You’d think that means updates should happen more often and you’d be right. But I intentionally throttle the updates back to HA, in order to avoid overwhelming either the esp32 or HA. So even though it attempts to send the status update to HA as soon as the bedjet acknowledges the command, because of this rapid stream of status updates across a different service, that recent command tends to get “reverted” quickly, that revert gets sent to HA, and then the rapid stream of status notifications changes as well, but by then the reverted status was already sent, and now it’s in its “throttle” period and won’t update with the latest status until the next polling period (30sec by default, but can be configured by setting the polling interval).
>
> A better improvement would be to make it smarter about sending status updates caused by a recent command; something like: if we recently sent a temp change, wait until the next update is received that matches that temp, before sending the current status. Or similar for mode, etc. Right now it’s just a naive throttle.

Another way to handle this would be to remove the need for the throttle at all, by allowing the BLE Client to configure the notification rate. If we processed and sent every status notification, we would quickly overwhelm the ESP, and unnecessarily send a ton of status updates to HA.

Alternatively we could be smarter about recognizing if the current status is meaningfully different from the last sent status, and drop the ones that look uninteresting. The status notification includes things like a runtime counter for the current mode or biorhythm program, and that isn't really something we need to include - so simply comparing the current status object to previously sent status might not be good enough because the status content will be constantly changing.

I plan to look into better ways to manage the frequent status updates, in a way that keeps the HA status properly up to date without being overly chatty.

jhansche, 2022-06-02:
I'm about to open a PR that refactors the component into a Hub with child components (currently only Climate as a child). As part of that refactor, I've also been able to address the "throttle" issues, and I'm seeing much more responsive updates after sending a command.

Try it out, using [`external_components`](https://esphome.io/components/external_components.html) in your config. I'll update here once the PR is ready.

jhansche, 2022-06-03:
https://github.com/esphome/esphome/pull/3522 is the PR. You can use the external_components link above to test it.

Note that this is a breaking change and requires shifting the yaml config around (explained in the PR, visualized in the PR's test1.yaml changes, and also documented in the linked docs PR as well).

mjoshd, 2022-06-04:
I finally got a chanse to this set up today and it is really snappy! 
Thank you very much for looking into this 😺

jhansche, 2022-08-22:
The above [PR](https://github.com/esphome/esphome/pull/3522) is now merged and available as of [2022.8.0b1](https://github.com/esphome/esphome/pull/3705).

jhansche, 2022-10-19:
@mjoshd can this be closed?

kquinsland, 2022-10-19:
> can this be closed?

I think so. My HA interface has been _very responsive_ since this shipped. If i press a button on the remote, the HA ui reflects that in just a few seconds

closed 2022-10-20 by mjoshd