esphome/issues #3309: remote_transmitter: transmit_nec --> value must be at most 65535

moonraker46, 2022-05-23:
### The problem

I do migrate my ESP devices step by step from Tasmota to ESPHome.

One device has an IRsender to send commands to a cheap lampion fairy light https://www.amazon.de/gp/product/B07SBZPKW9/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

This worked so far with Tasmota (I got those IR Codes with an IR Receiver and the official remote).

But using the same commands with ESPHome leads to the issue that ESPHome tells me when compiling the config: value must be at most 65535.

I guess this is because the ESPHome NEC IR command wants to use a 16Bit command but I do have a 32bit command. 

How can this be fixed?

```
remote_transmitter:
  pin: 16
  carrier_duty_percent: 50%

switch:
  - platform: template
    name: "mode5"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x2FD09F6
        command: 0x40BF906F 
```

```
INFO Reading configuration /config/lichterkette.yaml...
Failed config

switch.template: [source /config/lichterkette.yaml:68]
  platform: template
name: mode5
  
  value must be at most 65535.
  turn_on_action: 
    remote_transmitter.transmit_nec: 
      
      value must be at most 65535.
      address: 50137590
      command: 1086296175
```

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

esp8285

### Component causing the issue

remote__transmitter

### Example YAML snippet

```yaml
remote_transmitter:
  pin: 16
  carrier_duty_percent: 50%

switch:
  - platform: template
    name: "mode5"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x2FD09F6
        command: 0x40BF906F
```

ngilles, 2022-05-25:
@moonraker46 The official NEC standard only species a 32 bit message, with 8 bit address and commands. The format of the message "on air" is `| addr | addr inverse | cmd | cmd inverse |` (32 bit message).

There then was the "extended" NEC format which basically made addresses 16 bit, the messages still being 32-bit long though: `| 16-bit addr | cmd | cmd inverse |`.

The current ESPHome code does allow for a bit of freedom int actual values, but still respects that 32 bit message format (hence the 16-bitness for `address` and `command`.

I myself have run into an issue with a device (standing fan) that looks like it's using NEC protocol, but it actually repeats the cmd + cmd inverse parts for a total 48bits message. I'm working on PR to solve this, but your specific device seems to be even wilder.

It looks like there are a bunch of devices that are using the NEC timings/signaling, but with their own message format/length, maybe what is needed is the implementation of a "nec_generic" protocol or something, which just handles bytes (that would still exclude protocols with arbitrary bit-lengths that aren't a multiple of 8).

moonraker46, 2022-05-25:
@ngilles I guess you are right. Seems to be a extended NEC IR protocol.

Anyway..using this specific Arduino Sketch I was able to send the extended IR NEC Codes to the receiver successfully. I just had to use the decimal of the IR Command (0x40BF6897 --> 1086285975) and it worked like a charm.

https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRServer/IRServer.ino

Is there a way to intergrate this extended NEC protocol into ESPHome? If not how could I use this sketch as a custom code in a ESPHome image?

Emrvb, 2022-09-22:
@moonraker46 I think you've got something confused from your receiver dumps. To me it looks like 0x40BF906F is a raw Extended NEC dump with address 0x817F and command 0x20. You could give that a try with esphome.

DjordjeMandic, 2025-05-26:
This seems resolved. Current implementation takes 16bits for address and 16bits for command.

Higher 8 bits of address are inverse of lower 8 bits in case of normal NEC protocol. In case of extended NEC, they differ.

Command should always have inverted byte but current implementation let's user supply full 16 bits just in case of specific use.

Note in docs should be enough for this. https://esphome.io/components/remote_transmitter.html#remote-transmitter-transmit-nec-action