esphome/issues #3310: Connection with SDP3X failed! after 2022.4.0 update

upnorth0416, 2022-05-23:
### The problem

Logs show "Connection with SDP3X failed!" after 2022.4.0. 

2022.3.2 working.

### Which version of ESPHome has the issue?

2022.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2022.4.7

### What platform are you using?

ESP32

### Board

featheresp32

### Component causing the issue

SDP3X

jddonovan, 2022-06-02:
I have the same issue. Considering it fails very early, I suspect the issue in registering the sensirion_common, which is handling i2c communication.

@Azimath

sdp3x_ns = cg.esphome_ns.namespace("sdp3x")
SDP3XComponent = sdp3x_ns.class_(
    "SDP3XComponent", cg.PollingComponent, sensirion_common.SensirionI2CDevice
)

Azimath, 2022-06-02:
Thanks for pinging about this, I'll have a look after work.

Azimath, 2022-06-05:
I plugged into serial on my sensor and got a problem with a checksum. My logs have `CRC8 Checksum invalid at pos 1! 0xFF != 0xAC` in them. Can you check if your boot logs have a similar output? It will only show up when connected to the serial port afaik.

Azimath, 2022-06-05:
Found the problem, will have a fix soon.

closed 2022-06-07 by jesserockz

Azimath, 2022-06-16:
> via esphome/esphome#3531 <https://github.com/esphome/esphome/pull/3531>.

Latest esphome release includes the fix for this issue.

jddonovan, 2022-06-16:
> Latest esphome release includes the fix for this issue.
> […](#)

Thank you so much!

upnorth0416, 2022-06-16:
Tested and confirmed working. Thank you!