esphome/issues #3314: Bedjet: Heat mode is unpredictable

jhansche, 2022-05-26:
### The problem

This is a result of PR https://github.com/esphome/esphome/pull/3437 (cc @jgissend10 - Thanks for your PR! Apologies for not jumping in when it was first opened... I was on vacation for 3 weeks and did not see it)

That change replaces the behavior of the hvac "heat" mode to activate the Bedjet "HEAT" button (from "EXT HT"), and moves EXT HT to a preset.

But I disagree with this change. The reason I intentionally used EXT HT and not HEAT, is because the HEAT mode is described in the manual as:

> Strong warming for up to a few hours
> Temp Range: 72°F to 104 °F
> Auto-shutoff: Variable based on fan and temp, typically less than 4 hours

while EXT HT is described as:

> Moderate warming for all night use
> Temp range: 72°F to 92°F
> Auto-shutoff: 12 hours

In my case, when my bedtime automation triggers, the Bedjet is turning itself off after only 15-30 minutes, and that change explains why I've been seeing this behavior (I thought it was a bug in the new Bedjet firmware at first). My opinion is that that behavior is incorrect for a HA/ESP climate device. When activating a "heat" hvac mode, it is a reasonable expectation that the mode should remain active throughout the night, and this one does not.

The 12-hour auto-shutoff time for EXT HT is the same as for COOL and DRY modes as well, which is why I intentionally chose EXT HT to accompany the rest of the standard HVAC modes. And I associated the standard Boost preset to the Bedjet TURBO mode.

My intention is to revert the above PR so that hvac "heat" mode continues to set EXT HT on the Bedjet. But in the spirit of keeping @jgissend10's ability to activate the Bedjet HEAT mode separately, I would instead like to move that mode into a separate custom preset, so that it's slightly more reasonable that a preset mode will shut off sooner than the hvac mode (Boost preset already shuts off after only 10 minutes).

I opened this as an issue first, in order to start a discussion, in case anyone disagrees with that proposal.

### Which version of ESPHome has the issue?

2022.5.0b2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.5

### What platform are you using?

ESP32

### Component causing the issue

bedjet

mwolter805, 2022-05-28:
Recommend leaving Heat and EXT HEAT as is (not reverting 3437) nor adding a heat preset.

Instead, add the ability to view and set the auto off time. 

**Advantages**
- Consistent user experience between the BedJet Remote and ESPHome for users that prefer to use HA but also use the remote. 
- Provide the ability to view and set the auto-off-time via a service call initiated by a dashboard button or automation.

This way ESPHome and BedJet functions would match, and be intuitive for users that switch between using HA and the remote to control the BedJet. Plus it would make it a little easier for future developers to "follow behind" and add features. This would be especially helpful if the ESP or HA goes down.

Having the ability to view and set the auto-off-time would simplify automating the BedJet. For instance, an HA automation could be created that triggers for bed presence change and auto-off-time below a threshold, say 10 minutes. These same triggers would then be used as conditions and an action option would then turn on, off, or extend the auto-off-time, fully automating the BedJet.

We wouldn't have the ability to set the auto off time through the UI entity editor, but a custom dashboard popup could easily be created (ie in UI-Lovelace-Minimalist or using button-card and browser_mod) that has the proper fields.

On that note, my "ultimate" automation would be adjusting the BedJet's temperature/fan according to "comfort" by using the room temp, a temp sensor hidden under the covers, or maybe an IR sensor that tracks the cover temp. I'm sure many of us bought a BedJet so we don't need to run the central heating/air at night. How many times have you woken up due to either being too hot or cold as the room's temperature changed? Probably many more times than we realize. 

Thank you

jhansche, 2022-05-29:
My goal for this component wouldn't be to mirror the device's remote though. I find the behavior erratic and inconsistent. I never use heat mode, because I expect it to run throughout the night, and a 30 min timer is not what I asked for when I click a button called heat. My point of view is one that corrects what I think is unexpected behavior.

Once it's in the HA ecosystem, the goal (in my opinion) should not be to mirror a bedjet remote, the goal is to make the device part of the smart home system, which means making it consistent with other entities in HA. The current behavior is not.

When I integrated my Nest thermostat, and I want to turn on the AC, I don't have to also tell it that I want it to stay on, it just does what I tell it. Your suggestion is just that: I should have to turn it on and then tell it not to turn off after 30 minutes. Never mind what the bedjet remote would do; it's not what a thermostat would do.

If the goal was to mirror the remote I would have cracked open the remote and soldered buttons to gpio outputs. So I think my counter to "Consistent user experience between the BedJet Remote and ESPHome" is that it's more important to maintain consistency _within_ the smart home, rather than retain unexpected behavior, even if that might be consistent with the device's remote.

I don't have any argument against exposing and setting the shutoff timer, and there's another feature request to support that. Though as discussed in another issue recently, a climate device is not able to expose anything that's not in the Climate device type, and I don't know of a way to create a separate entity type that will share the upstream BLE client, which is a requirement if we're going to add multiple component types. So the only workaround I can think of now is to give you a chance to inspect the BLE status packet as it's processed, and pair that with template sensors to extract the data you want. I'm open to other ideas, but I'm just familiar enough with ESPHome to know the "right" way to do that.

jhansche, 2022-05-29:
As for the suggestion for my use case, I use a scene for this, not a script or automation, and it's not possible in a scene to set the preset only; it has to also set the state (aka mode). This worked for several months by using the heat mode for ext-ht, but now I have to set both `state: heat` and `preset_mode: EXT-HT`. It does seem to work (I wasn't sure at first if it would), but it's not intuitive and I don't think it's proper. It also means the controller has to send two button commands for no reason.

It's not lost on me that in fact by undoing this, would flip the issue around, and someone who wants the behavior of bedjet's heat button would have the same problem. But it's still not clear to me why the heat button should be more useful than the ext-ht button. I've given my reasons for why the ext-ht button is more consistent, more natural, and behaves more like all the other modes of bedjet, but so far I have not heard any compelling reason that heat should be the first class mode.

jgissend10, 2022-05-29:
The inconsistency with the remote was why I made the change. My wife hit
the heat button in HA and was confused by the behavior of extended heat
being what got activated. We actually never use extended heat and use heat
as the primary setting. If an option for choosing the mapping could be made
that would be a great compromise.

jhansche, 2022-05-29:
I was just about to say the same. Maybe that's the best option. A yaml schema to select whether heat is short or extended, and put the other in presets... 🤔

jhansche, 2022-05-29:
That being said, I'm still curious why heat and not extended? Is the expectation that you only want a short time on heat and then shut off for the rest of the night? An automation could do that as well (bedjet was not designed for the smart home, which could explain why they made this a dedicated button).

I may not agree with the need for it to be the primary heating option (luckily no one has asked for Turbo to be primary 😂), but that doesn't mean we can't find a way to make it suit more users.

```
climate:
  - platform: bedjet
    heating_mode: [heat | extended | turbo ]
```

mwolter805, 2022-05-29:
My guess why heat’s auto-off-time is inconsistent with the other modes, is power savings. Either they are going for some energy rating or didn’t want to be blamed for higher electric bills.  Inductive heaters are inefficient and to raise the temperature by just a few degrees requires at least 150 to 200 watts continuously while on. Multiply this by eight hours then by 30 days and this can add up to a significant increase on your electric bill. Depending upon where you live, this can be $15 to $20 or more extra per BedJet. In SoCal, we pay triple the national average.

mwolter805, 2022-05-29:
Allowing the user to change the mapping seems like a great compromise.

mwolter805, 2022-05-29:
Regarding “exposing and setting the shutoff timer”, I’m not familiar enough with ESPHome. Plus I’m not familiar with the format the data is received from the BedJet. If I can make a recommendation, the people in the ESPHome discord seem to be very helpful and knowledgeable. That might be a good place to start.

jhansche, 2022-05-30:
I'm in the ESPHome discord server already, but so far I have not had any feedback that seems to be the de facto way to go.

There's a separate feature request for exposing additional data from the BedJet status packet: https://github.com/esphome/feature-requests/issues/1747. It's not necessarily ideal, because it could be seen as outside the scope of a "climate" component, but it seems to be the easiest compromise at this time. Feel free to continue that conversation at that feature request (don't want to get off-topic here).

closed 2022-05-31 by jesserockz