esphome/issues #3316: Cannot use !include with color

michaelblight, 2022-05-26:
### The problem

I have a number of components stored in separate files, and tried to do the same for color.
```
binary_sensor:
- <<: !include include/binary_sensor/connection-status.yaml

color:
- <<: !include include/color/named.yaml
```
The IDE is happy with it (no red x on line 1), but installing produces the compile error shown below. If I paste the content of the file (shown below) instead of the !include, it all works ok.
```
- id: color_blue
  red: 0%
  green: 100%
  blue: 0%
- id: color_green
  red: 0%
  green: 100%
  blue: 0%
- id: color_orange
  red: 100%
  green: 65%
  blue: 0%
- id: color_orange_deep
  red: 93%
  green: 46%
  blue: 0%
- id: color_red
  red: 100%
  green: 0%
  blue: 0%
```
The same thing occurs if I try to do it in a file referenced in `device_base` under `packages`.

### Which version of ESPHome has the issue?

2022.5.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.4

### What platform are you using?

ESP32

### Board

m5stick-c

### Component causing the issue

color

### Example YAML snippet

```yaml
See above
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/m5stick-c-1.yaml...
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING Pin 10 (9-10) might already be used by the flash interface in QUAD IO flash mode.
INFO Generating C++ source...
INFO Compiling app...
Processing m5stick-c-1 (board: m5stack-core-esp32; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0.1
Compiling /data/m5stick-c-1/.pioenvs/m5stick-c-1/src/main.cpp.o
/config/esphome/m5stick-c-1.yaml: In lambda function:
/config/esphome/m5stick-c-1.yaml:44:32: error: 'color_orange_deep' was not declared in this scope
       it.print(80, 10, id(font_1), color_orange_deep, TextAlign::TOP_LEFT, "Test 1");
                                ^
/config/esphome/m5stick-c-1.yaml:48:34: error: 'color_green' was not declared in this scope
       it.rectangle(80, 41, 1, 1, color_green);      // Centre
                                  ^
*** [/data/m5stick-c-1/.pioenvs/m5stick-c-1/src/main.cpp.o] Error 1
========================== [FAILED] Took 4.31 seconds ==========================
```

edenhaus, 2022-05-26:
Have you tried to remove the `-` before the include like:
```yaml
color:
  <<: !include include/color/named.yaml
```

The dash specifies a list entry in yaml meaning you are making a list in a list with your above configuration

michaelblight, 2022-05-26:
Dammit, you're right! All of my other uses have the "list in a list" situation - i.e. a `-` before the `<<:` and also in the included file, like I did with the `binary_sensor` above. It's probably more surprising that they work. I also changed the `binary_sensor` to remove the `-`, and it's happy like that. Good that this is recorded here, for when I do the same thing again.

michaelblight, 2022-05-26:
And I forgot to close.

closed 2022-05-26 by michaelblight