esphome/issues #3318: OTA updates simply do not work (ERROR Error sending data: timed out)

reinisb, 2022-05-27:
### The problem

It is impossible to update my SparkFun ESP32 board over the air. It gets to uploading 5%, sometimes 20%, sometimes 50% but then times out. The error in the OTA console is `ERROR Error sending data: timed out` while the error in logs is `local: Ping timed out!`

I have a really solid Ubiquiti Unifi wifi network so I assume that it is not the cause.

[image]

### Which version of ESPHome has the issue?

2022.5.1 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant Core 2022.5.5, Home Assistant Supervisor 2022.05.3, Home Assistant OS 8.1, Kernel version 5.15.32-v8, Agent version 1.2.1

### What platform are you using?

ESP32

### Board

SparkFun ESP32 Thing

pmannk, 2022-05-29:
Are you using the bluetooth component of the ESP32?
I've been down this rabbit hole with ESP32's that have bluetooth enabled. The radio is a shared component of the hardware, so using both at the same time can create a world of problems.

If you don't have bluetooth enabled then I suggest looking at your WiFi equipment settings. I'm not using Ubiquiti (I use OpenWRT on a variety of hardware) and I found I had to reduce my 802.11b/g/n channel width from 40MHz to 20Mhz for a number of "IoT" devices to remain stable.

reinisb, 2022-05-30:
> Are you using the bluetooth component of the ESP32?
@pmannk  Thanks for the suggestion! I'm not using the bluetooth component, hence if I'd be glad to disable it if it causes issues to OTA. I did a quick google search and I cannot seem to find a way how to disable the bluetooth component on ESP32. Do you happen to know how to do that? Thanks!

pmannk, 2022-05-30:
The component would be in the ESPHome configuration for your device. Eg:
esp32_ble_tracker:
xiaomi_ble:
etc

ScIT-Raphael, 2023-01-12:
Same issue here, I upgraded my net with wifi6, aswell switched to the current version of esphome (got stuck due to the repo change with homeassistant). Since then, I just get the error above. Does anyone have an idea how to fix it?

masman69, 2023-01-14:
Same with (fake) ESP32 Mini ESP-WROOM-32 board (HiLetgo version on Amzn).  Uninstalled and reinstalled ESPHome to insure latest version:  2022.12.3
Ubiquity networking, latest version.  Verified that 2.4 GHz is channel width 20 MHz as suggested above.
Same error with multiple devices of same brand.

mcinnes01, 2023-01-24:
I get the same on some of my sonoff 4ch pros. I have a ubiquiti setup too and the signal strength is -22db. This seems to happen on my upstairs ones constantly which are near the upstairs AP, and I've tried moving the AP nearer and further away and get the same issue. Where as the same sonoff 4ch pro devices in my basement setup, also fairly near the basement AP have no issues at all.

reinisb, 2023-01-24:
Hey! Wanted to share a quick update - somehow the issue got resolved and I've been able to nicely do OTA updates to my ESP32 via ESPHome. Though I'm not sure how it got resolved. I am running the latest version of Home Assistant OS on my Raspberry Pi and the latest version of ESPHome as well.

That being said, I see that others are still experiencing this hence the issue is still around.

ScIT-Raphael, 2023-01-25:
Just updated to the latest esphome version, but still have the same issue:

```
Uploading: [====                                                        ] 7% 
ERROR Error sending data: timed out
```

Already made sure the IOT network is 2.4ghz only and a few other settings I've found. Issue occured either due to esphome update but probaly is more related due to the change to unifi wifi6 aps.

FlyingEagle, 2023-02-03:
Same here.

I```
NFO Compressed to 342209 bytes
Uploading: [========================                                    ] 40% 
ERROR Error sending data: timed out

```

Version: 2022.12.8

ScIT-Raphael, 2023-02-14:
Found the solution here: https://community.ui.com/questions/Big-instabilities-with-UDM-and-home-assistant-mDNS-ESP8266-ESP32/43ab26c1-6339-4608-95fb-ace2ac83eb39#answer/34613a21-2f75-4d8e-8dfa-79415c764041

After do that specific adjustments, flashing is back at work :).

gid204, 2023-04-01:
I just wanted to add my two cents as I had this same issue which was driving me nuts.

The issue for me, as silly as it will sound, was a poor ethernet termination to my AP. It was 'okay' enough for Wi-Fi to work quite reliably (for most devices) but after digging around for ages I noticed that it was connected at 100Mb opposed to 1Gbe. After more digging I found it was dropping around 50% of all Rx packets (surprised I didn't pick up on this earlier)...

Anyway, moral of the story - [check this simple things, it's probably the solution.](https://en.wikipedia.org/wiki/Occam's_razor)

ssunny1081, 2023-07-29:
Had the same issue, suddenly without any Change. Also using Ubiquity, if've seen in the logs the the esp was roaming between two APs even roaming is deactivated on the IOT network. Locking the device to a single AP resolved the issue.

AdamN1, 2023-08-01:
In my case problem have been with power control of the Wifi, try `power_save_mode: none` in the wifi section

https://esphome.io/components/wifi.html

mcinnes01, 2023-08-01:
I've not solved my issue yet, but I've tried all the above and none have worked. When I get a minute I'm pretty sure the AP where the devices are having issues, will turn out to have a faulty network cable. I think the OTA issue is down to packet loss and clearly the tolerance of this is fairly low with either esphome or the esp network stack or both in combination with ubiquiti APs.

proflan, 2023-09-17:
power_save_mode: none did it for me

RichFoley, 2023-10-14:
> > 
>
> try `ping -l 1492 AP_IP`. If problem is with network cable you will have losses.

@AdamN1 : is there a reason for 1492 bytes? I can ping with 1472 max, anything over that is 100% loss. 

Also on Ubiquiti with a newish WiFi 6 AP (don't think I've done an OTA update since upgrading). Channel width is 20. ESP WiFi power save mode already off.

pmannk, 2023-10-15:
I have over 50 ESP8266/ESP32/BK7231 (ESPHome LibreTiny) devices connected to Ubiquiti APs (2x U6 Pros) with zero issues. I moved from OpenWRT to Ubiquiti a few months ago and it's actually been more reliable than my OpenWRT based setup.
@RichFoley here's the advanced config I use for my "IoT" SSID if it helps:
[image]
Other Global AP settings: 
- 20MHz channel width on my 2.4GHz radio
- Channel optimisation is off
- Wireless meshing is off

A bit about my deployment:
- Only "IoT" style devices on the 2.4GHz SSID. All other clients are on a 5GHz-only SSID. Obviously no band steering as a result
- 2x U6 Pro APs
- Non-Ubiquiti POE switch and OpenWRT x86 router (so no UDM or Edgerouter products)

AdamN1, 2023-10-15:
> > try `ping -l 1492 AP_IP`. If problem is with network cable you will have losses.
> @AdamN1 : is there a reason for 1492 bytes? I can ping with 1472 max, anything over that is 100% loss.

Standard MTU is 1500 minus payload it gives 1492.
If you don't have losses with 1472 bytes it means that wifi is working correctly.

hein4105, 2025-04-27:
i have problem with oat upload, I got the following error message: ERROR Unsupported OTA version 2.

randybb, 2025-04-27:
> i have problem with oat upload, I got the following error message: ERROR Unsupported OTA version 2.
Are you updating some super old device? Just switch to `version: 1` and after update remove it. https://esphome.io/components/ota/esphome

hein4105, 2025-04-27:
no, it just a actual model atom m5. I will try using version 1 in the yaml, a config parameter.

hein4105, 2025-04-27:
how can I implement the parameter in the yaml script, under ota:
ota:
  version: 1

this is a error for esphome