esphome/issues #3321: CAN bus integration: using variable can_id_mask throws error

Jpsy, 2022-05-28:
### The problem

Usage of the variable `can_id_mask` in the component CAN bus throws when compiling.
Even the example from the [ESPHome docs](https://esphome.io/components/canbus.html) does not compile:

The error message complains about `can_id` but it is actually triggered by using `can_id_mask`:
```
  Must be string, got <class 'NoneType'>. did you forget putting quotes around the value?.
  on_frame: 
    - 
      Unable to find action with the name 'can_id'.
      can_id: 64
      can_id_mask: 520110016
      use_extended_id: True
      remote_transmission_request: False
      then: 
        - lambda: return;
```

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.5

### What platform are you using?

ESP8266

### Board

D1 mini

### Component causing the issue

CAN bus

### Example YAML snippet

```yaml
# example from the docs (only platform added)
canbus:
  - platform: mcp2515
    cs_pin: D5
    can_id: 4
    on_frame:
    - can_id:      0b00000000000000000000001000000
      can_id_mask: 0b11111000000000011111111000000
      use_extended_id: true
      remote_transmission_request: false
      then:
        - lambda: |-
            return;
```

iot49, 2022-07-28:
I do not get an error.

Instead the option appears to be ignored - per documentation I'd expect setting `can_id_mask` to `0` should accept all `can_id`s, instead I get nothing. Triggering on a specific `can_id` works as expected.

closed 2022-12-12 by github-actions[bot]

ssieb, 2022-12-12:
> I do not get an error.
> 
> Instead the option appears to be ignored - per documentation I'd expect setting `can_id_mask` to `0` should accept all `can_id`s, instead I get nothing. Triggering on a specific `can_id` works as expected.

You would also need to set the `can_id` to 0 in that case.  The masked value has to match the id you give.