esphome/issues #3325: SSD1306 causing perfomance issues

ShadyMCShadesen, 2022-05-29:
### The problem

If i add the SSD1306 component to any of my projects, the complete program of the ESP gets extremely slow and everything takes longer (getting states for entities from Home Assistant for example). Even if the display code consists of drawing only one static text.

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

SSD1306

### Additional information

This bug occured the first time over half a year ago. Since then i can't really use the display in projects that need to have a fast responsiveness.

vincentkoevoets, 2022-06-03:
I have the same ESPhome version, but a ESP32 board (LilyGo TTGo T-cam). I also found that it matters if the values are pulled from HA (either via mqtt or native api), or the text is defined static in the ESPhome lambda. 
If I define the text in the lambda itself, the logs still give me this:
```yaml
Component ssd1306_base took a long time for an operation (0.22 s)
Components should block for at most 20-30ms
```
I can bring the block-time down by increasing the i2c frequency (to 400 khz for example), but still it blocks for 0.05s which is way too much. 
Simple text:
```yaml
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    rotation: 180
    address: 0x3C
    id: esp_deurbel_display
    lambda: |-
      it.printf(64, 0, 0, TextAlign::CENTER_HORIZONTAL, "%s", "This is line 1" );
      it.printf(64, 20, 0, TextAlign::CENTER_HORIZONTAL, "%s", "This is line 2" );
      it.printf(64, 40, 0, TextAlign::CENTER_HORIZONTAL, "%s", "This is line 3" );
```

If I try to get some values from a mqtt text sensor, block times stay about the same but ESPhome begins to reconnect to mqtt broker very frequently. I also tried the native api for this with the same results. 
This might be a separate problem, but the case remains that the display component blocks the loop for way to long. 
Text from mqtt:
```yaml
text_sensor:
  - platform: mqtt_subscribe
    topic: deurbel_tekst_regel1
    name: "ESP deurbel tekst regel 1"
    id: mqtt_deurbel_regel1
  - platform: mqtt_subscribe
    topic: deurbel_tekst_regel2
    name: "ESP deurbel tekst regel 2"
    id: mqtt_deurbel_regel2
  - platform: mqtt_subscribe
    topic: deurbel_tekst_regel3
    name: "ESP deurbel tekst regel 3"
    id: mqtt_deurbel_regel3
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    rotation: 180
    address: 0x3C
    id: esp_deurbel_display
    lambda: |-
      it.printf(64, 0, 0, TextAlign::CENTER_HORIZONTAL, "%s", strdup(id(mqtt_deurbel_regel1).state.c_str()) );
      it.printf(64, 20, 0, TextAlign::CENTER_HORIZONTAL, "%s", strdup(id(mqtt_deurbel_regel2).state.c_str()) );
      it.printf(64, 40, 0, TextAlign::CENTER_HORIZONTAL, "%s", strdup(id(mqtt_deurbel_regel3).state.c_str()) );
```

closed 2022-11-01 by github-actions[bot]

dany547, 2022-11-18:
any solution for this problem? im having the exact same issue with the latest version of ESPHOME. If i dont enable Verbose logging the device wont event connect to WIFI.

ssieb, 2022-11-20:
Did you increase the i2c frequency?