esphome/issues #3326: Display Pages of st7920 Problem

SciTii, 2022-05-29:
### The problem

It looks like pages are not working for the ST7920 display. 

When i do put traditional Lambda it displays it correctly but when i put 2 pages it is displaying nothing.

I have checked the same code for SD1306 and it is working without any problems

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.5.3

### What platform are you using?

ESP32

### Board

ESP32 R1 

### Component causing the issue

Display

### Example YAML snippet

```yaml
esphome:
  name: hydroponika-stacja-1

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

logger:

api:
  password: !secret api_password
  encryption:
        key: !secret encryption_key
ota:
  safe_mode: true
  password: !secret ota_password
  id: my_ota

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  
  ap:
    ssid: "Hydroponika-Stacja-1"
    password: !secret wifi_password_fallback

captive_portal:

spi:
  clk_pin: GPIO19
  mosi_pin: GPIO13
  
font:
  - file: "fonts/conthrax.ttf"
    id: font1
    size: 9
    
  - file: "fonts/conthrax.ttf"
    id: font2
    size: 33

display:
  - platform: st7920
    setup_priority: -300
    width: 128
    height: 64
    id: display1
    cs_pin:
      number: GPIO22
      inverted: true
    pages:
      - id: page1
        lambda: |-
          it.print(0, 10, id(font1), "This is page 1!");
      - id: page2
        lambda: |-
          it.print(0, 10, id(font1), "This is page 2!");
```

### Additional information

Traditional Lambda woking without any problems

display:
  - platform: st7920
    setup_priority: -300
    width: 128
    height: 64
    id: display1
    cs_pin:
      number: GPIO22
      inverted: true
    lambda: |-
          it.print(0, 10, id(font1), "This is page 1!");

Wummeke, 2022-08-04:
I have exact the same problem, with ESPHome version 2022.6.2.

SciTii, 2022-08-05:
It looks like they don't really care to repair it ;/

Wummeke, 2022-08-05:
I noticed in the source code for this integration that st7920/display.py is missing a variables concerning pages and some additional code I do see when I look at the source for other displays. Unfortunately, just adding these, did not fix this. Perhaps it is missing stuff in the CPP code as well, but I'm not very good at understanding that.

I'm still not sure if the code is lacking some lines to make this work, or that the hardware is not capable of displaying pages. But from what I understand now, I don't think the latter is the case. If I interpreted the code correctly, pages is just a different way of showing lambdas. But I could be wrong, of course. Hopefully someone smarter than me will have a look soon.

Wummeke, 2022-11-21:
There's a workaround: https://github.com/esphome/feature-requests/issues/1480#issuecomment-1322378834

mateusz-jaworski, 2023-10-31:
I use ESPHome 2023.10.4 on esp8266 with st7920 and there is the same case

IschdeFelin, 2025-05-10:
I can confirm that this issue still exists in ESPHome v2025.4.1 with an ESP32 DevKit and an ST7920 128x64 display.

Even with a correctly configured pages: section, nothing is rendered on the display.

Static content using a lambda: outside of pages: works fine, so the display itself is functional. But as soon as I switch to a pages: configuration, the screen stays blank.