esphome/issues #3327: ESP32 does not reboot fully after upgrades

user897943, 2022-05-29:
### The problem

ESP32-based board (model https://www.ebay.co.uk/itm/154316034047?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=wyRWbzZyQZW&sssrc=2349624&ssuid=dBbZzHsUQyu&var=&widget_ver=artemis&media=COPY)

Whenever I update ESPHome this board does not restart cleanly. I am currently using it to control a bedjet climate device with ESPHome. See code from serial log below.

### Which version of ESPHome has the issue?

2022.6.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Not applicable

### What platform are you using?

ESP32

### Board

See link above for exact board

### Component causing the issue

esphome

### Anything in the logs that might be useful for us?

```txt
assert failed: vTaskPriorityDisinheritAfterTimeout tasks.c:4751 (pxTCB != pxCurrentTCB[xPortGetCoreID()])

Backtrace:0x40081d4e:0x3ffce880 0x40090d75:0x3ffce8a0 0x40096f15:0x3ffce8c0 0x40093519:0x3ffce9e0 0x40091d9e:0x3ffcea00 0x400dd064:0x3ffcea40 0x401a0751:0x3ffceb40 0x401a0811:0x3ffceb60 0x400e2c1e:0x3ffceb80 0x400e4a6a:0x3ffcebb0 0x400db6af:0x3ffcecb0

ELF file SHA256: 4346ac4e2285620a

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7592
load:0x40078000,len:14912
load:0x40080400,len:3688
entry 0x4008067c
I (27) boot: ESP-IDF 4.3.2 2nd stage bootloader
I (27) boot: compile time 00:18:49
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (82) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 app0             OTA app          00 10 00010000 001c0000
I (97) boot:  4 app1             OTA app          00 11 001d0000 001c0000
I (104) boot: End of partition table
I (109) boot_comm: chip revision: 1, min. application chip revision: 0
I (116) esp_image: segment 0: paddr=001d0020 vaddr=3f400020 size=2d770h (186224) map
I (192) esp_image: segment 1: paddr=001fd798 vaddr=3ffbdb60 size=02880h ( 10368) load
I (196) esp_image: segment 2: paddr=00200020 vaddr=400d0020 size=d5a38h (875064) map
I (514) esp_image: segment 3: paddr=002d5a60 vaddr=3ffc03e0 size=02088h (  8328) load
I (517) esp_image: segment 4: paddr=002d7af0 vaddr=40080000 size=1cc34h (117812) load
I (568) esp_image: segment 5: paddr=002f472c vaddr=50000000 size=00010h (    16) load
I (582) boot: Loaded app from partition at offset 0x1d0000
I (582) boot: Disabling RNG early entropy source...
I (594) cpu_start: Pro cpu up.
I (594) cpu_start: Starting app cpu, entry point is 0x40081da0
I (581) cpu_start: App cpu up.
I (611) cpu_start: Pro cpu start user code
I (611) cpu_start: cpu freq: 160000000
I (611) cpu_start: Application information:
I (615) cpu_start: Project name:     bedjetv3-fb1
I (621) cpu_start: App version:      2022.6.0-dev
I (626) cpu_start: Compile time:     May 17 2022 00:17:24
I (632) cpu_start: ELF file SHA256:  4346ac4e2285620a...
I (638) cpu_start: ESP-IDF:          4.3.2
I (643) heap_init: Initializing. RAM available for dynamic allocation:
I (650) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (656) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (662) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (668) heap_init: At 3FFCAD38 len 000152C8 (84 KiB): DRAM
I (675) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (681) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (687) heap_init: At 4009CC34 len 000033CC (12 KiB): IRAM
I (695) spi_flash: detected chip: generic
I (698) spi_flash: flash io: dio
I (705) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
```
```

user897943, 2022-05-29:
@jhansche you might be interested in this, wasn't sure if your component had to do anything with the issue so I'm flagging for your attention only, in case it is. Looks more like a ESPHome or Espressif issue?

jhansche, 2022-05-30:
I haven't had this issue myself. If it's an issue with booting, I would expect either a hardware issue, or a compatibility issue with this specific board. I don't think the component would cause this, but I'm not familiar enough with ESPHome and how it's bootstrapped.

user897943, 2022-05-30:
> I haven't had this issue myself. If it's an issue with booting, I would expect either a hardware issue, or a compatibility issue with this specific board. I don't think the component would cause this, but I'm not familiar enough with ESPHome and how it's bootstrapped.

What exact board are you using?

jhansche, 2022-05-30:
> What exact board are you using?

I use this one: https://www.amazon.com/gp/product/B07V42CJRF/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
but importantly, this is the only ESP32 board I've used before, and other than that I've only used one other ESP8266 board. So my experience is limited.

user897943, 2022-05-30:
That board is no longer available. But from the images I can see it also
uses the wroom chip..  mine just has an Arduino sort of format. Same chip.
What does your yaml look like (full)…? please redact sensitive data

jhansche, 2022-05-30:
```yaml
esphome:
  name: bedjet_proxy
  platform: ESP32
  board: esp32dev

## FIXME: Allows ble_tracker to start working again - but only if bedjet components are disabled.
# esp32:
#   board: esp32dev
#   framework:
#     type: esp-idf

wifi:
  ssid: !secret wifi_runtime_ssid
  password: !secret wifi_runtime_key
  ap:
    ssid: "Bedjet Proxy Fallback Hotspot"
    password: !secret captive_fallback_bedjet

time:
  - platform: homeassistant
    id: hass_time

# Enable logging
logger:
  # level: VERY_VERBOSE
  # level: VERBOSE
  #baud_rate: 0

api:
  password: !secret ota_password
ota:
  password: !secret ota_password
  on_begin:
    then:
      - logger.log: "Disconnecting clients for OTA update..."
      - switch.turn_off: bedjet_j_monitor
      - switch.turn_off: bedjet_l_monitor

# Mandatory to set up the BLE stack
esp32_ble_tracker:

binary_sensor:
  - platform: template
    id: bedjet_j_connected
    device_class: connectivity
    name: "BedJet JOENJET Connected"
  - platform: template
    id: bedjet_l_connected
    device_class: connectivity
    name: "BedJet PUFFY Connected"

ble_client:
  - mac_address: C4:4F:33:xx:xx:xx
    id: bedjet_joenjet_client
    on_connect:
      then:
        - lambda: |-
            id(bedjet_j_connected).publish_state(true);
    on_disconnect:
      then:
        - lambda: |-
            id(bedjet_j_connected).publish_state(false);
  - mac_address: C4:4F:33:xx:xx:xx
    id: bedjet_puffy_client
    on_connect:
      then:
        - lambda: |-
            id(bedjet_l_connected).publish_state(true);
    on_disconnect:
      then:
        - lambda: |-
            id(bedjet_l_connected).publish_state(false);

switch:
  - platform: ble_client
    id: bedjet_j_monitor
    name: "BedJet Enable (JOENJET)"
    ble_client_id: bedjet_joenjet_client
  - platform: ble_client
    id: bedjet_l_monitor
    name: "BedJet Enable (PUFFY)"
    ble_client_id: bedjet_puffy_client
  - platform: restart
    name: "BedJet Proxy Restart"

climate:
  - platform: bedjet
    ble_client_id: bedjet_joenjet_client
    id: bedjet_j_climate
    name: JOENJET
    time_id: hass_time
    receive_timeout: 0s
  - platform: bedjet
    ble_client_id: bedjet_puffy_client
    id: bedjet_l_climate
    name: PUFFY
    time_id: hass_time
    receive_timeout: 0s
```

user897943, 2022-05-30:
Thanks. I believe you have two (for dual zone) right?

Do me a big favor. Repost the config but clean up the code removing any
unnecessary stuff and comment out the 2nd device stuff (I may buy a 2nd
unit later and would be easier to amend with my details).

I'll flash your config and see what happens as I noticed several
differences.

I don't wanna assume I can remove parts and then get bad results because of
mistakes.

I'll report back with results.

user897943, 2022-05-30:
I edited, please take a look.

```
esphome:
  name: bedjetv3-fb1
  platform: ESP32
  board: esp32dev

## FIXME: Allows ble_tracker to start working again - but only if bedjet components are disabled.
# esp32:
#   board: esp32dev
#   framework:
#     type: esp-idf

wifi:
  ssid: mywifi
  password: !secret wifisecret
  manual_ip:
  static_ip: 10.0.x.x
  gateway: 10.0.x.x
  subnet: 255.255.x.x
  dns1: 10.0.x.x
  dns2: 10.0.x.x
  ap:
    ssid: "Bedjet FB Bridge Recovery"
    password: !secret wifire

time:
  - platform: homeassistant
    id: hass_time

# Enable logging
logger:
  # level: VERY_VERBOSE
  # level: VERBOSE
  #baud_rate: 0

api:
  password: !secret passapi
ota:
  password: !secret passota
  on_begin:
    then:
      - logger.log: "Disconnecting clients for OTA update..."
      - switch.turn_off: bedjet_fb1_monitor
#      - switch.turn_off: bedjet_fb2_monitor

# Mandatory to set up the BLE stack
esp32_ble_tracker:

binary_sensor:
  - platform: template
    id: bedjet_fb1_connected
    device_class: connectivity
    name: "BedJet FB 1 Connected"
#  - platform: template
#    id: bedjet_fb2_connected
#    device_class: connectivity
#    name: "BedJet FB 2 Connected"

ble_client:
  - mac_address: 08:3A:F2:x:x:x
    id: bedjet_fb1_client
    on_connect:
      then:
        - lambda: |-
            id(bedjet_fb1_connected).publish_state(true);
    on_disconnect:
      then:
        - lambda: |-
            id(bedjet_fb1_connected).publish_state(false);
#  - mac_address: C4:4F:33:xx:xx:xx
#    id: bedjet_fb2_client
#    on_connect:
#      then:
#        - lambda: |-
#            id(bedjet_fb2_connected).publish_state(true);
#    on_disconnect:
#      then:
#        - lambda: |-
#            id(bedjet_fb2_connected).publish_state(false);

switch:
  - platform: ble_client
    id: bedjet_fb1_monitor
    name: "BedJet Enable (FB 1)"
    ble_client_id: bedjet_fb1_client
#  - platform: ble_client
#    id: bedjet_fb2_monitor
#    name: "BedJet Enable (FB 2)"
#    ble_client_id: bedjet_fb2_client
  - platform: restart
    name: "BedJet FB Bridge Restart"

climate:
  - platform: bedjet
    ble_client_id: bedjet_fb1_client
    id: bedjet_fb1_climate
    name: BedJet 1 - Front room
    time_id: hass_time
    receive_timeout: 0s
#  - platform: bedjet
#    ble_client_id: bedjet_2_client
#    id: bedjet_fb2_climate
#    name: BedJet 2 - Front room
#    time_id: hass_time
#    receive_timeout: 0s
button:
  - platform: template
    name: "Check BedJet FB1 Firmware"
    on_press:
      then:
      - lambda: |-
          id(bedjet_fb1_climate).upgrade_firmware();
#  - platform: restart
#    name: "BedJet FB Bridge Restart"
#  - platform: safe_mode
#    name: "BedJet FB Bridge Restart (Safe Mode)"
```

jhansche, 2022-05-30:
This looks like it should work the same as mine (but just one unit: you're correct that I have dual units). The only thing I'm not sure about, but maybe someone else has more experience, is how important the board is for this. I know there are some hardware differences, but as we're not using gpios here, I'm not sure how important that would be. Could always try it and see, worst case you might need to edit the board entry and reflash.

dentra, 2022-06-11:
I have the same assertion on Lolin32 Lite and ESPHome 2022.5.1 on my custom component that also works with BLE. In my case it crashes constantly. Here is decoded stack trace:
```
0x40081e46: panic_abort at /esphome/.temp/platformio/packages/framework-espidf/components/esp_system/panic.c:404
0x40090e99: esp_system_abort at /esphome/.temp/platformio/packages/framework-espidf/components/esp_system/system_api.c:112
0x40097039: __assert_func at /esphome/.temp/platformio/packages/framework-espidf/components/newlib/assert.c:85
0x4009363d: vTaskPriorityDisinheritAfterTimeout at /esphome/.temp/platformio/packages/framework-espidf/components/freertos/tasks.c:4751 (discriminator 1)
0x40091ec2: xQueueSemaphoreTake at /esphome/.temp/platformio/packages/framework-espidf/components/freertos/queue.c:1629
0x400dc780: esphome::esp32_ble_tracker::ESP32BLETracker::loop() at /workspaces/esphome-tion/.esphome/build/kidsroom-tion/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:103
0x401a4c8d: esphome::Component::call_loop() at /workspaces/esphome-tion/.esphome/build/kidsroom-tion/src/esphome/core/component.cpp:75
0x401a4d4d: esphome::Component::call() at /workspaces/esphome-tion/.esphome/build/kidsroom-tion/src/esphome/core/component.cpp:97
0x400e4d80: esphome::Application::loop() at /workspaces/esphome-tion/.esphome/build/kidsroom-tion/src/esphome/core/application.cpp:74 (discriminator 2)
0x400e7362: loop() at /workspaces/esphome-tion/.esphome/build/kidsroom-tion/tion-home-base-dev.yaml:58
0x400dacc2: esphome::loop_task(void*) at /workspaces/esphome-tion/.esphome/build/kidsroom-tion/src/esphome/components/esp32/core.cpp:74 (discriminator 1)
```
It fails on xSemaphoreTake just after esp_ble_gap_stop_scanning:
```c++
         if (client->state() == ClientState::DISCOVERED) {
            esp_ble_gap_stop_scanning();
            if (xSemaphoreTake(this->scan_end_lock_, 10L / portTICK_PERIOD_MS)) {
              xSemaphoreGive(this->scan_end_lock_);
            }
          }
```

It 100% reproducable and can be repeated by ble_client switch, steps are:
 - ble_node->parent()->set_enabled(false);
 - wait some time or not
 - ble_node->parent()->set_enabled(true);
 - crash
First connection is always OK.

jhansche, 2022-06-11:
That could explain why it happens in OP's case, and I've seen it happen myself now a couple times (without logs) where I have to unplug/replug to get it back. I added an ota.on_begin automation to turn off the ble_switch. I only noticed it myself on idf framework, but not on Arduino.

Any initial thoughts on what could be causing it?

jhansche, 2022-06-11:
I also have some fallback code in bedjet component that tries to detect a stuck state and reset the BLE connection by toggling the parent client's enabled state, but that occurs only if you enable the `receive_timeout` config in the bedjet component.

@user897943 are you using idf or arduino framework? If this is limited to idf, we might be able to comment out those two configs (receive timeout and ota automation) to see if it helps, and maybe recommend them only if on Arduino framework.

The longer term option is to track down why the semaphore fails in this loop and try to mitigate or fix that. Maybe BLEClient's enabled flag needs better control over when and how it toggles its connection.

user897943, 2022-06-11:
When I opened this issue I was using the IDF framework. I'm on Arduino now.

user897943, 2022-06-11:
Having said that, remember I was on a different board, the one I am using now works flawlessly. The previous board had the wifi and BT antenna as a printed trace on the board. This one I'm using now has a connector for an external antenna so the signal is much more reliable.

dentra, 2022-06-11:
I think it is not bedjet related but ESP32BLETracker. So here is quick fix, do not know how it's correct but working:
```diff
            esp_ble_gap_stop_scanning();
---            if (xSemaphoreTake(this->scan_end_lock_, 10L / portTICK_PERIOD_MS)) {
+++            if (xSemaphoreTake(this->scan_end_lock_, 0)) {
              xSemaphoreGive(this->scan_end_lock_);
```

dentra, 2022-06-11:
And yes, it is only IDF related

michaelblight, 2022-06-19:
> I think it is not bedjet related but ESP32BLETracker. So here is quick fix, do not know how it's correct but working:
>
> ```diff
>             esp_ble_gap_stop_scanning();
> ---            if [...]

This works for me too, and I was having the problem on Arduino and IDF on m5stamps. Bit strange given that nothing has changed in the code for 6 months, and that particular code has been there for 3 years; and nothing has changed in my ESPHome config either.

jhansche, 2022-06-19:
I just made the change yesterday to test it. So far it looks ok, but I'll know for sure after a couple more days.

@dentra do you want to maybe open a PR for this, where we can get other people involved, who are more familiar with it? I don't know the implications of making this change, or why the original value was chosen in the first place (must have been for some reason, right? 🤷‍♂️).

jhansche, 2022-06-19:
> After 55 minutes it came online, and has stayed that way since.

You mean it was unavailable for 55 minutes, and then recovered? Mine usually won't recover until I power cycle it, when it hits this.

michaelblight, 2022-06-20:
> You mean it was unavailable for 55 minutes, and then recovered? Mine usually won't recover until I power cycle it, when it hits this.
Yes, weird. I don't have serial connected, so only see the logs over wifi, which only connects after that time. However, at boot priority 600, I turn the builtin LED off then red after 500ms, and once it connects to wifi then yellow, then green once the API is connected. The LED is staying solid red - if it was boot-looping I'd expect to see the LED blinking red.

dentra, 2022-06-21:
@jhansche OK, I will open PR for this with preprocessor directive to not affect arduino users.

jhansche, 2022-06-26:
> I just made the change yesterday to test it. So far it looks ok, but I'll know for sure after a couple more days.

So it's been about a week now, and I'm seeing that both units have remained connected and stable. One of the units does this thing where it flaps (drop and reconnect) every 30 min, almost like clockwork. It hasn't really been a problem for me, and it usually reconnects in a matter of 2-3 seconds. The other unit never used to have any problems, but I moved the esp32 across the room it was in a week or so ago, and now I see much more sporadic connectivity issues in that one. I'll move it back closer to where it was before, to confirm that distance was the cause, and not this change.

Overall I think this has been an improvement.

closed 2022-08-07 by jesserockz