esphome/issues #3329: Unable to communicate to Tuya MCU on fan controller

hydazz, 2022-05-31:
### The problem

I have succesfully flashed my brilliant fan controller [(This One)](https://www.brilliantlighting.com.au/product/smart-wifi-ac-fan-remote-0a11e7) with esphome, but using the yaml below i am unable to communicate with the tuya MCU.
does anyone know why this would be?
this is the error im getting:
```yaml
[19:54:04][C][tuya:029]: Tuya:
[19:54:04][C][tuya:032]:   Initialization failed. Current init_state: 0
[19:54:04][C][tuya:037]:   If no further output is received, confirm that this is a supported Tuya device.
[19:54:04][E][tuya:391]: Initialization failed at init_state 0
[19:54:04][E][tuya:391]: Initialization failed at init_state 0
...
```

[Tasmota template for older model](https://templates.blakadder.com/brilliantsmart_99111.html) 

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

TYWE1S

### Component causing the issue

tuya

### Example YAML snippet

```yaml
# settings
esphome:
  name: living-room-fan
esp8266:
  board: esp_wroom_02

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.2.133
    gateway: 192.168.2.1
    subnet: 255.255.255.0
    dns1: 192.168.2.1
  domain: .iot

logger:
  baud_rate: 0

api:
  encryption:
    key: !secret api_key

ota:
  password: !secret ota_password

# configuration

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

tuya:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/living-room-fan.yaml...
INFO Starting log output from 192.168.2.133 using esphome API
INFO Successfully connected to 192.168.2.133
[19:54:03][I][app:102]: ESPHome version 2022.5.1 compiled on May 31 2022, 09:47:55
[19:54:03][C][wifi:491]: WiFi:
[19:54:03][C][wifi:353]:   Local MAC: 34:AB:95:15:BD:B2
[19:54:03][C][wifi:354]:   SSID: [redacted]
[19:54:03][C][wifi:355]:   IP Address: 192.168.2.133
[19:54:03][C][wifi:356]:   BSSID: [redacted]
[19:54:03][C][wifi:358]:   Hostname: 'living-room-fan'
[19:54:03][C][wifi:360]:   Signal strength: -63 dB ▂▄▆█
[19:54:03][C][wifi:364]:   Channel: 6
[19:54:03][C][wifi:365]:   Subnet: 255.255.255.0
[19:54:03][C][wifi:366]:   Gateway: 192.168.2.1
[19:54:03][C][wifi:367]:   DNS1: 192.168.2.1
[19:54:03][C][wifi:368]:   DNS2: 0.0.0.0
[19:54:03][C][logger:274]: Logger:
[19:54:03][C][logger:275]:   Level: DEBUG
[19:54:03][C][logger:276]:   Log Baud Rate: 0
[19:54:03][C][logger:277]:   Hardware UART: UART0
[19:54:04][C][uart.arduino_esp8266:102]: UART Bus:
[19:54:04][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[19:54:04][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[19:54:04][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[19:54:04][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[19:54:04][C][uart.arduino_esp8266:109]:   Data Bits: 8
[19:54:04][C][uart.arduino_esp8266:110]:   Parity: NONE
[19:54:04][C][uart.arduino_esp8266:111]:   Stop bits: 1
[19:54:04][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[19:54:04][C][tuya.fan:052]: Tuya Fan 'MyFan'
[19:54:04][C][tuya.fan:164]:   Speed: YES
[19:54:04][C][tuya.fan:165]:   Speed count: 3
[19:54:04][C][tuya.fan:054]:   Speed has datapoint ID 3
[19:54:04][C][tuya.fan:056]:   Switch has datapoint ID 1
[19:54:04][C][mdns:084]: mDNS:
[19:54:04][C][mdns:085]:   Hostname: living-room-fan
[19:54:04][C][ota:085]: Over-The-Air Updates:
[19:54:04][C][ota:086]:   Address: 192.168.2.133:8266
[19:54:04][C][ota:089]:   Using Password.
[19:54:04][C][api:138]: API Server:
[19:54:04][C][api:139]:   Address: 192.168.2.133:6053
[19:54:04][C][api:141]:   Using noise encryption: YES
[19:54:04][C][tuya:029]: Tuya:
[19:54:04][C][tuya:032]:   Initialization failed. Current init_state: 0
[19:54:04][C][tuya:037]:   If no further output is received, confirm that this is a supported Tuya device.
[19:54:04][E][tuya:391]: Initialization failed at init_state 0
```

ssieb, 2022-06-07:
Try baud rate 115200.

ssieb, 2022-06-07:
Also, try adding `debug:` to the uart config at 9600 first.

hydazz, 2022-06-10:
this is what i get using 9600 and 115200 baud rates
```yaml
[18:44:52][C][api:141]:   Using noise encryption: YES
[18:44:52][C][tuya:029]: Tuya:
[18:44:52][C][tuya:032]:   Initialization failed. Current init_state: 0
[18:44:52][C][tuya:037]:   If no further output is received, confirm that this is a supported Tuya device.
[18:45:04][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[18:45:05][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[18:45:05][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[18:45:05][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[18:45:05][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[18:45:06][E][tuya:391]: Initialization failed at init_state 0
```

sumur1au, 2022-07-17:
I also have this issue with my Fanco Smart Infinity-iD unit. The fan no longer works even with the remote after I swapped out the WR4 to an ESP-07S chip. Interestingly I have another Fanco Smart Infinity-iD that was purchased Oct 2021 that doesn't have any issues communicating with the Tuya MCU. I think there is either newer firmware or hardware with newer units. 

```
[16:39:04][C][wifi:491]: WiFi:
[16:39:04][C][wifi:353]:   Local MAC: 58:BF:25:DB:52:C0
[16:39:04][C][wifi:354]:   SSID: [redacted]
[16:39:04][C][wifi:355]:   IP Address: 192.168.2.21
[16:39:04][C][wifi:356]:   BSSID: [redacted]
[16:39:04][C][wifi:358]:   Hostname: 'master-balcony-fan'
[16:39:04][C][wifi:360]:   Signal strength: -65 dB ▂▄▆█
[16:39:04][C][wifi:364]:   Channel: 11
[16:39:04][C][wifi:365]:   Subnet: 255.255.255.0
[16:39:04][C][wifi:366]:   Gateway: 192.168.2.1
[16:39:04][C][wifi:367]:   DNS1: 192.168.0.11
[16:39:04][C][wifi:368]:   DNS2: 0.0.0.0
[16:39:04][C][logger:275]: Logger:
[16:39:04][C][logger:276]:   Level: DEBUG
[16:39:04][C][logger:277]:   Log Baud Rate: 0
[16:39:04][C][logger:278]:   Hardware UART: UART0
[16:39:04][C][uart.arduino_esp8266:102]: UART Bus:
[16:39:05][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[16:39:05][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[16:39:05][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[16:39:05][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[16:39:05][C][uart.arduino_esp8266:109]:   Data Bits: 8
[16:39:05][C][uart.arduino_esp8266:110]:   Parity: NONE
[16:39:05][C][uart.arduino_esp8266:111]:   Stop bits: 1
[16:39:05][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[16:39:05][C][tuya.fan:051]: Tuya Fan 'Master Balcony Fan'
[16:39:05][C][tuya.fan:147]:   Speed: YES
[16:39:05][C][tuya.fan:148]:   Speed count: 7
[16:39:05][C][tuya.fan:053]:   Speed has datapoint ID 3
[16:39:05][C][tuya.fan:055]:   Switch has datapoint ID 1
[16:39:05][C][web_server:125]: Web Server:
[16:39:05][C][web_server:126]:   Address: master-balcony-fan.local:80
[16:39:05][C][mdns:084]: mDNS:
[16:39:05][C][mdns:085]:   Hostname: master-balcony-fan
[16:39:05][C][ota:085]: Over-The-Air Updates:
[16:39:05][C][ota:086]:   Address: master-balcony-fan.local:8266
[16:39:05][C][ota:089]:   Using Password.
[16:39:05][C][api:138]: API Server:
[16:39:05][C][api:139]:   Address: master-balcony-fan.local:6053
[16:39:05][C][api:143]:   Using noise encryption: NO
[16:39:05][C][tuya:033]: Tuya:
[16:39:05][C][tuya:038]:   Configuration will be reported when setup is complete. Current init_state: 3
[16:39:05][C][tuya:041]:   If no further output is received, confirm that this is a supported Tuya device.
```

ssieb, 2022-07-17:
> this is what i get using 9600 and 115200 baud rates
>
> ```yaml
> [18:44:52][C][api:141]:   Using noise encryption: YES
> [18:44:52][C][tuya:029]: Tuya:
> [18:44:52][C][tuya:032]:   Initialization failed. [...]

I don't know what else to suggest.  It's not getting a reply from the MCU.

closed 2022-11-04 by hydazz