esphome/issues #3331: mcp23017 - interrupt specification missing/incomplete

fdepalo, 2022-06-01:
### The problem

Is the interrupt part to free the i2c bus missing of some integration ?
I can't find how to configure it

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

i2c

### Example YAML snippet

```yaml
i2c:
  sda: 26
  scl: 27
  scan: true
  id: bus_a
  frequency: 100khz

binary_sensor:
  - platform: gpio
    name: "MCP23017 Pin #1"
    pin:
      mcp23xxx: mcp23017_hub
      # Use pin number 1
      number: 0

      # One of INPUT or INPUT_PULLUP
      mode: INPUT_PULLUP
      inverted: true
  - platform: gpio
    name: "MCP23017 Pin #2"
    pin:
      mcp23xxx: mcp23017_hub
      # Use pin number 1
      number: 0
      # One of INPUT or INPUT_PULLUP
      mode: INPUT_PULLUP
      inverted: true
```

ssieb, 2022-09-16:
You need to provide more explanation of what you're trying to do.

fdepalo, 2022-10-06:
Hello @ssieb, i'm trying to use the interrupt pin to not overload the i2c bus by using the interrupt pin by reading the inputs only when the pin is changing state, am i wrong ?

podexperfectussum, 2022-12-07:
I'm also having trouble with this.  I have two MCP23017s connected to an ESP32 via I2C:

```
esphome:
  name: esp32uino-test
  platform: ESP32
  board: esp32dev
logger:
  level: DEBUG
i2c:
   - id: bus_a
     sda: GPIO21
     scl: GPIO22
     scan: True
     frequency: 100khz
mcp23017:
  - id: 'mcp23017_1'
    address: 0x20
  - id: 'mcp23017_2'
    address: 0x24
binary_sensor:
  - platform: gpio
    name: "grnsw1"
    pin:
      mcp23xxx: mcp23017_1
      number: 0
      interrupt: change
      mode:
        input: true
        pullup: true
      inverted: false
```

When I attached a button to pin 0 which should be A0 on the MCP and press the button, nothing gets logged to the ESPHome "logs" console.  According to the [MCP23017 datasheet](https://ww1.microchip.com/downloads/aemDocuments/documents/APID/ProductDocuments/DataSheets/MCP23017-Data-Sheet-DS20001952.pdf), pin 20 on the DIP package is INTA so if I connect this pin on the DIP package to a pin on the ESP32, the ESP32 should get an interrupt signal and look for action from the MCP.  The ESPHome documentation seems to acknowledge this:

> open_drain_interrupt (Optional, boolean): Configure the interrupt pin to open-drain mode. Useful when the MCP23008’s power supply is greater than 3.3 volts. Note that this pin will require a pull-up resistor (to 3.3 volts) when this mode is enabled.

> interrupt (Optional): Set this pin to trigger the INT pin on the component. Can be one of CHANGE, RISING, FALLING.

However, it doesn't make note anywhere of how I tell ESPHome which pin on the ESP32 the INT pin from the MCP is wired to.

ssieb, 2022-12-07:
esphome doesn't use the interrupt pin.  It always checks the pin states every loop.  The interrupt option just enables the interrupt pin for your use if you want.