esphome/issues #3333: Fail to install firmware on D1 mini ESP32 board

Stefan-van-Hattum, 2022-06-02:
### The problem

From within HomeAssistant with the ESPHome add-on I am able to install the firmware on ESP8266 boards (NodeMCU and also D1 mini) and also ESP32 NodeMCU boards without any issues, but now I have some ESP32 D1 mini boards but these fail to get the firmware installed via ESPHome.
When I click install in ESPHome and use the "Plug into this computer" option and download the .bin file and connect via ESPHome Web (which is successful) and try to upload the .bin It first says " Connecting" and after a while I get an error "Failed to initialize. Try resetting your device or holding the BOOT button while selecting your serial port until it starts preparing the installation."  

### Which version of ESPHome has the issue?

Home Assistant Add-on ESPHome version 2022.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Just updated to core-2022.6.0 but previous version also has same issue

### What platform are you using?

ESP32

### Board

Mini D1 ESP32

### Example YAML snippet

```yaml
esphome:
  name: test

esp32:
  board: esp32dev
  framework:
    type: arduino
```

### Anything in the logs that might be useful for us?

```txt
When using ESPHome-Flasher tool I see the following when I try to use the downloaded firmware from ESPHome in Home Assistant:

Using 'COM7' as serial port.
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting..........
Detecting chip type... ESP32
Connecting....

Chip Info:
- Chip Family: ESP32
- Chip Model: ESP32-D0WD-V3 (revision 3)
- Number of Cores: 2
- Max CPU Frequency: 240MHz
- Has Bluetooth: YES
- Has Embedded Flash: NO
- Has Factory-Calibrated ADC: YES
- MAC Address: 24:D7:EB:0F:4A:40
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
- Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)
```

dashdrum, 2022-06-02:
I have used the ESP32 D1 Mini boards successfully, and my esphome: section reads:

```
esphome:
  platform: ESP32
  board: wemos_d1_mini32
```

I have to guess that

```
esp32:
  board: wemos_d1_mini32
```

would work as well.

Stefan-van-Hattum, 2022-06-02:
I just tried your proposed solution but it fails again.
The board is okay as I can install different firmware (i.e. EMS-ESP-3_4_1-ESP32.bin) on it using esphome-flasher tool

Stefan-van-Hattum, 2022-06-02:
Okay, I managed to install it using a manual download and choose legacy format so I could install it via the esphome-flasher tool.
Still I think it is a bug in the ESPHome-web and I would love to see it would also work from there.
This is just a workaround for now.

closed 2022-10-16 by github-actions[bot]