esphome/issues #3335: Nextion LCD goto_page call works only one

nagyv, 2022-06-04:
### The problem

I'm trying to set up a two-page display that would need to switch the page every 10s. Currently, the HMI is set up to start with `page1`. The logs show that the page is expected to move regularly, but it happens only once. 

After installation, `page1` shows up, then it switches to `page0` and stays there forever. At the same time, the logs report the switch every 10 sec as expected.

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.1

### What platform are you using?

ESP8266

### Board

WEMOS-D1-MINI

### Component causing the issue

display

### Example YAML snippet

```yaml
esphome:
  name: internal-display

esp8266:
  # board: esp01_1m
  board: d1_mini

uart:
  rx_pin: D1
  tx_pin: D6
  baud_rate: 115200

time:
  - platform: homeassistant
    id: homeassistant_time
    on_time:
      - seconds: 0, 20, 40
        then:
        - lambda:
            ESP_LOGD("main", "move to page %d", 1);
            id(internal_display).goto_page("page1");
      - seconds: 10, 30, 50
        then:
        - lambda:
            ESP_LOGD("main", "move to page %d", 0);
            id(internal_display).goto_page("page0");
            
text_sensor:
  - platform: nextion
    nextion_id: internal_display
    component_name: page0.hour
    id: clock0
  - platform: nextion
    nextion_id: internal_display
    component_name: page1.hour
    id: clock1
    
display:
  - platform: nextion
    id: internal_display
    update_interval: 5s
```

### Anything in the logs that might be useful for us?

```txt
[10:13:13][I][app:102]: ESPHome version 2022.5.1 compiled on Jun  4 2022, 10:01:16
[10:13:13][C][wifi:491]: WiFi:
[10:13:13][C][wifi:353]:   Local MAC: 40:91:51:53:1A:B8
[10:13:13][C][wifi:354]:   SSID: [redacted]
[10:13:13][C][wifi:355]:   IP Address: 192.168.0.85
[10:13:13][C][wifi:356]:   BSSID: [redacted]
[10:13:13][C][wifi:358]:   Hostname: 'internal-display'
[10:13:13][C][wifi:360]:   Signal strength: -35 dB ▂▄▆█
[10:13:13][C][wifi:364]:   Channel: 9
[10:13:13][C][wifi:365]:   Subnet: 255.255.255.0
[10:13:13][C][wifi:366]:   Gateway: 192.168.0.1
[10:13:13][C][wifi:367]:   DNS1: 94.140.15.15
[10:13:13][C][wifi:368]:   DNS2: 192.168.0.1
[10:13:13][C][logger:274]: Logger:
[10:13:13][C][logger:275]:   Level: DEBUG
[10:13:13][C][logger:276]:   Log Baud Rate: 115200
[10:13:13][C][logger:277]:   Hardware UART: UART0
[10:13:13][C][uart.arduino_esp8266:102]: UART Bus:
[10:13:13][C][uart.arduino_esp8266:103]:   TX Pin: GPIO12
[10:13:13][C][uart.arduino_esp8266:104]:   RX Pin: GPIO5
[10:13:13][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[10:13:13][C][uart.arduino_esp8266:108]:   Baud Rate: 115200 baud
[10:13:13][C][uart.arduino_esp8266:109]:   Data Bits: 8
[10:13:13][C][uart.arduino_esp8266:110]:   Parity: NONE
[10:13:13][C][uart.arduino_esp8266:111]:   Stop bits: 1
[10:13:13][C][uart.arduino_esp8266:115]:   Using software serial
[10:13:13][C][nextion:117]: Nextion:
[10:13:13][C][nextion:118]:   Device Model:     NX4832F035_011R
[10:13:13][C][nextion:119]:   Firmware Version: 48
[10:13:13][C][nextion:120]:   Serial Number:    94893401C09B2E54
[10:13:13][C][nextion:121]:   Flash Size:       16777216
[10:13:13][C][nextion:122]:   Wake On Touch:    True
[10:13:13][C][homeassistant.time:010]: Home Assistant Time:
[10:13:13][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[10:13:13][C][mdns:084]: mDNS:
[10:13:13][C][mdns:085]:   Hostname: internal-display
[10:13:13][C][ota:085]: Over-The-Air Updates:
[10:13:13][C][ota:086]:   Address: internal-display.local:8266
[10:13:13][C][ota:089]:   Using Password.
[10:13:13][C][api:138]: API Server:
[10:13:13][C][api:139]:   Address: internal-display.local:6053
[10:13:13][C][api:141]:   Using noise encryption: YES
[10:13:20][D][main:041]: move to page 1
[10:13:30][D][main:046]: move to page 0
[10:13:40][D][main:041]: move to page 1
[10:13:50][D][main:046]: move to page 0
[10:14:00][D][main:041]: move to page 1
```

### Additional information

The related HMI file is available at https://www.dropbox.com/s/ko38a0l13vc10k8/internal-display.HMI?dl=0 (GitHub does not allow uploading HMI)

kbx81, 2022-07-26:
would love to hear if the above-linked PR fixes this issue for you! 😄

nagyv, 2022-08-01:
@kbx81 I decided to simplify the UI and solve everything in a single place. As the device is already locked, I can not test it easily.

I recommend closing the issue.

closed 2026-01-22 by kbx81