esphome/issues #3336: Black and empty screen on add-on's opening

Cecchellone, 2022-06-04:
### The problem

When opening the add-on from the sidebar, the view remains black without displaying nothing
When opening the add-on from the extension page, the view remains black for some minutes and then displays an `error 504, timed out` thrown by NGINX without displaying anything else
The docker container is started and seems to work

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.1

### What platform are you using?

ESP8266

### Anything in the logs that might be useful for us?

```txt
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-banner.sh: executing... 
-----------------------------------------------------------
 Add-on: ESPHome
 ESPHome add-on for intelligently managing all your ESP8266/ESP32 devices
-----------------------------------------------------------
 Add-on version: 2022.5.1
 You are running the latest version of this add-on.
 System: Ubuntu 22.04 LTS  (amd64 / qemux86-64)
 Home Assistant Core: 2022.6.1
 Home Assistant Supervisor: 2022.05.3
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] 10-requirements.sh: executing... 
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing... 
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] 30-dirs.sh: executing... 
[cont-init.d] 30-dirs.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[17:11:09] INFO: Waiting for dashboard to come up...
[17:11:09] INFO: Starting ESPHome dashboard...
2022-06-04 17:11:10,148 INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[17:11:10] INFO: Starting NGINX...
```

### Additional information

NGINX is used for reverse proxy
My hassio configuration uses SSL

When opening from sidebar:
[image]
When opening from add-on's page:
[image]

bearpawmaxim, 2022-06-06:
Hi @Cecchellone 
I have the same problem just after the homeassistant after startup due to the slow hardware (I have an old [cubietruck](https://linux-sunxi.org/Cubietech_Cubietruck) board). After some time (approx 2-3 minutes) the addon becomes available.
Maybe the same problem for you?

Cecchellone, 2022-06-06:
Hi! Thanks for the answer!
I'm running Hassio and all the docker-works on a dedicated x64 server with pretty good hardware (Intel i5 and 6GB of Ram on an NVME) so I suppose it's not an hardware problem.
Anyway i tryed leaving the add-on "loading" for half an hour and, unfortunately, nothing changes...

closed 2022-10-21 by github-actions[bot]