esphome/issues #3337: After enabling encryption, a lot of esp8266 devices are shown as offline, but are online?

fribse, 2022-06-05:
### The problem

I have activated encryption on the API interface.
After doing that, quite a lot of devices are shown as offline in the ESPHome list in Home Assistant.

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.2

### What platform are you using?

ESP8266

### Board

8285 in plug

### Component causing the issue

encryption

### Example YAML snippet

```yaml
api:
  password: !secret apipw
  encryption:
    key: !secret encryption
```

### Additional information

[image]

Without the encryption activated, everything is shown as online.

fribse, 2022-06-05:
I can see it sort of goes haywire on the esp8285, some of them really will not come online after that, so I'm once again removing the encryption from the API. It seems to mess up other stuff on the HA as well, but I can't really put a finger on it.
I have about 32 devices on esphome.

MasterCATZ, 2022-06-10:
mine have been doing the same thing last last update

fribse, 2022-06-10:
I've again given up on activating encryption, it got very unstable on all the esp8285 devices, really sad state.

closed 2022-10-23 by github-actions[bot]

itsieber, 2022-12-05:
Same on NSPanel