esphome/issues #3338: Sensor filters drop NAN, masking sensor problems

DAVe3283, 2022-06-05:
### The problem

If a sensor reports NAN, this will be silently dropped & ignored by most filters. If this happens rarely, it is not a problem. But if a sensor has a hard fault and stops reporting entirely, this causes the last filtered output value to get published over and over, even though it should be publishing NAN (Unknown).

This can be a fairly critical problem if the ESP device is hooked to a chiller, for example. If the sensor fails, but the thermostat component doesn't get a NAN reported, it can cause a chiller to stay running indefinitely, leading to freezing and broken pipes.

I am happy to create a PR to make at least a couple of the filters handle NAN properly (by publish NAN once enough NANs come in).

### Which version of ESPHome has the issue?

2022.5.1 (likely all versions)

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Any (I am using 2022.5.5)

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

sensor

### Example YAML snippet

```yaml
esphome:
  name: d1-mini-proto

esp8266:
  board: d1_mini

logger:
  level: DEBUG

api:
  password: !secret d1_mini_proto_api
  encryption:
    key: !secret d1_mini_proto_key

ota:
  password: !secret d1_mini_proto_ota

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "D1-Mini-Proto Fallback Hotspot"
    password: !secret d1_mini_proto_ota

captive_portal:

spi:
  clk_pin: GPIO14
  mosi_pin: MOSI
  miso_pin: MISO

external_components:
  # Waiting for https://github.com/esphome/esphome/pull/3505
  - source: github://DAVe3283/esphome@dev
    components: ["max31865"]

sensor:
  - platform: max31865
    name: "MAX31865 Debug"
    cs_pin: GPIO15
    reference_resistance: 4.3 kΩ
    rtd_nominal_resistance: 1 kΩ
    rtd_wires: 3
    update_interval: 5s
    filters:
      - median
```

### Anything in the logs that might be useful for us?

```txt
Notice how despite not being able to read the temperature, the filtered sensor reports a valid temperature, when it should be reporting NAN (unknown):

[11:26:47][E][max31865:099]: SPI bus read all 0 or all 1 (0x0000), check MAX31865 wiring & power.
[11:26:52][E][max31865:099]: SPI bus read all 0 or all 1 (0x0000), check MAX31865 wiring & power.
[11:26:57][E][max31865:099]: SPI bus read all 0 or all 1 (0x0000), check MAX31865 wiring & power.
[11:27:02][E][max31865:099]: SPI bus read all 0 or all 1 (0x0000), check MAX31865 wiring & power.
[11:27:07][E][max31865:099]: SPI bus read all 0 or all 1 (0x0000), check MAX31865 wiring & power.
[11:27:07][D][sensor:124]: 'MAX31865 Debug': Sending state 22.32755 °C with 2 decimals of accuracy
[11:27:12][E][max31865:099]: SPI bus read all 0 or all 1 (0x0000), check MAX31865 wiring & power.
[11:27:17][E][max31865:099]: SPI bus read all 0 or all 1 (0x0000), check MAX31865 wiring & power.
[11:27:22][E][max31865:099]: SPI bus read all 0 or all 1 (0x0000), check MAX31865 wiring & power.
[11:27:27][E][max31865:099]: SPI bus read all 0 or all 1 (0x0000), check MAX31865 wiring & power.
[11:27:32][E][max31865:099]: SPI bus read all 0 or all 1 (0x0000), check MAX31865 wiring & power.
[11:27:32][D][sensor:124]: 'MAX31865 Debug': Sending state 22.32755 °C with 2 decimals of accuracy
[11:27:37][E][max31865:099]: SPI bus read all 0 or all 1 (0x0000), check MAX31865 wiring & power.
[11:27:42][E][max31865:099]: SPI bus read all 0 or all 1 (0x0000), check MAX31865 wiring & power.
[11:27:47][E][max31865:099]: SPI bus read all 0 or all 1 (0x0000), check MAX31865 wiring & power.
[11:27:52][E][max31865:099]: SPI bus read all 0 or all 1 (0x0000), check MAX31865 wiring & power.
[11:27:57][E][max31865:099]: SPI bus read all 0 or all 1 (0x0000), check MAX31865 wiring & power.
[11:27:57][D][sensor:124]: 'MAX31865 Debug': Sending state 22.32755 °C with 2 decimals of accuracy
```
```

nagyrobi, 2022-06-06:
I just add `- filter_out: NAN` as the first filter before all. That will prevent NAN ftom reaching the next filters.

DAVe3283, 2022-06-07:
`- filter_out: NaN` has no net effect, with the previous value published by the sensor remaining indefinitely.

I am trying to figure out how to use a lambda filter to publish a NaN when the raw value is NaN. If I put that filter first, it gets eaten by the next filter (e.g. `median`). If I put the lambda last, it never receives a NaN. Maybe I can read the raw value from within the lambda, I will play with it tonight.

It looks like I could possibly use [`throttle_average` ](https://esphome.io/components/sensor/index.html?highlight=filter#throttle-average), but I still think the other filters should handle NaN properly. IMO it shouldn't be this complicated to properly handle this edge case. I will likely try my hand at editing the filters to handle NANs rather than just ignore them.

nagyrobi, 2022-06-07:
https://esphome.io/components/sensor/index.html#on-raw-value

closed 2022-07-28 by jesserockz