esphome/issues #3339: Dashboard Authentication CLI Arguments not Secret

DaAwesomeP, 2022-06-05:
### The problem

CLI arguments are visible to all users and processes on a machine, making the `--username` and `--password` arguments not secret. Authentication secrets should never be passed through CLI arguments (at least on Linux).

This can be seen by any user running `ps aux | grep esphome`.

These options should become environment variables like `ESPHOME_USERNAME` and `ESPHOME_PASSWORD`. Simply using variables like `USERNAME` and `PASSWORD` may conflict with system variables.

### Which version of ESPHome has the issue?

all

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

N/A

### Component causing the issue

dashboard

OttoWinter, 2022-06-14:
Those env variables already exist `USERNAME` and `PASSWORD`. The username/password is only passed via env variables for our docker/hassio images

DaAwesomeP, 2022-06-16:
It looks like the `PASSWORD` env var can be used in any installation, so that nulls the biggest concern that I had:

https://github.com/esphome/esphome/blob/06440d0202930de8c44cc6197281c19e566a3cd4/esphome/dashboard/dashboard.py#L64-L67

May I add this to the docs/CLI help?

OttoWinter, 2022-06-17:
Sure :+1:

closed 2022-10-31 by github-actions[bot]