esphome/issues #3340: Binary sensor not updated

Drealine, 2022-06-05:
### The problem

Binary sensor not updated when filter delayed_off is set more than 6seconds.

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.2

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

GPIO

Drealine, 2022-06-05:
Logs output show the initial state but is not updated on HA frontend

ssieb, 2022-09-16:
Your description is not clear.  Maybe you want `publish_initial_state: true`?  If that doesn't solve it, then reopen and add more details.

closed 2022-09-16 by ssieb