esphome/issues #3342: Nextion brightness argument breaks the script

nagyv, 2022-06-05:
### The problem

I have a script that runs fine and everything displays as expected on a Nextion display, but the display is really bright during the evening, so I'd like to decrease its brightness programmatically. Seems to be trivial, doesn't it?

Changing the brightness from script or using the `brightness` option of the display component makes the script fail. No updates happen to the screen.

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.2

### What platform are you using?

ESP8266

### Board

WEMOS-D1-MINI

### Component causing the issue

display/nexion

### Example YAML snippet

```yaml
esphome:
  name: internal-display

esp8266:
  # board: esp01_1m
  board: d1_mini

uart:
  rx_pin: D1
  tx_pin: D6
  baud_rate: 115200

# Enable logging
logger:
#  baud_rate: 0  # disables logging

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

time:
  - platform: homeassistant
    id: homeassistant_time
    on_time:
    - seconds: 0
      minutes: /1
      then:
      - lambda:
          auto time = id(homeassistant_time).now();
          id(nextion_clock).set_state(time.strftime("%H:%M"), false, true);

text_sensor:
  - platform: nextion
    nextion_id: internal_display
    id: nextion_clock
    component_name: page0.clock
  - platform: nextion
    nextion_id: internal_display
    id: nextion_temperature
    component_name: page0.temperature
      
sensor: 
  - platform: homeassistant   # Outside temperature
    id: temperature_outside    
    entity_id: sensor.trisensor_air_temperature
    on_value:
      then:
        - lambda:
            char buffer [6];
            int n;
            n=sprintf (buffer, "%2.1f", x);
            id(nextion_temperature).set_state(buffer, false, true);

display:
  - platform: nextion
    id: internal_display
    update_interval: 5s
    # Removing the following comment fails the script
    #brightness: 10%
```

### Anything in the logs that might be useful for us?

```txt
-
```

jesserockz, 2022-06-06:
Interesting you are having issues. 

I am able to set the brightness using the `brightness: 10%` config in yaml, and via 

```yaml
number:
  - platform: template
    name: "${friendly_name} Brightness"
    min_value: 0
    max_value: 1
    step: 0.1
    optimistic: true
    on_value:
      - lambda: id(nextion_display)->set_backlight_brightness(x);
```

nagyv, 2022-06-07:
@jesserockz, is there any difference between `id(nextion_display)->set_backlight_brightness(x);` and `id(nextion_display).set_backlight_brightness(x);`? I used the `dot` version.

nagyv, 2022-06-08:
@jesserockz Interesting. I added your code snippet to my working code, and seemingly nothing works now. No brightness updates, no clock, no temperature.

closed 2022-06-09 by jesserockz