esphome/issues #3343: Rename causes devices to become Unavailable

tiaanv, 2022-06-06:
### The problem

When using the new rename feature:

- I rename the device from some_device_name to some-device-name
- Esphome automatically compiles and updates the device.

After the update is complete, the device entities are no longer available.  The device is "online" on the ESPhome page, as per logs

Corrective measures taken in order:

- I restarted ESPHome - No improvement
- I restarted HomeAssistant - No improvement
- I rebooted the host - No Improvement

Ultimately to get the device and entities so work again I had to rename the devices back "manually" using the use_address option, since renaming back to a name with underscores cannot be dine (for good reason).

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.6.2

### What platform are you using?

ESP8266

### Board

D1 mini

### Component causing the issue

Rename

### Example YAML snippet

```yaml
esphome:
  name: "tv-strip"
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "VILJOEN"
  password: "wearesecure"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/tv-strip.yaml...
INFO Starting log output from tv-strip.local using esphome API
INFO Successfully connected to tv-strip.local
[07:08:33][I][app:102]: ESPHome version 2022.5.1 compiled on Jun  5 2022, 13:45:57
[07:08:33][C][wifi:491]: WiFi:
[07:08:33][C][wifi:353]:   Local MAC: 2C:F4:32:20:4C:CC
[07:08:33][C][wifi:354]:   SSID: [redacted]
[07:08:33][C][wifi:355]:   IP Address: 192.168.0.54
[07:08:33][C][wifi:356]:   BSSID: [redacted]
[07:08:33][C][wifi:358]:   Hostname: 'tv-strip'
[07:08:33][C][wifi:360]:   Signal strength: -56 dB ▂▄▆█
[07:08:33][C][wifi:364]:   Channel: 11
[07:08:33][C][wifi:365]:   Subnet: 255.255.255.0
[07:08:33][C][wifi:366]:   Gateway: 192.168.0.1
[07:08:33][C][wifi:367]:   DNS1: 192.168.0.1
[07:08:33][C][wifi:368]:   DNS2: 0.0.0.0
[07:08:33][C][logger:274]: Logger:
[07:08:33][C][logger:275]:   Level: DEBUG
[07:08:33][C][logger:276]:   Log Baud Rate: 115200
[07:08:33][C][logger:277]:   Hardware UART: UART0
[07:08:33][C][light:104]: Light 'TV Room RGBW Light'
[07:08:33][C][light:106]:   Default Transition Length: 1.0s
[07:08:33][C][light:107]:   Gamma Correct: 2.80
[07:08:33][C][mdns:084]: mDNS:
[07:08:33][C][mdns:085]:   Hostname: tv-strip
[07:08:33][C][ota:085]: Over-The-Air Updates:
[07:08:33][C][ota:086]:   Address: tv-strip.local:8266
[07:08:33][C][api:138]: API Server:
[07:08:33][C][api:139]:   Address: tv-strip.local:6053
[07:08:33][C][api:143]:   Using noise encryption: NO
```

### Additional information

Warning in the Log:
```
Logger: homeassistant.components.esphome
Source: components/esphome/__init__.py:314
Integration: ESPHome ([documentation](https://www.home-assistant.io/integrations/esphome), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+esphome%22))
First occurred: June 5, 2022, 20:17:52 (2045 occurrences)
Last logged: 07:09:29
```

Name of device dining_strip changed to dining-strip, potentially due to IP reassignment
Name of device tv_strip changed to tv-strip, potentially due to IP reassignment

bkaufx, 2022-06-06:
I think you'll have to remove the device from Home Assistant and re-add after renaming.  I'll be watching this to see if anyone has any better ideas.  I've had the same problem before when manually renaming.  It would be nice if the Home Assistant integration could handle this a little better but I'm not sure if it is possible.

ssieb, 2022-10-14:
HA and the esphome dashboard are completely independent.  HA does not know that you renamed the device, it only sees that the device doesn't match what it's expecting.  You have to remove the old name from the HA integrations list and add the new device.

tiaanv, 2022-10-14:
I'm not sue this has to be this way, and the integration can take care of renames appropriately.  Be that as it may.  This issue got no love, so I'm just closing it to move on.

closed 2022-10-14 by tiaanv

ssieb, 2022-10-14:
The HA integration has no way to know that you did a rename either.  The dashboard where you did the rename is completely independent from anything else in HA.

tiaanv, 2022-10-14:
I guess I don't understand the inner workings, but in my mind, at worst, when I rename an EspHome node, it should be the same as adding a new one, and the HA integration should discover it.  I am OK though with the solution being that you have to remove the Device from the Integrations page. Makes sense. Thx for the feedback.

ssieb, 2022-10-14:
It *is* the same as adding a new one.  The problem is that HA is already expecting something else at that IP address, so you have to remove it.