esphome/issues #3346: Doc issue: LEDC documentation may refer to the wrong board

Joshfindit, 2022-06-06:
### The problem

The documentation [here](https://github.com/esphome/esphome-docs/blob/current/components/output/ledc.rst) states

> The ESP8266 for instance has usually a frequency of 1000Hz with a resolution of 10 bits. This means that there are only 4 steps between each value.

This is directly below the chart for the bit-depth to frequency relationship of the ESP32 (specifically the boards with LEDC).

### Which version of ESPHome has the issue?

current

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

current

### What platform are you using?

ESP32

### Board

ESP32-Mini

Joshfindit, 2022-10-13:
Still an issue: documentation still has the same wording

ssieb, 2022-10-13:
This should be filed in the docs repo instead.  Can you make a PR to fix it?

Joshfindit, 2022-10-14:
@ssieb I can do the PR for sure as long as someone confirms that it should reference the ESP32 instead

ssieb, 2022-10-14:
I'm not sure what the intent of that line was.  It might just be a comparison of how an ESP8266 compares to using ledc on an ESP32.  It's not misleading or anything, just maybe confusing.