esphome/issues #3348: `deep_sleep` component fails to build on ESP8266

trombik, 2022-06-07:
### The problem

`deep_sleep` component fails to build on ESP8266.

### Which version of ESPHome has the issue?

2022.5.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

deep_sleep

### Example YAML snippet

```yaml
---
esphome:
  name: foo

esp8266:
  board: nodemcuv2

logger:
  level: debug

debug:
  update_interval: 5s

deep_sleep:
  run_duration: 10s
  sleep_duration: 3s
```

### Anything in the logs that might be useful for us?

```txt
esphome run --device /dev/cuaU0  config/test.yml
INFO Reading configuration config/test.yml...
INFO Generating C++ source...
INFO Compiling app...
Processing foo (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
Compiling .pioenvs/foo/src/esphome/components/deep_sleep/deep_sleep_component.cpp.o
Compiling .pioenvs/foo/src/esphome/core/component_iterator.cpp.o
Compiling .pioenvs/foo/src/esphome/core/controller.cpp.o
Compiling .pioenvs/foo/src/esphome/core/entity_base.cpp.o
Compiling .pioenvs/foo/src/esphome/core/helpers.cpp.o
Compiling .pioenvs/foo/src/esphome/core/log.cpp.o
Compiling .pioenvs/foo/src/esphome/core/scheduler.cpp.o
Compiling .pioenvs/foo/src/esphome/core/util.cpp.o
In file included from src/esphome/core/log.h:8,
                 from src/esphome/components/deep_sleep/deep_sleep_component.cpp:4:
src/esphome/components/deep_sleep/deep_sleep_component.cpp: In member function 'void esphome::deep_sleep::DeepSleepComponent::begin_sleep(bool)':
src/esphome/components/deep_sleep/deep_sleep_component.cpp:118:36: error: expected ')' before '__INT64'
  118 |     ESP_LOGI(TAG, "Sleeping for %" PRId64 "us", *this->sleep_duration_);
      |                                    ^~~~~~
/home/trombik/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:39:76: note: in definition of macro 'FPSTR'
   39 | #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
      |                                                                            ^~~~~~~~~~~~
src/esphome/core/log.h:67:36: note: in expansion of macro 'F'
   67 | #define ESPHOME_LOG_FORMAT(format) F(format)
      |                                    ^
src/esphome/core/log.h:105:58: note: in expansion of macro 'ESPHOME_LOG_FORMAT'
  105 |   esp_log_printf_(ESPHOME_LOG_LEVEL_INFO, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |                                                          ^~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:150:28: note: in expansion of macro 'esph_log_i'
  150 | #define ESP_LOGI(tag, ...) esph_log_i(tag, __VA_ARGS__)
      |                            ^~~~~~~~~~
src/esphome/components/deep_sleep/deep_sleep_component.cpp:118:5: note: in expansion of macro 'ESP_LOGI'
  118 |     ESP_LOGI(TAG, "Sleeping for %" PRId64 "us", *this->sleep_duration_);
      |     ^~~~~~~~
/home/trombik/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:40:34: note: to match this '('
   40 | #define F(string_literal) (FPSTR(PSTR(string_literal)))
      |                                  ^~~~
/home/trombik/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:39:76: note: in definition of macro 'FPSTR'
   39 | #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
      |                                                                            ^~~~~~~~~~~~
src/esphome/core/log.h:67:36: note: in expansion of macro 'F'
   67 | #define ESPHOME_LOG_FORMAT(format) F(format)
      |                                    ^
src/esphome/core/log.h:105:58: note: in expansion of macro 'ESPHOME_LOG_FORMAT'
  105 |   esp_log_printf_(ESPHOME_LOG_LEVEL_INFO, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |                                                          ^~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:150:28: note: in expansion of macro 'esph_log_i'
  150 | #define ESP_LOGI(tag, ...) esph_log_i(tag, __VA_ARGS__)
      |                            ^~~~~~~~~~
src/esphome/components/deep_sleep/deep_sleep_component.cpp:118:5: note: in expansion of macro 'ESP_LOGI'
  118 |     ESP_LOGI(TAG, "Sleeping for %" PRId64 "us", *this->sleep_duration_);
      |     ^~~~~~~~
*** [.pioenvs/foo/src/esphome/components/deep_sleep/deep_sleep_component.cpp.o] Error 1
```

### Additional information

see https://github.com/esphome/esphome/commit/8be704e591ce236e027448d1b40f72ab74378295

`time` component also has this issue.

https://github.com/esphome/esphome/blob/dev/esphome/components/time/automation.cpp#L45-L50

mjsambol, 2022-06-26:
I believe I'm seeing the same issue:

```
Compiling .pioenvs/upstairs_thermostat/src/esphome/components/time/automation.cpp.o
Compiling .pioenvs/upstairs_thermostat/src/esphome/components/time/real_time_clock.cpp.o
Compiling .pioenvs/upstairs_thermostat/src/esphome/components/web_server/list_entities.cpp.o
Compiling .pioenvs/upstairs_thermostat/src/esphome/components/web_server/web_server.cpp.o
In file included from src/esphome/core/log.h:8,
                 from src/esphome/components/time/automation.cpp:2:
src/esphome/components/time/automation.cpp: In member function 'virtual void esphome::time::CronTrigger::loop()':
src/esphome/components/time/automation.cpp:47:111: error: expected ')' before '__INT64'
   47 |     ESP_LOGD(TAG, "Second=%02u Minute=%02u Hour=%02u DayOfWeek=%u DayOfMonth=%u DayOfYear=%u Month=%u time=%" PRId64,
      |                                                                                                               ^~~~~~
/home/moshe/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:39:76: note: in definition of macro 'FPSTR'
   39 | #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
      |                                                                            ^~~~~~~~~~~~
src/esphome/core/log.h:67:36: note: in expansion of macro 'F'
   67 | #define ESPHOME_LOG_FORMAT(format) F(format)
      |                                    ^
src/esphome/core/log.h:92:59: note: in expansion of macro 'ESPHOME_LOG_FORMAT'
   92 |   esp_log_printf_(ESPHOME_LOG_LEVEL_DEBUG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |                                                           ^~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:152:28: note: in expansion of macro 'esph_log_d'
  152 | #define ESP_LOGD(tag, ...) esph_log_d(tag, __VA_ARGS__)
      |                            ^~~~~~~~~~
src/esphome/components/time/automation.cpp:47:5: note: in expansion of macro 'ESP_LOGD'
   47 |     ESP_LOGD(TAG, "Second=%02u Minute=%02u Hour=%02u DayOfWeek=%u DayOfMonth=%u DayOfYear=%u Month=%u time=%" PRId64,
      |     ^~~~~~~~
/home/moshe/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:40:34: note: to match this '('
   40 | #define F(string_literal) (FPSTR(PSTR(string_literal)))
      |                                  ^~~~
/home/moshe/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:39:76: note: in definition of macro 'FPSTR'
   39 | #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
      |                                                                            ^~~~~~~~~~~~
src/esphome/core/log.h:67:36: note: in expansion of macro 'F'
   67 | #define ESPHOME_LOG_FORMAT(format) F(format)
      |                                    ^
src/esphome/core/log.h:92:59: note: in expansion of macro 'ESPHOME_LOG_FORMAT'
   92 |   esp_log_printf_(ESPHOME_LOG_LEVEL_DEBUG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |                                                           ^~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:152:28: note: in expansion of macro 'esph_log_d'
  152 | #define ESP_LOGD(tag, ...) esph_log_d(tag, __VA_ARGS__)
      |                            ^~~~~~~~~~
src/esphome/components/time/automation.cpp:47:5: note: in expansion of macro 'ESP_LOGD'
   47 |     ESP_LOGD(TAG, "Second=%02u Minute=%02u Hour=%02u DayOfWeek=%u DayOfMonth=%u DayOfYear=%u Month=%u time=%" PRId64,
      |     ^~~~~~~~
*** [.pioenvs/upstairs_thermostat/src/esphome/components/time/automation.cpp.o] Error 1

```

trombik, 2022-08-02:
for those of you who would like to work around the issue:

```diff
--- components/deep_sleep/deep_sleep_component.cpp.orig	2022-08-02 07:30:45.252204000 +0700
+++ components/deep_sleep/deep_sleep_component.cpp	2022-08-02 07:33:40.745285000 +0700
@@ -115,7 +115,7 @@
 
   ESP_LOGI(TAG, "Beginning Deep Sleep");
   if (this->sleep_duration_.has_value())
-    ESP_LOGI(TAG, "Sleeping for %" PRId64 "us", *this->sleep_duration_);
+    ESP_LOGI(TAG, "Sleeping for %" "lld" "us", *this->sleep_duration_);
 
   App.run_safe_shutdown_hooks();
 
```

mjsambol, 2022-09-30:
Thanks for this suggestion, but I'm struggling wiht how to implement it. I ran esphome clean <component> then esphome compile <component>, got the error, edited the .cpp file under .esphome/build, ran compile again, and there it is again. It seems esphome is overwriting the files in .esphome/build each time I run compile. Is there a step I'm missing in the build process, or is there a different location to edit the source?